zephyrproject-rtos/zephyr #16786: [Coverity CID :198946]Memory - corruptions in /subsys/bluetooth/host/gatt.c

aasthagr, 2019-06-12:
Static code scan issues seen in File: /subsys/bluetooth/host/gatt.c
 Category: Memory - corruptions
 Function: bt_gatt_indicate
 Component: Bluetooth
 CID: 198946
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-06-12:
```
*** CID 198946:  Memory - corruptions  (OVERRUN)
/subsys/bluetooth/host/gatt.c: 1609 in bt_gatt_indicate()
1603     	handle = params->attr->handle ? : find_static_attr(params->attr);
1604     	if (!handle) {
1605     		return -ENOENT;
1606     	}
1607     
1608     	/* Check if attribute is a characteristic then adjust the handle */
>>>     CID 198946:  Memory - corruptions  (OVERRUN)
>>>     Overrunning struct type bt_uuid_16 of 4 bytes by passing it to a function which accesses it at byte offset 16.
1609     	if (!bt_uuid_cmp(params->attr->uuid, BT_UUID_GATT_CHRC)) {
1610     		struct bt_gatt_chrc *chrc = params->attr->user_data;
1611     
1612     		if (!(chrc->properties & BT_GATT_CHRC_INDICATE)) {
1613     			return -EINVAL;
1614     		}
```

jhedberg, 2019-06-13:
This one was marked as a false positive before this issue was opened - please check the Coverity state before creating new GitHub issues.

closed 2019-06-13 by jhedberg