zephyrproject-rtos/zephyr #16787: [Coverity CID :198945]Null pointer dereferences in /subsys/bluetooth/controller/ll_sw/ull_conn.c

aasthagr, 2019-06-12:
Static code scan issues seen in File: /subsys/bluetooth/controller/ll_sw/ull_conn.c
 Category: Null pointer dereferences
 Function: ull_conn_tx_lll_enqueue
 Component: Bluetooth
 CID: 198945
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-06-12:
```
*** CID 198945:  Null pointer dereferences  (FORWARD_NULL)
/subsys/bluetooth/controller/ll_sw/ull_conn.c: 1195 in ull_conn_tx_lll_enqueue()
1189     }
1190     
1191     void ull_conn_tx_lll_enqueue(struct ll_conn *conn, u8_t count)
1192     {
1193     	bool pause_tx = false;
1194     
>>>     CID 198945:  Null pointer dereferences  (FORWARD_NULL)
>>>     Passing "conn" to "is_enc_req_pause_tx", which dereferences null "conn->tx_ctrl_last".
1195     	while (conn->tx_head &&
1196     	       ((
1197     #if defined(CONFIG_BT_CTLR_DATA_LENGTH)
1198     		 !conn->llcp_length.pause_tx &&
1199     #endif /* CONFIG_BT_CTLR_DATA_LENGTH */
1200     #if defined(CONFIG_BT_CTLR_PHY)
```

cvinayak, 2019-06-14:
My understanding of the detected NULL pointer dereference could be with that the _conn->tx_ctrl_last_ is assigned NULL in the _while_ loop and would be dereferenced inside _is_enc_req_pause_tx_. But _conn->tx_ctrl_last_ is assigned NULL alongwith _conn->tx_ctrl_ and _is_enc_req_pause_tx_ tries to dereference _conn->tx_ctrl_last_ after checking for  _conn->tx_ctrl_ is not NULL.

This issue can be triaged as false positive.

Could @mped-oticon take a second look?

mped-oticon, 2019-06-14:
I've looked at the sources and reach the same conclusion as @cvinayak.

aescolar, 2019-06-14:
Closing then as false positive. @cvinayak please annotate it so in the coverity server.

closed 2019-06-14 by aescolar