zephyrproject-rtos/zephyr #16797: [Zephyr v1.14.0] stm32: MCUboot bootloader results in Hardware exception

gon1332, 2019-06-13:
**Describe the bug**
I tried to support MCUboot bootloader in STM32 platform SOCs but with no success. After building the bootloader from the mcuboot repository, when flashed it results in a Hardware Exception.

I've tried with the following Zephyr revisions:
* tags/v1.14.0
* v1.14-branch (6950a38cb3)
* master (f5d2d99762)

The target socs/boards I've tried are:
* nucleo_f401re (there is already DTS suppport)
* nucleo_l496zg (I've added DTS support)
* custom board with STM32L496RE

**To Reproduce**
Steps to reproduce the behavior:
1. `cd mcuboot/boot/zephyr`
2. `mkdir build; cd build`
3. `cmake -GNinja -DBOARD=nucleo_f4014re ..`
4. `ninja`
5. `ninja flash`
6. See the error

**Expected behavior**
See the bootloader boot.

**Impact**
Blockage

**Screenshots or console output**
* Zephyr@v1.14.0
```
***** USAGE FAULT *****
  Illegal load of EXC_RETURN into PC
***** Hardware exception *****
Current thread ID = 0x00000000
Faulting instruction address = 0x8008060
```

* Zephyr@v1.14-branch
```
***** USAGE FAULT *****
  Illegal load of EXC_RETURN into PC
***** Hardware exception *****
Current thread ID = 0x00000000
Faulting instruction address = 0x8008058
```

* Zephyr@master crashes at cmake
```
[1/138] Preparing syscall dependency handling

[23/138] Building C object CMakeFiles/app.dir/os.c.obj
FAILED: CMakeFiles/app.dir/os.c.obj 
ccache /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v1.14.0-1460-gf5d2d9976248 -DKERNEL -DSTM32F401xE -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../include -I../targets -I/home/gon1332/Development/Centaur/mcuboot/boot/bootutil/include -I/home/gon1332/Development/zephyr/kernel/include -I/home/gon1332/Development/zephyr/arch/arm/include -I/home/gon1332/Development/zephyr/include -I/home/gon1332/Development/zephyr/include/drivers -Izephyr/include/generated -I/home/gon1332/Development/zephyr/soc/arm/st_stm32/stm32f4 -I/home/gon1332/Development/zephyr/lib/libc/minimal/include -I/home/gon1332/Development/zephyr/drivers -I/home/gon1332/Development/zephyr/ext/hal/cmsis/Include -I/home/gon1332/Development/zephyr/ext/hal/st/stm32cube/stm32f4xx/soc -I/home/gon1332/Development/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/include -I/home/gon1332/Development/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/include/Legacy -isystem /opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include -isystem /opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include-fixed -Og -nostdinc -imacros /home/gon1332/Development/Centaur/mcuboot/boot/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -g -fno-common -mthumb -mcpu=cortex-m4 -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/gon1332/Development/zephyr=. -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -std=c99 -MD -MT CMakeFiles/app.dir/os.c.obj -MF CMakeFiles/app.dir/os.c.obj.d -o CMakeFiles/app.dir/os.c.obj   -c ../os.c
../os.c:27:10: fatal error: mbedtls/platform.h: No such file or directory
 #include <mbedtls/platform.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
[25/138] Building C object CMakeFiles/app.dir/home/gon1332/Development/Centaur/mcuboot/boot/bootutil/src/image_validate.c.obj
FAILED: CMakeFiles/app.dir/home/gon1332/Development/Centaur/mcuboot/boot/bootutil/src/image_validate.c.obj 
ccache /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v1.14.0-1460-gf5d2d9976248 -DKERNEL -DSTM32F401xE -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../include -I../targets -I/home/gon1332/Development/Centaur/mcuboot/boot/bootutil/include -I/home/gon1332/Development/zephyr/kernel/include -I/home/gon1332/Development/zephyr/arch/arm/include -I/home/gon1332/Development/zephyr/include -I/home/gon1332/Development/zephyr/include/drivers -Izephyr/include/generated -I/home/gon1332/Development/zephyr/soc/arm/st_stm32/stm32f4 -I/home/gon1332/Development/zephyr/lib/libc/minimal/include -I/home/gon1332/Development/zephyr/drivers -I/home/gon1332/Development/zephyr/ext/hal/cmsis/Include -I/home/gon1332/Development/zephyr/ext/hal/st/stm32cube/stm32f4xx/soc -I/home/gon1332/Development/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/include -I/home/gon1332/Development/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/include/Legacy -isystem /opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include -isystem /opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include-fixed -Og -nostdinc -imacros /home/gon1332/Development/Centaur/mcuboot/boot/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -g -fno-common -mthumb -mcpu=cortex-m4 -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/gon1332/Development/zephyr=. -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -std=c99 -MD -MT CMakeFiles/app.dir/home/gon1332/Development/Centaur/mcuboot/boot/bootutil/src/image_validate.c.obj -MF CMakeFiles/app.dir/home/gon1332/Development/Centaur/mcuboot/boot/bootutil/src/image_validate.c.obj.d -o CMakeFiles/app.dir/home/gon1332/Development/Centaur/mcuboot/boot/bootutil/src/image_validate.c.obj   -c /home/gon1332/Development/Centaur/mcuboot/boot/bootutil/src/image_validate.c
In file included from /home/gon1332/Development/Centaur/mcuboot/boot/bootutil/src/image_validate.c:28:
/home/gon1332/Development/Centaur/mcuboot/boot/bootutil/include/bootutil/sha256.h:41:14: fatal error: mbedtls/sha256.h: No such file or directory
     #include <mbedtls/sha256.h>
              ^~~~~~~~~~~~~~~~~~
compilation terminated.
[28/138] Building C object CMakeFiles/app.dir/home/gon1332/Development/Centaur/mcuboot/boot/bootutil/src/loader.c.obj
ninja: build stopped: subcommand failed.
```
**Environment (please complete the following information):**
 - OS: GNU/Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: various

gon1332, 2019-06-13:
@galazari

jli157, 2019-06-17:
You need to enable multi thread support for the board.

erwango, 2019-06-18:
@gon1332 , can you provide MCUBoot SHA1?

> You need to enable multi thread support for the board in MCUboot.

@jli157, is that something that has changed from 1.14 and current master?

gon1332, 2019-06-18:
@jli157 Indeed, it works. But why?

@erwango fc07eab118e673abaa0eb8634848b00c3487d42a

jli157, 2019-06-18:
MCUBoot is disabling the multithreading by default. However, I guess the flash driver on STM32 needs multi-threading support. This is why it crashes if you don't enable multithreading. 

@erwango the issue is being kept for long time. I have read an issue conversation page in which some guys suggested to enable the multi-threading for MCUBoot by default while the code owner insisted that MCUBoot doesn't need that.

gon1332, 2019-06-18:
I see, but for master of Zephyr everything is or seem OK, even if the multithreading is off from the MCBoot side.

jli157, 2019-06-18:
> I see, but for master of Zephyr everything is or seem OK, even if the multithreading is off from the MCBoot side.

I've not been following the master for a while. Maybe something has changed to address the issue?

carlescufi, 2019-07-22:
@gon1332 is this still an issue with the latest zephyr and mcuboot masters?

gon1332, 2019-07-24:
@carlescufi I hope that I check it until the end of the week. However the issue was with Zephyr v1.14. Nevertheless I'll check for master, v1.14 and v1.14.1-rc1.

carlescufi, 2019-08-13:
@gon1332 please do check this again since both v1.14-branch and master have moved quite a bit. If you can't reproduce please close the issue.

gon1332, 2019-08-14:
Well,

With MCUboot@master (ceb43f5f98951461407307640fc8867ee485a287):

- Zephyr@master (): OK
- Zephyr@v1.14.1-rc1 & v1.14-branch (5b8488a936): Build failing with:
```
~/D/m/b/z/build ❯❯❯ ninja                                                                                                                                                                                                               master
[1/218] Preparing syscall dependency handling

[18/218] Building C object CMakeFiles/app.dir/main.c.obj
FAILED: CMakeFiles/app.dir/main.c.obj 
ccache /home/gon1332/Software/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=v1.14.1-rc1 -DKERNEL -DMBEDTLS_CONFIG_FILE=\"mcuboot-mbedtls-cfg.h\" -DSTM32F401xE -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../include -I../targets -I/home/gon1332/Development/mcuboot/boot/bootutil/include -I/home/gon1332/Development/zephyrproject/zephyr/kernel/include -I/home/gon1332/Development/zephyrproject/zephyr/arch/arm/include -I/home/gon1332/Development/zephyrproject/zephyr/include -I/home/gon1332/Development/zephyrproject/zephyr/include/drivers -Izephyr/include/generated -I/home/gon1332/Development/zephyrproject/zephyr/soc/arm/st_stm32/stm32f4 -I/home/gon1332/Development/zephyrproject/zephyr/lib/libc/minimal/include -I/home/gon1332/Development/zephyrproject/zephyr/drivers -I/home/gon1332/Development/zephyrproject/zephyr/ext/hal/cmsis/Include -I/home/gon1332/Development/zephyrproject/zephyr/ext/hal/st/stm32cube/stm32f4xx/soc -I/home/gon1332/Development/zephyrproject/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/include -I/home/gon1332/Development/zephyrproject/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/include/Legacy -I/home/gon1332/Development/zephyrproject/zephyr/ext/lib/crypto/mbedtls/include -I/home/gon1332/Development/zephyrproject/zephyr/ext/lib/crypto/mbedtls/configs -isystem /home/gon1332/Software/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include -isystem /home/gon1332/Software/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include-fixed -Og -nostdinc -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -imacros /home/gon1332/Development/mcuboot/boot/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -Wno-main -fno-common -mthumb -mcpu=cortex-m4 -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Werror=implicit-int -Wpointer-arith -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -std=c99 -MD -MT CMakeFiles/app.dir/main.c.obj -MF CMakeFiles/app.dir/main.c.obj.d -o CMakeFiles/app.dir/main.c.obj   -c ../main.c
../main.c:22:10: fatal error: drivers/timer/system_timer.h: No such file or directory
 #include <drivers/timer/system_timer.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[23/218] Building C object CMakeFiles/app.dir/os.c.obj
ninja: build stopped: subcommand failed.
```

After resolving the compiler error and building successfully on v1.14-branch, the bootloader still  results in Hardware exception:
```
***** Hardware exception *****
Current thread ID = 0x00000000
Faulting instruction address = 0x48024001
***** USAGE FAULT *****
  Illegal load of EXC_RETURN into PC
***** Hardware exception *****
Current thread ID = 0x00000000
Faulting instruction address = 0x48024001

```

carlescufi, 2019-08-15:
The issue here is that MCUboot master may not work anymore with v1.14-branch. Maybe @nvlsianpu can help you figure out which version of MCUboot will work with v1.14. We are working towards adding MCUboot as a west project so that this uncertainty in versions is eliminated.

nvlsianpu, 2019-08-16:
@gon1332 From conversation I'm understanding that the Hw exception for zephyr v1.14 you had is caused by a flash driver which requires synchronisation primitives (not working while multithreding id disabled). Didn't you try to backport driver bugfix to v1.14-branch.

I believe that apart from [this patch](https://github.com/JuulLabs-OSS/mcuboot/pull/497) mcuboot master is compatible with zephyr v1.14

gon1332, 2019-08-19:
> @gon1332 From conversation I'm understanding that the Hw exception for zephyr v1.14 you had is caused by a flash driver which requires synchronisation primitives (not working while multithreding id [...]

@nvlsianpu You are correct. The actual problem is the multithreading support. It's not a problem for me since the time I opened the issue. But it is still a bad first impression if a newcomer tries to use them and observe this issue.

Is there a backport fixing this driver? Could you please point me to it?

@carlescufi Also adding MCUboot as a west project is a good idea.

carlescufi, 2019-08-19:
@gon1332 thank you, this is being tracked elsewhere already so I will close this issue.

closed 2019-08-19 by carlescufi