zephyrproject-rtos/zephyr #16809: TCP2: Steps to intergrate

laperie, 2019-06-13:
The final steps of preparing new TCP stack for integration to Zephyr tree will be tracked in this issue.  As a result of these steps, the TCP2 should be available under certain KConfig flag (mutually exclusive with existing TCP stack) with existing TCP unit tests passing, existing samples functional and comparable Maxwell Pro results

These are the steps to integrate TCP2 to the existing net stack:
- [ ] Use specific KConfig option
- [ ] Integrate to connection.c  (RX side of things)
- [ ] Integrate with net_context API (RX and TX side of things)  

We expect to run the following tests on the TCP2
- [ ] Current TCP unit tests (passing)
- [ ] Maxwell Pro  (comparable pass rate)
- [ ] Samples (all working)

jukkar, 2019-06-13:
>     * Maxwell Pro  (comparable pass rate)

For integration work, I think this is not really needed. It is very much needed when we want to replace TCP1 by TCP2.

jukkar, 2019-10-29:
TCP2 was merged to upstream last Friday so we will concentrate next to test it.

dleach02, 2019-11-05:
There are a number of issues on this ticket that look like they will be v2.2 (and potentially beyond) items. It would make sense to mapout the "todo" items over the various release milestones.

carlescufi, 2020-03-11:
@jukkar could you update the description with any progress made in the last couple of months?

jukkar, 2020-03-11:
I need to pass this req to the source, @ozhuraki could you update the description about what has been done and what is still missing.

carlescufi, 2020-04-21:
@jukkar is this on track for 2.3?

jukkar, 2020-04-21:
> is this on track for 2.3?

Yes, we are doing our best to improve the functionality and test coverage more.

jukkar, 2020-08-10:
TCP2 is now the default -> closing.

closed 2020-08-10 by jukkar