zephyrproject-rtos/zephyr #16814: support for RISCV64 and SMP

thomascp, 2019-06-13:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Add support for RISCV64 and SMP.

**Describe alternatives you've considered**
No.

**Additional context**
I had added some quick support for it, please refer to 
https://github.com/thomascp/zephyr
This quick support had been tested on Renode Hifive Unleashed SMP board.
work based on branch : v1.14-branch

thomascp, 2019-06-26:
Will anyone comment on this feature request? Or let me know if I made the wrong progress.

carlescufi, 2019-06-26:
@thomascp why not send a Draft Pull Request with the contents of your branch?
@npitre are you by any chance looking at adding support for this as well?

thomascp, 2019-06-26:
@carlescufi I don't really know the workflow here, should I  send the pull request first?
It is only tested on Renode simulate board, is that enough? Is there any document to guide this?
Thanks!

PiotrZierhoffer, 2019-06-26:
@thomascp steps you should take:

1. read https://github.com/zephyrproject-rtos/zephyr/blob/master/CONTRIBUTING.rst and https://docs.zephyrproject.org/latest/contribute/index.html
2. clean up you repo a little - remove Renode from the tree, prepare commits so that they have significant and well described changes, following the above guidelines
3. submit a PR, possibly marking it as work in progress - [WIP], and describe all your concerns about testing and whatever insight about this code you'd like to share with others

npitre, 2019-06-26:
> @npitre are you by any chance looking at adding support for this as well?

Yes. But the core Zephyr code has to be 64-bit clean before that can 
happen. As proposed, I don't see how the RV64 support could ever work.

thomascp, 2019-06-26:
@npitre I had added RV64 and SMP support, please check https://github.com/thomascp/zephyr/commit/4ba217beabf552793619f935ce59be9484cc7276.

npitre, 2019-06-26:
On Wed, 26 Jun 2019, thomascp wrote:

> @npitre I had added RV64 and SMP support, please check https://github.com/thomascp/zephyr/commit/4ba217beabf552793619f935ce59be9484cc7276.

I did check it. This is far far from sufficient.

I can guarantee you that this won't pass the Zephyr test suite, and that 
it would even regress existing platforms.

Furthermore your commit contains a large amount of unrelated changes.

You should split your changes into several smaller changes, each with 
their own explanation. For example, renaming riscv32 to riscv is one 
such change that has to be made separately. Then you make sure that 
change alone doesn't break the test suite.

Then the introduction of LR/SR into the assembly code. And so on.

I also suggest you have a look at the z64_1.14 branch of 
https://github.com/npitre/zephyr which contains all the preliminary 
work before RV64 in mainline Zephyr could be possible.

thomascp, 2019-06-27:
@npitre Yes, I agree with you, this is just quick support for RV64 and SMP.
I am new in Zephyr, but add these just for fun. It is far away from a commercial and stable version, but it can run with multi-thread and multi-core which is I want. Actually, I don't like commit thing, this is not fun and may only cause trouble, should be done by some man who works in a company for this?
Thank you for advice, yes, it must be split into several commits.
About z64_1.14 branch, nice preliminary jobs, right now I only modified some file directly for RV64 support, I think my modifications should be merged after your jobs.

npitre, 2024-02-08:
This was completed long time ago.

closed 2024-02-08 by npitre