zephyrproject-rtos/zephyr #16823: k_busy_wait() on nRF5x boards isn't waiting long enough

andyross, 2019-06-13:
The nrf52840_pca10056 uses an arch-supplied implementation for k_busy_wait().  And it seems the clock being used is consistently slow by 10%, resulting in busy wait times that are incorrect.  This can be detected in tests now that high tick rates are being used (c.f. #16782)

See the code below.  It requests successively larger delays to k_busy_wait(), checks them against the cycle counter, and and there is a persistent 10% error between the two (even as the delays get into the multiple seconds range and should be utterly swamping any per-call overhead), e.g.:

```
***** Booting Zephyr OS zephyr-v1.14.0-1473-gf441944ad0e7 *****
us 30 cyc 2 (== ~61us, err = 103%)
us 37 cyc 2 (== ~61us, err = 64%)
us 46 cyc 2 (== ~61us, err = 32%)
us 57 cyc 2 (== ~61us, err = 7%)
us 71 cyc 3 (== ~91us, err = 28%)
us 88 cyc 4 (== ~122us, err = 38%)
us 110 cyc 4 (== ~122us, err = 10%)
us 137 cyc 5 (== ~152us, err = 10%)
...
us 6067090 cyc 219502 (== ~6698669us, err = 10%)
us 7583862 cyc 274425 (== ~8374786us, err = 10%)
us 9479827 cyc 343224 (== ~10474365us, err = 10%)
```

Disabling CONFIG_ARCH_HAS_CUSTOM_BUSY_WAIT produces correct behavior (as would be expected -- the default implementation is based on k_cycle_get_32() of course).  The error drops rapidly to zero as the delays get large.

The buggy z_arch_busy_wait() implementation is just a call to a HAL layer, so I don't have any visibility into how to fix it.  Presumably the relevant clock rates are simply misconfigured?  I don't know if this is generically wrong for all nRF5x SoC's or if the wrong rate is specific to the nrf52840_pca10056.

```c
#include <zephyr.h>
#include <misc/printk.h>

void main(void)
{
	u32_t us, t0, t1, cyc, expect_us, err_pct;

	/* Start at a busy wait of 30us, increment by 25% until we run
	 * out of precision
	 */
	for (us = 30; us < 10000000; us += us / 4) {

		/* Timestamp a busy wait on either side with the cycle
		 * counter
		 */
		t0 = k_cycle_get_32();
		k_busy_wait(us);
		t1 = k_cycle_get_32();

		/* Compute the cycles we delayed, convert to
		 * microseconds, then compute an error vs. the
		 * requested delay as a percent.
		 */
		cyc = t1 - t0;
		expect_us = (int)((cyc * 1000000ull)
				  / CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC);
		err_pct = (100ull * (expect_us - us)) / us;

		printk("us %d cyc %d (== ~%dus, err = %d%%)\n", us, cyc,
		       expect_us, err_pct);
	}
}
```

andyross, 2019-06-13:
Stumbled into this while tuning the timer stuff.   10% off hasn't really been noticeable so far, but it is now.  FWIW: having the busy wait implementation and the timing implementation be on (seemingly) different hardware clock bases may not be the best choice anyway.   The default implementation is super small and leverages the code already present for the cycle counter.  I'd suggest sticking with that.

pabigot, 2019-06-13:
This sort of delay can happen if the wait loop code isn't aligned to the proper offset for caching; see [this comment](https://github.com/pabigot/nrfcxx/blob/master/src/nrf52/delay_cycles.S#L4).  Dunno why that would suddenly be occurring with Nordic's code, though, since it has an align directive.  I tested it back in the 1.14 days and it was dead on.

If the current standard busywait implementation is correct the custom busywait could be reverted.  It was way wrong (terminated early) in previous versions of Zephyr (see  #11626).

pabigot, 2019-06-14:
@andyross Your 10% overhead is because #16782 is causing the timer ISR to be entered every 60 us.

The SOC busy-wait is looping for the number of CPU clock cycles required to cover the requested duration.  It does this with interrupts enabled, so any extra work that happens during the loop is added to the apparent delay.  This is a cost we pay for having accurate short-duration (<= 100 us) delays; see da3f7feaf8f48b04d018b25ee8c729013880361a.

Reverting use of that independently-measured delay as done by f82822c600e0c95fcd860663ca6b969097d211b2 hides this behavior.

I'd suggest using a logic analyzer to verify that the timer ISR is being called as you expect.

pizi-nordic, 2019-06-14:
@andyross: The default k_busy_wait() implementation is imprecise due to 2 reasons:
1. It bases on CPU execution speed (which is affected by icache).
2. It runs on RC clock by default (which is imprecise).

The (1) has been chosen as the default implementation because it can run on any Nordic SoC
The (2) is the default as RC oscillator requires less power than crystal.

To get the most precise busy wait, you have to do 2 things:
- Use DWT instead of NOP-based delay (just set NRFX_DELAY_DWT_BASED to 1 in ext/hal/nordic/nrfx_glue.h)
- Switch CPU clock to crystal by calling clock_control_on(device_get_binding(DT_NORDIC_NRF_CLOCK_0_LABEL "_16M"), UINT_TO_POINTER(1));

andyross, 2019-06-14:
Hah.  Yeah, @pabigot saw it.  I'd broken the logic at the end of the process and was spamming interrupts.  Running that test code against master (which I would have sworn I did originally, but clearly didn't) works fine.  Closing.

closed 2019-06-14 by andyross

andyross, 2019-06-14:
Hah again.  Actually the version I pushed seems to have fixed that bug anyway.  We get correct ISR counts and correct busy wait durations (to within about .5%, but it's high and not low and tests don't see it).  Not even sure which version I tested this one against originally.

Clearly I rushed this report.  Sorry :)

pabigot, 2019-06-14:
@andyross Not sure how you're doing things, but I reproduced the problem of spamming ISRs on f82822c600e0c95fcd860663ca6b969097d211b2 which is #16782, with the application in this report, using this patch.
```
diff --git a/drivers/timer/nrf_rtc_timer.c b/drivers/timer/nrf_rtc_timer.c
index ddff1f5fd1..60c4635083 100644
--- a/drivers/timer/nrf_rtc_timer.c
+++ b/drivers/timer/nrf_rtc_timer.c
@@ -50,6 +50,7 @@ static u32_t counter(void)
 void rtc1_nrf_isr(void *arg)
 {
 	ARG_UNUSED(arg);
+	NRF_GPIO->OUTSET = (1U << 30);
 	RTC->EVENTS_COMPARE[0] = 0;
 
 	k_spinlock_key_t key = k_spin_lock(&lock);
@@ -70,6 +71,7 @@ void rtc1_nrf_isr(void *arg)
 
 	k_spin_unlock(&lock, key);
 	z_clock_announce(IS_ENABLED(CONFIG_TICKLESS_KERNEL) ? dticks : 1);
+	NRF_GPIO->OUTCLR = (1U << 30);
 }
 
 int z_clock_driver_init(struct device *device)
@@ -85,6 +87,11 @@ int z_clock_driver_init(struct device *device)
 
 	clock_control_on(clock, (void *)CLOCK_CONTROL_NRF_K32SRC);
 
+	NRF_GPIO->PIN_CNF[30] = ((GPIO_PIN_CNF_SENSE_Disabled << GPIO_PIN_CNF_SENSE_Pos)
+			    | (GPIO_PIN_CNF_DRIVE_S0S1 << GPIO_PIN_CNF_DRIVE_Pos)
+			    | (GPIO_PIN_CNF_INPUT_Disconnect << GPIO_PIN_CNF_INPUT_Pos)
+			       | (GPIO_PIN_CNF_DIR_Output << GPIO_PIN_CNF_DIR_Pos));
+
 	/* TODO: replace with counter driver to access RTC */
 	nrf_rtc_prescaler_set(RTC, 0);
 	nrf_rtc_cc_set(RTC, 0, CYC_PER_TICK);
```

That GPIO toggles at 16 KiHz on a logic analyzer, while there aren't supposed to be any timers set.

Also after a while I get:
```
***** BUS FAULT *****     err = -1%)                    
  Unstacking error       s, err = 0%)                   
***** Hardware exception *****
***** HARD FAULT *****
  Fault escalation (see below)
***** BUS FAULT *****
  Precise data bus error
  BFAR Address: 0xbd525fb4
```
Please recheck the PR contents against what you're testing.

andyross, 2019-06-14:
FWIW: the spam bug was the case where **long** timeouts (e.g. K_FOREVER) would because of the modular space be indistinguishable from missed ones, so it would set "idle" timeouts for 60ms.  It's definitely fixed in my tree, which has a few other things added as I work through the full test suite.  I just never put the clues together that this bug was just a symptom of the other issue (because my brain was in a "poll a counter" paradigm and I didn't realize the busy wait was a calibrated spin).  I'll push the new version as soon as I get it cleaned up.

Regardless, this bug as reported was wrong.  It works fine.