zephyrproject-rtos/zephyr #16851: west flash error on zephyr v1.14.99

xuefli, 2019-06-15:
enviroment:
host: macOS 10.13.6 High Sierra
board: stm32 EUCLEO-L476RG

run with sample/hello_word

can flash zephyr.elf with STM32 Cube Programer,but with west flash can not flash 
the console log like follow:

```
[zephyr]$west -v flash
ZEPHYR_BASE=/Users/xxx/zephyr/zephyr (origin: env)
ninja: no work to do.
Using runner: openocd
/usr/local/bin/openocd -f /Users/xxx/zephyr/zephyr/boards/arm/nucleo_l476rg/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase /Users/xxx/zephyr/zephyr/build/zephyr/zephyr.elf' -c 'reset halt' -c 'verify_image /Users/xxx/zephyr/zephyr/build/zephyr/zephyr.elf' -c 'reset run' -c shutdown
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
/Users/xxx/zephyr/zephyr/boards/arm/nucleo_l476rg/support/openocd.cfg:1: Error: Can't find board/st_nucleo_l4.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 60
at file "/Users/xxx/zephyr/zephyr/boards/arm/nucleo_l476rg/support/openocd.cfg", line 1
ERROR: command exited with status 1: /usr/local/bin/openocd -f /Users/xxx/zephyr/zephyr/boards/arm/nucleo_l476rg/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase /Users/xxx/zephyr/zephyr/build/zephyr/zephyr.elf' -c 'reset halt' -c 'verify_image /Users/xxx/zephyr/zephyr/build/zephyr/zephyr.elf' -c 'reset run' -c shutdown
Traceback (most recent call last):
  File "/Users/xxx/zephyr/.west/west/src/west/main.py", line 580, in main
    args.handler(args, unknown)
  File "/Users/xxx/zephyr/.west/west/src/west/main.py", line 332, in ext_command_handler
    command.run(*west_parser.parse_known_args(argv))
  File "/Users/xxx/zephyr/.west/west/src/west/commands/command.py", line 90, in run
    self.do_run(args, unknown)
  File "/Users/xxx/zephyr/zephyr/scripts/west_commands/flash.py", line 32, in do_run
    'ZEPHYR_BOARD_FLASH_RUNNER')
  File "/Users/xxx/zephyr/zephyr/scripts/west_commands/run_common.py", line 232, in do_run_common
    runner.run(command_name)
  File "/Users/xxx/zephyr/zephyr/scripts/west_commands/runners/core.py", line 407, in run
    self.do_run(command, **kwargs)
  File "/Users/xxx/zephyr/zephyr/scripts/west_commands/runners/openocd.py", line 90, in do_run
    self.do_flash(**kwargs)
  File "/Users/xxx/zephyr/zephyr/scripts/west_commands/runners/openocd.py", line 122, in do_flash
    self.check_call(cmd)
  File "/Users/xxx/zephyr/zephyr/scripts/west_commands/runners/core.py", line 466, in check_call
    subprocess.check_call(cmd)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/subprocess.py", line 291, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/local/bin/openocd', '-f', '/Users/xxx/zephyr/zephyr/boards/arm/nucleo_l476rg/support/openocd.cfg', '-c', 'init', '-c', 'targets', '-c', 'reset halt', '-c', 'flash write_image erase /Users/xxx/zephyr/zephyr/build/zephyr/zephyr.elf', '-c', 'reset halt', '-c', 'verify_image /Users/xxx/zephyr/zephyr/build/zephyr/zephyr.elf', '-c', 'reset run', '-c', 'shutdown']' returned non-zero exit status 1.
[zephyr]$
```

how to config board/st_nucleo_l4.cfg and which folder to put

ioannisg, 2019-06-18:
@erwango could you also take a look here?

erwango, 2019-06-19:
@xuefli , which version of openocd are you using?

carlescufi, 2019-06-19:
@erwango from the log: `Open On-Chip Debugger 0.10.0`

erwango, 2019-06-20:
> @erwango from the log: Open On-Chip Debugger 0.10.0

Thanks! the worst part is that I actually searched before asking.

erwango, 2019-06-20:
@xuefli , OpenOCD version 0.10.0 is a bit old.
In Zephyr SDK, OpenOCD version is 0.10.0+dev-g1df07a9a4-dirty.

So in your version, the file to be used is : st_nucleo_l476rg.cfg.
But the best way would be to align your OpenOCD on Zephyr SDK version.

erwango, 2019-07-23:
@xuefli , did the previous comment help to solve your issue?

nashif, 2019-12-12:
no response, closing.

closed 2019-12-12 by nashif

wuruoyu, 2023-02-08:
@erwango Hi, I also encounter this issue when flashing to ST NUCLEO-F746ZG, with OpenOCD version 0.10.0. I am new to embedded systems and do not find a good solution to upgrade the OpenOCD. Could you drop me some hints on how to "align your OpenOCD on Zephyr SDK version"? 
I really appreciate any help you can provide!

erwango, 2023-02-09:
> o encounter this issue when flashing to ST NUCLEO-F746ZG, with OpenOCD version 0.10.0. I am new to embedded systems and do not find a good solution to upgrade the OpenOCD. Could you drop me some [...]

I suggest to follow Zephyr getting started guide and get the latest SDK which provides the latest openocd
Otherwise, you can get it here: https://github.com/openocd-org/openocd