zephyrproject-rtos/zephyr #16861: nRF52: UARTE: Data corruption right after resuming device

endian-benjamin, 2019-06-17:
Providing a bare-bones reproducible example isn't easy in this case, but I can reproduce it reliably with customer code. It seems that when you suspend the UARTE and then resume it (using `device_set_pm_state()`), there is a window right after resume in which transmissions get corrupted. This is the setup:

1. pm hooks suspends UART and `cpu_idle()`
2. timer interrupt triggers; pm hooks resume UART, ISR pushes item to work queue
3. work item TX:es eight bytes and RX:es eight bytes

I have verified by oscilloscope that the bytes TXed are correct and the bytes RXed are correct on the serial line. However, when my application reads the RX bytes, they are garbled (and an extra byte is added). This can be reproduced consistently, but only if the power management hooks suspending/resuming the UART are enabled.

It can be worked around by adding a dummy RX prior to TXing, so it's not a high prio bug (for me atleast :P) but it's clearly indicative of something generating spurious events.

Mierunski, 2019-06-17:
Hi, thank you for sharing!

Could you please elaborate on which API you are using? Polling, interrupt or asynchronous?

ioannisg, 2019-08-19:
@endian-benjamin can you elaborate a bit more here?

endian-benjamin, 2019-08-19:
@ioannisg @Mierunski Using the interrupt driven approach (with a wrapper handler).

So

- pm hooks calls `device_set_power_state` to wake up UART
- work item sets tx irq handler and calls `uart_irq_tx_enable`
- work item enables rx interrupts and waits for response
- response on the line is, say, `AA BB CC DD EE`, verified by oscilloscope
- work item receives NOT five bytes, but SIX, and they are garbled. 

If I disable power manager, then no bytes are garbled and no bytes are added. The current workaround is to flush the RX line prior to every TX (by triggering an RX and simply discarding the bytes received), which works fine for our use case.

ioannisg, 2019-08-26:
@Mierunski could you follow up with this one?

ioannisg, 2019-08-27:
@endian-benjamin could you please, check the patch proovided by @Mierunski ?

PR #18696 (2019-08-28): drivers: uart: Properly configure pins in NRF UARTE power management
Configure pins before enabling and after disabling UARTE through
power management functions.

Pins without proper prior configuration could result in undefined behaviour of UARTE at start
Fixes #16861
PR #19477 (2019-11-07): [Backport v1.14-branch] drivers: uart: Properly configure pins in NRF UARTE power management
Backport #18696.

Fixes #16861
closed 2019-08-28 by ioannisg