zephyrproject-rtos/zephyr #16872: Bluetooth: LL: Peripheral crashes after a while with multiple Centrals

paulgavrikov, 2019-06-17:
**Describe the bug**
After two centrals have connected to the same peripheral the stack will crash after a while. No GATT operations are happening. Both connections are IDLE. This behavior was not replicated with zephyr-v1.14.0.

Log:

> [15:45:53:913] ***** Kernel OOPS! *****␍␊
> [15:45:53:925] Current thread ID = 0x20004040␍␊
> [15:45:53:925] Faulting instruction address = 0x91e␍␊
> [15:45:53:925] Fatal fault in ISR! Spinning...␍␊

addr2line:

> zephyr/subsys/bluetooth/controller/ll_sw/ctrl.c:8445

Faulting code line:
```
static void event_connection_prepare(u32_t ticks_at_expire,
				     u32_t remainder, u16_t lazy,
				     struct connection *conn)
{
	u16_t event_counter;

	LL_ASSERT(!_radio.ticker_id_prepare);
>>>>>	_radio.ticker_id_prepare = RADIO_TICKER_ID_FIRST_CONNECTION +
				   conn->handle;
```

**To Reproduce**
Steps to reproduce the behavior:
1. Setup peripheral (nRF52840-DK)
2. Connect multiple centrals (iPhone 6s, nRF52840-dongle via nRF Connect) to the same peripheral
3. Wait a bit (1-2min)
4. Enjoy Kernel OOPS

**Expected behavior**
Stack shouldn't crash.

**Impact**
Showstopper

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Kubuntu 18.04 
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK
 - Commit SHA or Version used: 591b0e1c7af2df9f1f192336073962ea0f83b9d1

cvinayak, 2019-06-17:
@paulgavrikov Could you please revert 8a92818c0a and check if the issue exists?

paulgavrikov, 2019-06-18:
@cvinayak works when reverting 8a92818c0a0ceea30a9540e3e1557dec2cdf6fbf

aescolar, 2019-06-18:
Closing assuming that after the revert, this works, per [your comment](https://github.com/zephyrproject-rtos/zephyr/issues/16872#issuecomment-503038373)

closed 2019-06-18 by aescolar