zephyrproject-rtos/zephyr #16880: Systematic *-zephyr-eabi/bin/ld: warning: cannot find entry symbol _start; defaulting to 000::00XXXXX

marc-hb, 2019-06-17:
**Describe the bug**

As of recent commit eaf994a7fc34 and SDK 0.10.0,  sanitycheck produces two warnings (unrelated to each other) **in almost every test case**. One of them is:
 
```
  CMakeOutput.log: ARCH-zephyr-eabi/bin/ld: warning: cannot find entry symbol _start; defaulting to 000000000000XXXXX 
```
This almost systematic `_start`warning makes about half of the entire `grep -ri '[[:blank:]]warn'` noise that obscures any work-in-progress warning or platform specific warning in the logs (typically: device tree warnings).

Note `arc-zephyr-elf/bin/ld` behaves (again?) differently from all other `*-zephyr-elf.bin/ld` in the same 0.10.0 SDK: the missing `__start` it complains about has a double underscore, all other linkers in the same SDK complain about a single underscore `_start`. Curious why?

**To Reproduce**
```
sanitycheck
grep -A3 -ri -E '[[:blank:]]warn' sanity-out/
```
The other, unrelated warning has a tentative fix in PR #16879 

**Expected behavior**
A small number of test or platform-specific warnings that fits on one screen.
Unrelated PR #16879 + fixing or working around this one is enough to get there.

**Impact**
Discouraging to look at warnings in sanitycheck results. Proof: these two warnings seem to have been present for at least weeks, maybe months. Dunno how old are the unrelated device tree warnings.

**Additional context**
According to @andyross:
- This is a very architecture-specific area and may require a number of different fixes independent from each other.
- This warning could be fairly new, not present in older SDKs.

marc-hb, 2019-06-19:
Considering this is now the only warning appearing in most tests run by default by sanitycheck and considering the current emphasis on certification/compliance/etc. I would recommend raising the priority one notch above the lowest. Now priorities are of course relative and I only have "tunnel vision" of all Zephyr priorities so please correct me with some priority guidelines/rules of thumb; thanks in advance!

PS: maybe this doesn't need to be tagged as a "bug" but rather some kind of task in case that helps.

marc-hb, 2019-06-20:
> Considering this is now the only warning appearing in most tests run by default by sanitycheck...

For completeness: a small number (30-) of platforms sharing an apparently smaller number of configurations also show a few Device Tree and Kconfig warnings in their cmake output. All other platforms have only this missing `_start` warning in their entire sanitycheck output.

Here's a good test to produce most warnings in `sanity-out/**/*.log`

`${ZEPHYR_BASE}/scripts/sanitycheck -T ${ZEPHYR_BASE}/samples/synchronization/  --all`

marc-hb, 2019-09-18:
> 1. Cache the result of toolchain_is_ok? 

Done in commit 6212ec9 / PR #18741 )

SebastianBoe, 2019-09-19:
Shall we close this then Marc? I actually wanted #18743 to close this, but I guess #18741 is enough.

closed 2019-09-19 by galak

marc-hb, 2019-09-19:
Yes of course! sorry I totally missed it was still open.

PR #18743 (2019-09-19): cmake: Explicitly set the entry point when testing toolchain flags
We are observing warnings when test-compiling toolchain flags that
look like:

cannot find entry symbol _start; defaulting to 00000000000010d4

To fix this warning we explicitly set an entry point to an arbitrary
address. The files are only compiled, not run, so the entry point does
not need to be correct.

This fixes #16880