zephyrproject-rtos/zephyr #16886: Bluetooth Mesh: Receive segmented message multiple times

spieth, 2019-06-18:
**Describe the bug**
I have two nRF52840 DK that send periodically messages to a group address both nodes  subscribed to. When these messages are big enough for separation and are sent close in time (around less than 2 seconds), one node receives the same message multiple times. To identify the messages I transmit an ID.
I wasn't able to figure out what's going on when I looked at the debug logs. With more devices it gets even worse and the messages are received up to 5 times. 

**To Reproduce**
See above.

**Expected behavior**
Each message should be received only once.

**Screenshots or console output**
Two devices:
\`\`\`
6790: got message 0 from node 1
7030: sending 15 bytes test data
8000: got message 0 from node 1
36790: got message 1 from node 1
37040: sending 15 bytes test data
38000: got message 1 from node 1
66820: got message 2 from node 1
67050: sending 15 bytes test data
67330: got message 2 from node 1
\`\`\`

Three devices:
\`\`\`
637240: sending 15 bytes test data
637840: got message 21 from node 1
638100: got message 9 from node 2
638280: got message 21 from node 1
638610: got message 9 from node 2
639100: got message 21 from node 1
639420: got message 9 from node 2
639910: got message 21 from node 1
640230: got message 9 from node 2
640720: got message 21 from node 1
641040: got message 9 from node 2
667250: sending 15 bytes test data
667840: got message 22 from node 1
668130: got message 10 from node 2
668290: got message 22 from node 1
668610: got message 10 from node 2
669100: got message 22 from node 1
669420: got message 10 from node 2
669910: got message 22 from node 1
670230: got message 10 from node 2
670720: got message 22 from node 1
671040: got message 10 from node 2
\`\`\`

**Environment:**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: v1.14.0

jhedberg, 2019-06-18:
I'm not sure this is a bug (it could be, but further analysis is needed). Since mesh is an unreliable bearer there is built-in redundancy in many cases. Usually the various duplication filters below the access layer should filter out messages, but there are valid cases where filtering happens above the access layer as well - that's why some of the standard model messages have a transaction ID field.

spieth, 2019-06-18:
Thanks for your reply. Could these valid cases explain why this happens only to segmented messages? And how does it come a node receives a message up to 5 times when my retransmit counter is lower or even set to 0?

jhedberg, 2019-06-18:
@spieth are you using model publication to send the messages? That has two kinds of retransmit parameters: the network layer retransmit (perhaps that's what you mean when you said you set something to 0), and the publish retransmit. The publish retransmit sends completely "new" messages each time, meaning duplicates need to be identified above the access layer.

Is the source code of your application available somewhere, so I could take a look to see if I spot something obviously wrong?

jhedberg, 2019-06-18:
@spieth I went through the RX code path for segmented messages and found that it isn't completely in line with how the spec describes replay protection to be done for segmented messages. I created a pull request to try to fix this: #16890

I am not at all certain if this would help your issue, but could you test it and report back if you see any change in behaviour? Thanks.

spieth, 2019-06-18:
I don't use model publication but `bt_mesh_model_send`. As I understand model publication allows only  
 to send fixed size messages, right? Eventually I want to exchange neighborhood information between the nodes for optimal relay selection and therefore message size can change over time.

The source code can be found at https://github.com/spieth/zephyr-duplicate-messages

spieth, 2019-06-18:
@jhedberg Unfortunately the pull request didn't fix this.

jhedberg, 2019-06-18:
@spieth I still can't explain the behaviour you're seeing, but I'd like to make a couple of points:

 - model publication can use variable size messages, so this is an option if you'd want
 - Sending segmented messages to group addresses is *very* in efficient with mesh because the segment acknowledgement responses are never sent then. This means that the sender will keep "blindly" resending the segments until a timer expires, without knowledge of whether the segments were received or not. Because of this I really wouldn't recommend combining segmented messages with group addresses. That said, this still shouldn't cause message duplication, so there's possibly a bug in the transport layer code for trying to send a new segmented message when the timers for a previous one are still active.

carlescufi, 2019-08-15:
@spieth is this still an issue on the latest master?

spieth, 2019-08-17:
@carlescufi I'll test it next week and report back.

spieth, 2019-08-21:
@carlescufi This issue can still be reproduced on the latest master. I'll hope to find some time next week to investigate further as it looks like a bug to me.

carlescufi, 2019-08-22:
@spieth please let us know once you've had a chance to investigate more since we are currently not actively investigating this until we get a little bit more information.

carlescufi, 2019-11-14:
@spieth is this still an issue?

carlescufi, 2020-05-14:
Closing this. If this is still anissue with the latest master please reopen @spieth.

closed 2020-05-14 by carlescufi

MarcianoPreciado, 2020-08-28:
This is still an issue. Even when sending unsegmented messages with `bt_mesh_model_send` the receiving node gets **every** message twice.

MarcianoPreciado, 2020-08-28:
Shouldn't the network layer be able to filter these extras out using the IV and SeqNum?