zephyrproject-rtos/zephyr #16893: Bluetooth: Multiple local IDs, privacy problem

MarekPieta, 2019-06-18:
**Describe the bug**
For Bluetooth peripheral device with multiple local IDs enabling privacy (CONFIG_BT_PRIVACY=y, CONFIG_BT_CTLR_PRIVACY=y) causes problems with Bluetooth address. Without privacy every local ID uses own Bluetooth address. After turining privacy on, device advertises with the same address for every local ID, which may cause connection attemps from host.

**Expected behavior**
Every local ID should have own Bluetooth address.

**Impact**
Cannot use multiple local IDs together with privacy.

**Commit**
Reproduced for commit: 7e64ac737daead12d2dbdda5daec04188243a81c

jhedberg, 2019-06-18:
Is this because the BT_DEV_RPA_VALID flag isn't cleared? Does this help:
```
diff --git a/subsys/bluetooth/host/hci_core.c b/subsys/bluetooth/host/hci_core.c
index c2209018cc..bbf380b039 100644
--- a/subsys/bluetooth/host/hci_core.c
+++ b/subsys/bluetooth/host/hci_core.c
@@ -5348,6 +5348,10 @@ int bt_le_adv_start_internal(const struct bt_le_adv_param *param,
        set_param.max_interval = sys_cpu_to_le16(param->interval_max);
        set_param.channel_map  = 0x07;
 
+       if (bt_dev.adv_id != param->id) {
+               atomic_clear_bit(bt_dev.flags, BT_DEV_RPA_VALID);
+       }
+
        /* Set which local identity address we're advertising with */
        bt_dev.adv_id = param->id;
        id_addr = &bt_dev.id_addr[param->id];
```

MarekPieta, 2019-06-18:
@jhedberg, seems like this change fixed issue. Thanks a lot.

PR #16895 (2019-06-19): Bluetooth: Fix expiring RPA when different local identities are used
When advertising with different identities we need to flag any
programmed RPA as invalid if it was generated using a different
identity.

Fixes #16893
commit b02157ef (2019-06-19): Bluetooth: Fix expiring RPA when different local identities are used

When advertising with different identities we need to flag any
programmed RPA as invalid if it was generated using a different
identity.

Fixes #16893
commit 04c54f47 (2019-06-19): Bluetooth: Fix expiring RPA when different local identities are used

When advertising with different identities we need to flag any
programmed RPA as invalid if it was generated using a different
identity.

Fixes #16893
commit 4e89d294 (2019-06-19): Bluetooth: Fix expiring RPA when different local identities are used

When advertising with different identities we need to flag any
programmed RPA as invalid if it was generated using a different
identity.

Fixes #16893
closed 2019-06-19 by jhedberg