zephyrproject-rtos/zephyr #16894: ARM: alignment problems in libc/newlib

ioannisg, 2019-06-18:
The following Kconfig option indicates that CONFIG_LIBC_ALIGNED_HEAP_SIZE only gets a non-zero value for ARMv7-M builds with MPU: Builds with ARMv8-M and NXP_MPU do not have the requirement for power-of-two & size alignment.

```
config NEWLIB_LIBC_ALIGNED_HEAP_SIZE
	int "Newlib aligned heap size"
	depends on MPU_REQUIRES_POWER_OF_TWO_ALIGNMENT
	depends on NEWLIB_LIBC
	depends on USERSPACE
	default 0

```

However, when building with CONFIG_USERSPACE, we still enclose the heap in an user-accessible memory partition:
```

static int malloc_prepare(struct device *unused)
{
	ARG_UNUSED(unused);

#if CONFIG_NEWLIB_LIBC_ALIGNED_HEAP_SIZE
	z_malloc_partition.start = (u32_t)heap_base;
#else
	z_malloc_partition.start = USED_RAM_END_ADDR;
#endif
	z_malloc_partition.size = MAX_HEAP_SIZE;
	z_malloc_partition.attr = K_MEM_PARTITION_P_RW_U_RW;
	return 0;
}
```
In the above snippet, `USED_RAM_END_ADDR `is taken directly from .end address in the linker, which might not be aligned with the minimum MPU granularity for ARM (32 bytes), triggering an ASSERT when attempting to program the region.

ioannisg, 2019-06-18:
@andrewboie FYI

andrewboie, 2019-06-18:
can we just align the _end symbol in the linker script? seems like a trivial fix

ioannisg, 2019-06-18:
> 
> 
> can we just align the _end symbol in the linker script? seems like a trivial fix

I think I would promote a more clean solution @andrewboie. I mean, in the libc hooks, we are _still_ creating a mem partition with nPRIV rights outside the #ifdef NEWLIB_LIBC_ALIGNED_HEAP_SIZE, which obviously needs MPU. So IMHO it's the responsibility of the modulte to make sure it uses the right address for the heap. It's not the linker script that is responsible for that...

andrewboie, 2019-06-18:
You can also round up `z_malloc_partition.start = USED_RAM_END_ADDR;` to whatever it needs to be.

ioannisg, 2019-06-18:
> You can also round up `z_malloc_partition.start = USED_RAM_END_ADDR;` to whatever it needs to be.

Yes @andrewboie we could do that. We also need to round up the _end inside the libc cause it is used directly as is.

PR #16949 (2019-06-26): Libc hooks fix alignment issues
Fixes #16894
In detail, the PR fixes the alignment issue for the heap, when building with user mode support in platforms with the NXP or the ARMv8-M MPU architecture.
closed 2019-06-26 by nashif