zephyrproject-rtos/zephyr #16898: bluetooth stack change affects timer behavior

pabigot, 2019-06-18:
Commit 8a92818c0a0ceea30a9540e3e1557dec2cdf6fbf broke my bluetooth mesh application, in that a work item submitted to the system work queue is never executed.

Prior to the commit the backtrace when idle was:
```
#0  k_cpu_idle () at /mnt/devel/external/zp/zephyr/arch/arm/core/cpu_idle.S:139
#1  0x0001bf94 in sys_power_save_idle () at /mnt/devel/external/zp/zephyr/kernel/idle.c:100
#2  idle (unused1=<optimized out>, unused2=<optimized out>, unused3=<optimized out>)
    at /mnt/devel/external/zp/zephyr/kernel/idle.c:155
#3  0x0001d5a2 in z_thread_entry (entry=0x30 <z_get_fd_obj+48>, p1=<optimized out>, p2=<optimized out>,
    p3=<optimized out>) at /mnt/devel/external/zp/zephyr/lib/os/thread_entry.c:29
#4  0x0001d5a2 in z_thread_entry (entry=0x30 <z_get_fd_obj+48>, p1=<optimized out>, p2=<optimized out>,
    p3=<optimized out>) at /mnt/devel/external/zp/zephyr/lib/os/thread_entry.c:29
```

At the commit the idle backtrace is:

```
#0  cpu_sleep () at /mnt/devel/external/zp/zephyr/subsys/bluetooth/controller/ll_sw/nordic/hal/cpu.h:11
#1  0x000184ca in role_disable (ticker_id_primary=ticker_id_primary@entry=5 '\005', ticker_id_stop=ticker_id_stop@entry=3 '\003') at /mnt/devel/external/zp/zephyr/subsys/bluetooth/controller/ll_sw/ctrl.c:10566
#2  0x0001a052 in radio_adv_disable () at /mnt/devel/external/zp/zephyr/subsys/bluetooth/controller/ll_sw/ctrl.c:10997
#3  0x0001a4d2 in ll_adv_enable (enable=<optimized out>) at /mnt/devel/external/zp/zephyr/subsys/bluetooth/controller/ll_sw/ll_adv.c:404
#4  0x00017990 in le_set_adv_enable (buf=0x2000523c <net_buf_hci_cmd_pool+28>, evt=0x20002724 <tx_thread_stack+516>) at /mnt/devel/external/zp/zephyr/subsys/bluetooth/controller/hci/hci.c:1650
#5  controller_cmd_handle (node_rx=0x2000273c <tx_thread_stack+540>, evt=0x20002724 <tx_thread_stack+516>, cmd=0x2000523c <net_buf_hci_cmd_pool+28>, ocf=10) at /mnt/devel/external/zp/zephyr/subsys/bluetooth/controller/hci/hci.c:1650
#6  hci_cmd_handle (cmd=0x2000523c <net_buf_hci_cmd_pool+28>, node_rx=node_rx@entry=0x20002754 <tx_thread_stack+564>) at /mnt/devel/external/zp/zephyr/subsys/bluetooth/controller/hci/hci.c:2241
#7  0x00017094 in cmd_handle (buf=0x2000523c <net_buf_hci_cmd_pool+28>) at /mnt/devel/external/zp/zephyr/subsys/bluetooth/controller/hci/hci_driver.c:384
#8  hci_driver_send (buf=0x2000523c <net_buf_hci_cmd_pool+28>) at /mnt/devel/external/zp/zephyr/subsys/bluetooth/controller/hci/hci_driver.c:434
#9  0x00007a98 in bt_send (buf=0x2000523c <net_buf_hci_cmd_pool+28>) at /mnt/devel/external/zp/zephyr/subsys/bluetooth/host/hci_core.c:4561
#10 send_cmd () at /mnt/devel/external/zp/zephyr/subsys/bluetooth/host/hci_core.c:3565
#11 process_events (ev=0x20005cac <events>, count=2) at /mnt/devel/external/zp/zephyr/subsys/bluetooth/host/hci_core.c:3589
#12 hci_tx_thread (p1=<optimized out>, p2=<optimized out>, p3=<optimized out>) at /mnt/devel/external/zp/zephyr/subsys/bluetooth/host/hci_core.c:3649
#13 0x0001d452 in z_thread_entry (entry=0x79c9 <hci_tx_thread>, p1=<optimized out>, p2=<optimized out>, p3=<optimized out>) at /mnt/devel/external/zp/zephyr/lib/os/thread_entry.c:29
#14 0x0001d452 in z_thread_entry (entry=0x79c9 <hci_tx_thread>, p1=<optimized out>, p2=<optimized out>, p3=<optimized out>) at /mnt/devel/external/zp/zephyr/lib/os/thread_entry.c:29
```
In short, it's waiting here:
```
		if (ret == TICKER_STATUS_BUSY) {
			/* wait for ticker to be stopped */
			while (ret_cb == TICKER_STATUS_BUSY) {
				cpu_sleep();
			}
		}
```
but whatever's supposed to change the ticker state doesn't happen.

Commenting out the sleep has no effect.

Changing the loop body to invoke `ticker_job_sched` like similar loops also has no effect.

pabigot, 2019-06-18:
The behavior is non-deterministic, i.e. the application may run for a few seconds before the conditions that introduce the failure occur.

Reverting 8a92818c0a0ceea30a9540e3e1557dec2cdf6fbf works.  The cause of the failure may be related to something like the diagnostic below (before reverting I'd just see nothing after the first or second `periodic`.
```
0:00:12.078: periodic
[00:00:11.887,634] <wrn> bt_mesh_net: Duplicate found in Network Message Cache
0:00:17.078: periodic
```

aescolar, 2019-06-18:
That commit has now been reverted from master.
Therefore I guess this issue can be closed?

closed 2019-06-18 by pabigot