zephyrproject-rtos/zephyr #16899: fs/nvs: might loop-up if storage was not erased before first run

nvlsianpu, 2019-06-18:
**bug**
If storage area content was not clear before NVS first run, that it is possible that in one of subsequent startup nvs will lock-up in infinite loop in GC procedure.

execution was looping infinitely here:
https://github.com/zephyrproject-rtos/zephyr/blob/54000fb886ea5ab21404f927666d60a9a0a238b4/subsys/fs/nvs/nvs.c#L476-L491

If storage was cleared before first run this issue will never occur.

**To Reproduce**
Was reported once by 3rd person - NVS area was occupied by a fcb storage before.

**Expected behavior**
NVS should be able to recoginize that it is looping in searching for valid data, and return an error.

**Impact**
This error issue shouldn't happen if device has been totally erased before been flashed. As `west flash` erases only programmed area, storage area might contains a garbage.

nvlsianpu, 2019-06-18:
cc @kapi-no

Laczen, 2019-06-19:
@nvlsianpu, can you somehow reproduce this ?

I have been going through the code how this could happen and might have spotted a reason for this:
a. nvs_prev_ate updates the address by going up in the ate range one ate at a time (until a sector end is reached when it jumps to the previous sector). When it jumps to the previous sector it will read the offset from the sector close ate or try to find it by stepping down one ate at a time. 
b. as long as it does steps by one ate nothing can go wrong.
c. where it can go wrong is when the ate address is updated from the sector close ate: if for whatever reason the sector close ate has an ok crc8 it is considered valid and the ate addr is updated from this. This might lead to a situation where the ate_addr is pointing outside the sector and the loop in gc would get back to the same sector close ate, thereby creating a loop.

To avoid this kind of behavior an extra check can be added to the update of the address based upon the sector close ate (nvs.c line 360):
```
	if (!nvs_ate_crc8_check(&close_ate)) {
		/* update the address so it points to the last added ate.
		 * do a check on close_ate.offset so that it does not point
		 * outside a sector.
		 */
		if (close_ate.offset < (fs->sector_size - ate_size)) {
			(*addr) &= ADDR_SECT_MASK;
			(*addr) += close_ate.offset;
			return 0;
		}
	}
```

nvlsianpu, 2019-08-07:
@Laczen Why the second code snapshot gives better protection?

nvlsianpu, 2019-08-07:
I commit your comments below the snippets - yes right.

Laczen, 2019-08-07:
@nvlsianpu, because it also makes sure that the alignment will remain correct

PR #18086 (2019-08-07): fs/nvs: preserve the previous ate searching
Previous ATE searching is accelerate by reading sector
close ate, which allow to skip reading whole storage.

ATE is already covered by crc8. That patch introduce
check on correctness offset read form close ate. This
increases meta-data integrity check level.
This also preserves against possible looping when read
incorrect or not a NVS storage image.

fixes #16899
commit f42617ad (2019-08-07): fs/nvs: preserve the previous ate searching

Previous ATE searching is accelerate by reading sector
close ate, which allow to skip reading whole storage.

ATE is already covered by crc8. That patch introduce
check on correctness offset read form close ate. This
increases meta-data integrity check level.
This also preserves against possible looping when read
incorrect or not a NVS storage image.

fixes #16899
PR #18114 (2019-08-15): fs/nvs: preserve the previous ate searching [Backport]
Previous ATE searching is accelerate by reading sector
close ate, which allow to skip reading whole storage.

ATE is already covered by crc8. That patch introduce
check on correctness offset read form close ate. This
increases meta-data integrity check level.
This also preserves against possible looping when read
incorrect or not a NVS storage image.

fixes #16899
commit baae719e (2019-08-15): fs/nvs: preserve the previous ate searching

Previous ATE searching is accelerate by reading sector
close ate, which allow to skip reading whole storage.

ATE is already covered by crc8. That patch introduce
check on correctness offset read form close ate. This
increases meta-data integrity check level.
This also preserves against possible looping when read
incorrect or not a NVS storage image.

fixes #16899
closed 2019-08-07 by carlescufi