zephyrproject-rtos/zephyr #16907: native_posix build fails with X86_64 on macOS

jhedberg, 2019-06-18:
I'm trying to make native posix build on macOS. It seems 32-bit support has been discontinued from the macOS compiler, and indeed the build fails almost immediately with that. I get much further when enabling `CONFIG_X86_64=y` for `samples/hello_world`, but that fails eventually as well, due to what looks like some sort of dependency issue (possibly circular) with various headers:
```
[8/84] Building C object zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj
FAILED: zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj 
ccache /usr/bin/gcc -DBUILD_VERSION=zephyr-v1.14.0-1601-gfaa398f078c4 -DKERNEL -D_FORTIFY_SOURCE=2 -D_POSIX_C_SOURCE=200809 -D_XOPEN_SOURCE=600 -D_XOPEN_SOURCE_EXTENDED -D__ZEPHYR__=1 -I../kernel/include -I../arch/posix/include -I../include -I../include/drivers -Izephyr/include/generated -I../soc/posix/inf_clock -I../boards/posix/native_posix -I../lib/libc/minimal/include -isystem /Library/Developer/CommandLineTools/usr/lib/clang/11.0.0/include -Os -nostdinc -imacros /Users/jhedberg/src/zephyr/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wno-pointer-sign -Wpointer-arith -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -ffunction-sections -fdata-sections -m32 -MMD -MP -include /Users/jhedberg/src/zephyr/zephyr/arch/posix/include/posix_cheats.h -std=c11 -MD -MT zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj   -c /Users/jhedberg/src/zephyr/zephyr/arch/posix/core/offsets/offsets.c
In file included from /Users/jhedberg/src/zephyr/zephyr/arch/posix/core/offsets/offsets.c:30:
In file included from ../kernel/include/kernel_structs.h:10:
In file included from ../include/kernel.h:17:
In file included from ../include/kernel_includes.h:34:
In file included from ../include/arch/cpu.h:15:
In file included from ../include/arch/x86_64/arch.h:9:
In file included from ../arch/posix/include/kernel_arch_func.h:15:
../arch/posix/include/posix_core.h:17:2: error: unknown type name 'k_thread_entry_t'
        k_thread_entry_t entry_point;
        ^
In file included from /Users/jhedberg/src/zephyr/zephyr/arch/posix/core/offsets/offsets.c:30:
In file included from ../kernel/include/kernel_structs.h:10:
In file included from ../include/kernel.h:17:
In file included from ../include/kernel_includes.h:34:
In file included from ../include/arch/cpu.h:15:
In file included from ../include/arch/x86_64/arch.h:9:
../arch/posix/include/kernel_arch_func.h:49:8: error: incomplete definition of type 'struct k_thread'
        thread->callee_saved.retval = value;
        ~~~~~~^
../include/sched_priq.h:32:8: note: forward declaration of 'struct k_thread'
struct k_thread;
       ^
In file included from /Users/jhedberg/src/zephyr/zephyr/arch/posix/core/offsets/offsets.c:30:
In file included from ../kernel/include/kernel_structs.h:10:
In file included from ../include/kernel.h:17:
In file included from ../include/kernel_includes.h:37:
../include/spinlock.h:75:10: warning: implicit declaration of function 'z_arch_irq_lock' is invalid in C99 [-Wimplicit-function-declaration]
        k.key = z_arch_irq_lock();
                ^
../include/spinlock.h:110:2: warning: implicit declaration of function 'z_arch_irq_unlock' is invalid in C99 [-Wimplicit-function-declaration]
        z_arch_irq_unlock(key.key);
        ^
2 warnings and 2 errors generated.
ninja: build stopped: subcommand failed.
ERROR: command exited with status 1: /usr/local/bin/cmake --build /Users/jhedberg/src/zephyr/zephyr/build
```

I.e. the problem seems to be that `include/kernel.h` very early includes a chain of headers which ends up requiring symbols that are only defined later inside kernel.h.

jhedberg, 2019-06-18:
Forgot to provide steps to reproduce:

1. apply following diff:
```
diff --git a/samples/hello_world/prj.conf b/samples/hello_world/prj.conf
index b2a4ba5910..ef4912518b 100644
--- a/samples/hello_world/prj.conf
+++ b/samples/hello_world/prj.conf
@@ -1 +1,2 @@
 # nothing here
+CONFIG_X86_64=y
```

2. run `west build -b native_posix samples/hello_world`

jhedberg, 2019-06-18:
Also, the obvious question: am I trying to do something that cannot be made to work? I just realised the the X86_64 Kconfig is an arch selection, just like ARCH_POSIX, so is it invalid to try to enable X86_64 when trying to build for the native_posix board? Should new 64-bit variants be created for the posix board and arch definitions?

jhedberg, 2019-06-18:
@aescolar @andyross any comments on this?

andrewboie, 2019-06-18:
I don't think this has ever worked, it requires a 32-bit host toolchain and libs. The kernel is not 64-bit clean yet, although there have been contributions recently to help fix this.

CONFIG_X86_64 isn't the way to go about it, that is an actual arch target and not a POSIX shim.

marc-hb, 2019-06-18:
Just mentioned on Slack: https://www.phoronix.com/scan.php?page=news_item&px=Ubuntu-Drops-32-bit-x86

marc-hb, 2019-06-19:
> It seems 32-bit support has been discontinued from the macOS toolchain,

Your error log has an -include reference to Xcode, can you confirm Xcode/clang is what you were trying to use?

https://docs.zephyrproject.org/latest/getting_started/ doesn't seem to mention any particular macOS toolchain.

Support for 32 bits applications has been removed in Xcode 10 (late 2018)
https://en.wikipedia.org/wiki/Xcode#9.x_series
https://developer.apple.com/documentation/xcode_release_notes/xcode_10_release_notes
In macOS 10.15 Catilina existing 32bits applications will stop working.

Did you ever get native_posix (or anything really...) working on macOS and approximately when/with what toolchain version(s)?

BTW: https://crosstool-ng.github.io/2018/11/26/macos.html but @nashif said it may still be doable.

jhedberg, 2019-06-19:
> Your error log has an -include reference to Xcode, can you confirm Xcode/clang is what you were trying to use?

Yes, that's correct.

> Did you ever get native_posix (or anything really...) working on macOS and approximately when/with what toolchain version(s)?

No. I tried it sometime last year without much luck. Having seen some clang and 64-bit patches going in lately I thought I'd try it again.

aescolar, 2019-06-19:
As @andyross commented, the issue today is:
* The kernel (and in 1 place the posix arch), and apps, not being 64 bit clean
* There isn't yet, a POSIX_ARCH_x64 or similar option, which would then set the compiler options to target it (today it explicitly targets x86 with `-m32`)
Note that there is some ongoing work to have these.
(Disclaimer: You may face other issues when trying to compile it or run it in macOS)

Anyhow, as what you described above is technically incorrect (selecting the wrong arch for a board), do you want to close this, and then add another feature request like "posix arch support on 64 bit hosts" or so? (and then maybe another "native_posix is not supported in macOS")

jhedberg, 2019-06-19:
> Anyhow, as what you described above is technically incorrect (selecting the wrong arch for a board), do you want to close this, and then add another feature request like "posix arch support on 64 bit hosts" or so? (and then maybe another "native_posix is not supported in macOS")

@aescolar I was kind of expecting this to morph into a feature request (e.g. I already updated the labels) but if you think it's cleaner to start off with a new issue that's fine as well.

aescolar, 2019-06-19:
@jhedberg I think it would be cleaner, yep.

jhedberg, 2019-06-19:
Closing as per discussion - will open new issues to track the specific feature requests

closed 2019-06-19 by jhedberg

jhedberg, 2019-06-24:
@aescolar curious how #17018 relates to this - is it really that simple to do a 64-bit posix build?

olofj, 2019-06-24:
BOARD=native_posix_64  and #17018 produces a valid build for me on Linux:

```
$ export ZEPHYR_TOOLCHAIN_VARIANT=host
$ . ~/z/zephyr/zephyr-env.sh
$ cd samples/subsys/shell/shell_module
$ mkdir b
$ cd b
$ cmake -DBOARD=native_posix_64 -GNinja ..
$ ninja
$ file zephyr/zephyr.exe
zephyr/zephyr.elf: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked (uses shared libs), for GNU/Linux 2.6.32, not stripped
```

jhedberg, 2019-06-24:
> BOARD=native_posix_64

@olofj where do you get that board definition from? It doesn't seem to exist upstream at least.

olofj, 2019-06-24:
Oh, my bad -- I was working on top of some WIP patches from @npitre that haven't yet been upstreamed.

aescolar, 2019-06-24:
@jhedberg Yes. That `-m32` is to explicitly tell the compiler to build for x86, otherwise it targets the host platform.
But what you need is to have the different 64bit issues solved around in Zephyr. Once @npitre solves them, you should be able to compile safely for 64bits (if your application is also 64bit clean). I thought he was planning to add a config option to native_posix to select so. But from olofj comment maybe he plans to have a 64 bit variant of the board all together instead.

marc-hb, 2019-06-24:
> That -m32 is to explicitly tell the compiler to build for x86, otherwise it targets the host platform.

Sorry for being pedantic but as usual in this area there's a high potential for confusion, so better safe than sorry: even with -m32 the compiler still targets "the host platform". Both native_posix (-m32) and native_posix_64 run on the host. The exact same host does both 32 and 64 bits simultaneously.

BTW https://www.phoronix.com/scan.php?page=news_item&px=Ubuntu-Partial-32-Bit-19.10

> (Disclaimer: You may face other issues when trying to compile it or run it in macOS)

You bet.