zephyrproject-rtos/zephyr #16926: NXP LPC54102（LPC54114）: question about dual core(M4 and M0) running on flash

Jeepgoing, 2019-06-19:
Hi, @MaureenHelm, I met a thorny problem, my board based on lpc54102 soc, applications distributed on M4 and M0, both M4 and M0 are running zephyr on flash, when i only enabled M0 timer0 with 600us,  the timer range is from 602us to 610, but when i enabled M4 application (used timer3 and spi0), ther range of M0 timer0 become large from 575us to 630us. 

allocation of resources is as follows:

> processor | peripheral
> M4 | TIMER3
> M4 | SPI0
> M4 | USART
> M4 | MAILBOX
> M0 | TIMER0
> M0 | TIMER1
> M0 | TIMER2
> M0 | SPI1
> M0 | ADC0
> M0 | MAILBOX

i have checked LPC5410xDataSheet, 
[image]
 
it's that mean i can't run both M4 and M0 on flash? why zephyr can't support LPC54102 soc, is there this problem on LPC54114?

agansari, 2019-06-19:
@Jeepgoing current zephyr and mcuxpresso implementation on LPC54114 uses M4 with flash memory and M0 uses one of the SRAMs for program memory. M0's code is copied from flash to the SRAM. We assume M4 is master and M0 is slave.

See the following files:
zephyr/boards/arm/lpcxpresso54114/lpcxpresso54114_m4.dts
`	chosen {
		zephyr,sram = &sram0;
		zephyr,flash = &flash0;
		zephyr,console = &usart0;
		zephyr,shell-uart = &usart0;
	};`
zephyr/boards/arm/lpcxpresso54114/lpcxpresso54114_m0.dts
`	chosen {
		zephyr,sram = &sram2;
		zephyr,flash = &sram1;
		/*zephyr,console = &usart0; uncomment to use console on M0  */
		/*zephyr,shell-uart = &usart0; uncomment to use shell on M0 */
	};`

Look for function _slave_init (zephyr/soc/arm/nxp_lpc/lpc54xxx/soc.c) on how secondary core (M0) is booted.
`	/* Copy second core image to SRAM */
	memcpy(CORE_M0_BOOT_ADDRESS, (void *)core_m0, sizeof(core_m0));`

In mcuxpresso you can look for functions ResetISR(void) and ResetISR2(void) on how cores are booted.

Jeepgoing, 2019-06-19:
@agansari  thank you. i know lpc54114 m0 run on ram, my platform is lpc54102, it's ram not enough, so i run both m4 and m0 on flash.

Jeepgoing, 2019-06-20:
Hi, @agansari @MaureenHelm i am trying run some codes on ram through section `.ramfunc`

agansari, 2019-06-20:
@Jeepgoing 
1. You need to make sure your project has config ARCH_HAS_RAMFUNC_SUPPORT enabled. This is needed to link ramfunc specific linker configurations.
2. Put __ramfunc in front of the function, e.g. __ramfunc void func(void), it's a macro that will place the function in the ramfunc section.

Jeepgoing, 2019-06-20:
@agansari

there are too many functions should be moved to SRAM, so how can i move .c file to  .ramfunc？

any ideas?

agansari, 2019-06-20:
@Jeepgoing you still need to enable ARCH_HAS_RAMFUNC_SUPPORT.

You would need to change a linker script to add your file's text section to a certain partition. This can be done as either: 
 - edit zephyr/soc/arm/nxp_lpc/lpc54xxx/linker.ld
 - or look at the following sample: zephyr\samples\application_development\code_relocation how it uses custom a linker file and how it's included in the project

What to add to the linker file? I think the information here is good.
https://mcuoneclipse.com/2014/10/06/putting-code-of-files-into-special-section-with-the-gnu-linker/

>  .generatedCode : 
>   { 
>     . = ALIGN(4);
>     *Generated_Code/*.o (.text .text*)  <<<<< use file.o (object)
>     . = ALIGN(4); 
>   }  > m_text2 /* new section for my special code files */

Jeepgoing, 2019-06-20:
@agansari  thank you, i will try.

Jeepgoing, 2019-07-03:
either LPC54114 or LPC54102 can't run m4 and m0 core both on flash.

closed 2019-07-03 by Jeepgoing