zephyrproject-rtos/zephyr #16934: drivers: flash: stm32l4: Erase wait time is not enough

gon1332, 2019-06-19:
**Describe the bug**
`flash_erase` cannot erase a page from the STM32L4 SOC flash. I've seen it, after investigating why MCUboot would never perform swapping and always resulted in *panic*.

I concluded that the erase time of a 2K page is not enough. STM32L496xx datasheet states in *6.3.10 Flash memory characteristics*, that the maximum erase time is 24.47ms. It is also noted that these characteristics are *guaranteed by design*. I tried wait times for `flash_stm32_wait_flash_idle` from 25ms (initial) up to 30ms. The 30ms one worked all the times I tested. It seems that the erase time is a little bit off.

I've also studied the code of [texane/stlink](https://github.com/texane/stlink/blob/d040db5585d33dc36bd8ae093c024583ef1b4f74/src/common.c#L1629) just to get an idea of how they erase a page. They just [poll on the `SR_BUSY` bit](https://github.com/texane/stlink/blob/d040db5585d33dc36bd8ae093c024583ef1b4f74/src/common.c#L597). So it works.

For a quick test I created a sample that replicates the issue on Nucleo L496ZG:

**main.c**:
```c
/*
 * Copyright (c) 2019 Centaur Analytics, Inc
 */

#include <zephyr.h>
#include <misc/printk.h>
#include <flash.h>

#define FLASH_TEST_OFFSET 0x080000

void main(void)
{
    struct device *flash_dev;
    int rc;

    flash_dev = device_get_binding(DT_FLASH_DEV_NAME);
    if (!flash_dev) {
        printk("Cannot bind to %s\n", DT_FLASH_DEV_NAME);
        goto fail_exit;
    }

    flash_write_protection_set(flash_dev, false);

    printk("Trying to erase one page at offset 0x%04x\n", FLASH_TEST_OFFSET);
    rc = flash_erase(flash_dev, FLASH_TEST_OFFSET, FLASH_PAGE_SIZE);
    if (rc == -EINVAL) {
        printk("Invalid range\n");
        goto fail_exit;
    } else if (rc == -EIO) {
        printk("IO error\n");
        goto fail_exit;
    } else if (rc < 0) {
        printk("Could not erase area\n");
        goto fail_exit;
    }

    printk("Success\n");

fail_exit:
    while (1) {
        k_sleep(1000);
    }
}
```

**prj.conf**:
```
CONFIG_SIZE_OPTIMIZATIONS=y
CONFIG_FLASH=y
CONFIG_FLASH_HAS_PAGE_LAYOUT=y
```

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nucleo\_l496zg -s samples/flash\_erase\_page
2. st-flash write build/zephyr/zephyr.bin 0x8000000
4. See error

**Expected behavior**
The page should have been erased in the time stated by the Datasheet.

**Impact**
no comment

**Screenshots or console output**
One can see the outcome if modifies the `STM32_FLASH_TIMEOUT` from 25ms to 30ms for `CONFIG_SOC_SERIES_STM32L4X`:

**@25ms**:
```
Trying to erase one page at offset 0x80000
IO Error
```

**@30ms**:
```
Trying to erase one page at offset 0x80000
Success
```

**Environment:**
 - OS: GNU/Linux
 - Toolchain: Zephyr SDK
 - Zephyr tags/v1.14.0

**Additional context**
> I know that, if I'm not missing something, this is not a Zephyr problem, but a ST one. I'm reporting here to be sure.

If the workaround seems OK, I could open a PR and also add a test sample.

gon1332, 2019-06-19:
@erwango you seem the most ST person.
@galazari @nikooiko

gon1332, 2019-06-20:
Moreover, I've tested on STM32L496ZG and STM32L496RE SOCs, and on frequencies of 2MHz (MSI) and 80MHz (PLL).

erwango, 2019-06-20:
@gon1332 

I can observe the same issue. Strangely, this doesn't happen at a lower offset (tested at 0x1e000).
Also reproduced using `samples/drivers/flash_shell`.
Also reproduced using nucleo_l476rg.

I'll check internally to get more inputs on that and get back to you.

One question:
> `st-flash write build/zephyr/zephyr.bin 0x8000000`

Out of curiosity, why not using west flash?

gon1332, 2019-06-20:
> Out of curiosity, why not using west flash?
I've been using st-flash quite some time, before west. Now with west, I make more use of west flash(openocd). I'm currently on the transition, but I like the fact that I can explicitly define the write addresses, eg. for cases like the above.

psidhu, 2019-07-03:
At least according to the STM32L4+ reference manual in 3.3.6:

```
Page erase
To erase a page, follow the procedure below:
1. Check that no Flash memory operation is ongoing by checking the BSY bit in the Flash status register (FLASH_SR).
2. Check and clear all error programming flags due to a previous programming. If not, PGSERR is set.
3. In Dual-bank mode (DBANK option bit is set), set the PER bit and select the page to erase (PNB) with the associated bank (BKER) in the Flash control register (FLASH_CR). In Single-bank mode (DBANK option bit is reset), set the PER bit and
select the page to erase (PNB). The BKER bit in the Flash control register (FLASH_CR) must be kept cleared.
4. Set the STRT bit in the FLASH_CR register.
5. Wait for the BSY bit to be cleared in the FLASH_SR register.
```

The same is said about mass erase operations.

At least for the STM32L4+, it appears we should poll on the busy bit only and not introduce a timeout of our own?

gon1332, 2019-07-04:
@psidhu That's true also for STM32L4x{5,6}. I thought that the wait time used in the driver in the first place is just a safeguard to avoid eternal (?) blocking.

erwango, 2019-07-04:
@gon1332 , @psidhu 

> I thought that the wait time used in the driver in the first place is just a safeguard to avoid eternal (?) blocking.

This is right. We're already polling the SR_BSY bit in `flash_stm32_wait_flash_idle`, so extending STM32_FLASH_TIMEOUT from the "maximum erase time" to a real time out value would be the only thing to do here (let's try to double it first).

gon1332, 2019-07-04:
@erwango So be it. Do you want me to do it?

erwango, 2019-07-04:
@gon1332, if you don't mind, it would be welcomed

closed 2019-07-05 by ioannisg