zephyrproject-rtos/zephyr #16935: Zephyr doc website: Delay search in /boards to the end of the search.

erwango, 2019-06-19:
**Is your enhancement proposal related to a problem? Please describe.**
When using search functionality in Zephyr Project Documentation, board/ directory hits are reported first. Though in general, this is not in board files that we can get insightful information on Zephyr.
Besides, some generic terms (west, i2c, ...) hit in all boards so we need to go after >100 of hits before getting the result we're looking for.

**Describe the solution you'd like**
Delay search in /boards to the end of the search.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

dbkinder, 2019-06-19:
Thanks for point this out.

Alas, we don't have this level of control over the Sphinx-generated search results via any settings, but we might be able to introduce a tweak to accomplish pushing the boards later in the search results (the search results seem to be sorted by filename path where the hit is found).  I'll look into this.

We once looked at replacing the site search with a customized google search, so searching for "i2c" would show something more like this: https://www.google.com/search?q=site%3Azephyrproject.org+i2c  
The problem with this though is there's a delay for google to index the site (whereas the current search system is updated daily).

erwango, 2019-06-20:
@dbkinder,

> Alas, we don't have this level of control over the Sphinx-generated search results via any settings, 

Yeah that's what I was supposing. Thought, I think the problem is worth investigating, as it is a real drawback of the current doc system. You usually don't know where to find things, so an efficient search functionality is really a plus.
 
> but we might be able to introduce a tweak to accomplish pushing the boards later in the search results (the search results seem to be sorted by filename path where the hit is found). I'll look into this.

Would be great, thanks for checking this!

erwango, 2019-10-18:
@dbkinder feel free to close.

dbkinder, 2019-12-11:
After further review, the call on the field is reversed.  There is indeed a way to tweak the search scoring function to do what we want, and I've submitted a PR #21312 to do just that.

PR #21312 (2019-12-19): doc: tweak search to to understate certain docs
All docs are treated equally in the search results. The built-in search
system knows to emphasize hits in titles and object names, but could use
some help understating hits in non-definitive docs.  In particular, hits
to docs in the  /boards, /samples, and /reference/kconfig docs are often
not as important as hits in other docs, so let's push them later in the
search result output.

We can tweak the search results scoring (and thereby the order of
display) via the ``html_search_scorer`` setting in ``conf.py`` along
with a piece of JavaScript to a [... cut]
commit efffdfc4 (2019-12-19): doc: tweak search to to understate certain docs

All docs are treated equally in the search results. The built-in search
system knows to emphasize hits in titles and object names, but could use
some help understating hits in non-definitive docs.  In particular, hits
to docs in the  /boards, /samples, and /reference/kconfig docs are often
not as important as hits in other docs, so let's push them later in the
search result output.

We can tweak the search results scoring (and thereby the order of
display) via the ``html_search_scorer`` setting in ``conf.py`` along
with a piece of JavaScript to  [... cut]
closed 2019-12-19 by nashif