zephyrproject-rtos/zephyr #16943: Missing test coverage for lib/os/crc*.c

andrewboie, 2019-06-19:
We have a bunch of C files in lib/os for computing CRC values. These are not being tested. LCOV shows 42.9% line coverage for crc16_sw.c, and zero coverage for crc32_sw.c, crc7_sw.c, crc8_sw.c. Need to get these up to 100% coverage including branches.

aescolar, 2019-06-20:
@andrewboie Did you account for its unit test? the codecov report seems to indicate they are quite well covered:
https://codecov.io/gh/zephyrproject-rtos/zephyr/tree/master/lib/os

andrewboie, 2019-06-20:
@aescolar we are not getting any of this stuff when we run coverage reports using gcovr/lcov.
Someone needs to fix that.

aescolar, 2019-06-20:
> @aescolar we are not getting any of this stuff when we run coverage reports using gcovr/lcov.
Someone needs to fix that.

@andrewboie Do you mean we don't when run with the QEMU target? The unit tests are their own target: ```sanitycheck -p unit_testing -C```
The old codecov report was based on everything run in the 1st CI instance (which should include unit_testing, native_posix and nrf52_bsim).

andrewboie, 2019-06-20:
>  and the crc is covered by unit tests (their only user actually))

Why are the CRC tests a special snowflake?
This is a total, inconsistent mess!!

aescolar, 2019-06-20:
Don't ask me, it was there before I joined the project.
But you can see the positive side of it: the CRC test is a test of the unit tests and all related infrastructure ;-)

nashif, 2019-06-20:
last time I tried I was not able to combine results generated by qemu targets with native_poisx and unit_test.

crc in any case is well covered with the unit_test, shows 100% here when I run locally.

andrewboie, 2019-06-28:
@nashif what should we do about this with respect to our code coverage reports?
Should I convert this into a standard ztest? Leave it alone?

PR #17263 (2019-07-05): tests: crc: convert to regular test case
The special 'unittest' target has largely been superseded by
native_posix, and converting this to a regular test will allow
us to see code coverage for the CRC functions in our coverage
reports.

Fixes: #16943
commit 6c9cfabf (2019-07-05): tests: crc: convert to regular test case

The special 'unittest' target has largely been supersesed by
native_posix, and converting this to a regular test will allow
us to see code coverage for the CRC functions in our coverage
reports.

Fixes: #16943
commit b5087fe8 (2019-07-05): tests: crc: convert to regular test case

The special 'unittest' target has largely been supersesed by
native_posix, and converting this to a regular test will allow
us to see code coverage for the CRC functions in our coverage
reports.

Fixes: #16943
commit eb07943b (2019-07-05): tests: crc: convert to regular test case

The special 'unittest' target has largely been supersesed by
native_posix, and converting this to a regular test will allow
us to see code coverage for the CRC functions in our coverage
reports.

Fixes: #16943
closed 2019-07-05 by ioannisg