zephyrproject-rtos/zephyr #16961: Modules: add a SHA1 check to avoid updating module in the past

erwango, 2019-06-20:
**Is your feature request related to a problem? Please describe.**
In case of several PR on going having dependencies on the same module, it could be possible that the module PR are not merged in the same order as the zephyr main repo PRs.
In that case, when merging the last zephyr PR we should not update the module SHA1, otherwise we'll update SHA1 to a predecessor version of the module and latest SHA1 of the module will be removed.

For instance:
zephyr/pull/23 has dependency on module/pull/13
zephyr/pull/25 has dependency on module/pull/15

module/pull/13 is merged first and get SHA1:000001
module/pull/15 is merged second and get SHA1:000002

zephyr/pull/25 is merged first, so west.yml set module SHA1:000002
zephyr/pull/23 is the merged later and west.yml set module SHA1:000001

=> zephyr/pull/25 looses the SHA1 its depends on..

**Describe the solution you'd like**
Add a script to see if west.yml update is going in the past or going forward.

**Describe alternatives you've considered**
Check manually

mbolivar, 2019-06-26:
I think this is not a west feature as much as a zephyr policy. West is designed to allow projects to move back in time if that is what the user wants.

erwango, 2019-06-27:
Actually I didn't think about this proposal as a west feature but a rather github automatic check to help contributors and maintainers.
With multiple PR on going with dependencies on the same module, the exercise is close to a air traffic controller activity. I think we can benefit from an automatic check, and given the number of checks we're doing today, I don't think one more will make any difference.

closed 2023-01-26 by erwango