zephyrproject-rtos/zephyr #16968: silabs/gecko/emlib/src/em_gpio.c:111:35: warning: ?: using integer constants in boolean context [-Wint-in-bool-context]

marc-hb, 2019-06-21:
**Describe the bug**
ccache can surface a hidden warning in `silabs/gecko/emlib/src/em_gpio.c`. This is a sanitycheck showstopper because sanitycheck adds -Werror.
```
modules/hal/silabs/gecko/emlib/src/em_gpio.c: In function 'GPIO_DriveModeSet':
modules/hal/silabs/gecko/emlib/src/em_gpio.c:111:35: warning: ?: using integer constants in boolean context [-Werror=int-in-bool-context]
   EFM_ASSERT(GPIO_PORT_VALID(port) && GPIO_DRIVEMODE_VALID(mode));
           ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ 
```

The warning seems valid and easy to fix with some `(bool)` casts. It looks like it gets "optimized away" when not using ccache.

**Workarounds**
- Disable ccache - and everything becomes slower
- `rm -rf ~/.ccache; export CCACHE_CPP2=anything` and everything becomes slower but just once.

Second workaround and possible explanation found at http://petereisentraut.blogspot.com/2011/09/ccache-and-clang-part-2.html 

**To Reproduce**
```
mv ~/.ccache ~/ccache.saved
unset CCACHE_CPP2
cd ${ZEPHYR_BASE}/samples/synchronization
rm -rf build/
cmake -B build -S . -DBOARD=efm32hg_slstk3400a -DEXTRA_CFLAGS=-Werror
make -C build
```

**Expected behavior**
Compiles.

**Impact**
sanitycheck showstopper

**Environment (please complete the following information):**
 - OS: Ubuntu 16.04, ccache version 3.2.4
 - Toolchain  Zephyr SDK 0.10.0
- https://github.com/zephyrproject-rtos/hal_silabs 9151e614c239

**Additional context**
I verified that ccache doesn't change the .obj file, it only adds the warning.

marc-hb, 2019-06-21:
Following a suggestion from @joshtriplett I put ccache aside and separated the pre-processing and actual compilation steps:
```
$ arm-zephyr-eabi-gcc <long arg list copied from VERBOSE=1> -c em_gpio.c -E -o em_gpio.i
$ arm-zephyr-eabi-gcc <long arg list copied from VERBOSE=1> -c em_gpio.i 
```
This does produce the same warning on the second, actual compilation step. This confirms that the Zephyr SDK behaves somewhat differently when performing both passes at once and that the http://petereisentraut.blogspot.com/2011/09/ccache-and-clang-part-2.html article is not clang specific.

Whether it gets "optimized away" or not and when, the warning seems valid and the simplest seems to just fix it.

marc-hb, 2019-06-21:
I found why I hit this issue only on Ubuntu 16.04 and why sanitycheck doesn't fail on Ubuntu 18.04 or Fedora: newer distributions changed the default ccache setting:
```
ubuntu16$ ccache --print-config | grep cpp
(default) run_second_cpp = false
```
Description updated accordingly.

chrta, 2019-06-22:
Info: The current gecko sdk (emlib) 5.8.0 still contains this issue.

chrta, 2019-07-27:
Since i neither use ubuntu 16.04 nor ccache, i had to use seperate preprocessing to reproduce your problem.
Neither bool-casts, nor adding a `!!` fixed the issue for me, only this:
```diff
-#define GPIO_PORT_VALID(port)          (_GPIO_PORT_MASK(port) )
+#define GPIO_PORT_VALID(port)          (_GPIO_PORT_MASK(port) != 0)
```

@marc-hb Could you please verify that this change fixes the issue for you?

marc-hb, 2019-08-21:
I just tested it and I confirm the following change fixes the compilation issue reported (and the mismatch between the macro name and its previous code)
```diff
--- a/gecko/emlib/inc/em_gpio.h
+++ b/gecko/emlib/inc/em_gpio.h
@@ -406,7 +406,7 @@ extern "C" {
     : 0)
 
 /** Validation of port and pin. */
-#define GPIO_PORT_VALID(port)          (_GPIO_PORT_MASK(port) )
+#define GPIO_PORT_VALID(port)          (_GPIO_PORT_MASK(port) != 0)
 #define GPIO_PORT_PIN_VALID(port, pin) (((_GPIO_PORT_MASK(port)) >> (pin)) & 0x1)
 
 #if defined(_GPIO_EXTIPINSELL_MASK)
```
Disclaimer: compilation is all I tested. I don't see how this could affect anything else though.

> Since i neither use ubuntu 16.04 nor ccache, i had to use seperate preprocessing to reproduce your problem.

The reproduction script I put in the description works on any Linux distribution. What operating system do you use? On pretty much any Linux distribution using ccache is as simple as `apt-get install ccache` or `dnf install ccache` and done, no configuration required. To stop using just uninstall ccache.

galak, 2020-07-22:
Is this still an issue?  We don't see any failures w/sanitycheck building gpio support on Silabs.

marc-hb, 2020-07-22:
tl:dr: if you don't care about the programming error and only care about sanitycheck showing green lights then feel free to close as wontfix. Otherwise apply the trivial one-line fix above.

The warning is still valid and the code is still wrong.

However in 2016 ccache disabling its pre-processing optimization in order to align with compilers and show fewer warnings. See https://github.com/ccache/ccache/issues/93

So no one using a recent ccache version will ever see this (valid!) warning.

The `GPIO_PORT_VALID()` macros still returns an integer instead of a boolean:
 https://github.com/zephyrproject-rtos/hal_silabs/blob/78da967feeac0d5/gecko/emlib/inc/em_gpio.h#L433

`em_gpio.c` still uses it as a boolean:
 https://github.com/zephyrproject-rtos/hal_silabs/blob/78da967feeac0d512/gecko/emlib/src/em_gpio.c#L105

PS: I clarified the very short summary at the top.

chrta, 2020-08-13:
This bug was fixed by merging #27042

closed 2020-08-13 by chrta