zephyrproject-rtos/zephyr #16971: DFU supported for hci_uart sample ?

icephyr, 2019-06-21:
Hi guys, 
 Now I am research on the DFU function by uart. I use nrf52832 running hci_uart sample as controller to connect with the Host. All I want to do is to upgrade the firmware of nrf52832 by the host via uart. I learned some articles on the website, but it seems not to work with the hci_uart sample. Here is the conf I added into the config file.
`CONFIG_MCUMGR_SMP_UART=y
CONFIG_MCUMGR=y
CONFIG_MCUMGR_CMD_IMG_MGMT=y
`

but error occurs during the compilation.

> In file included from /home/icephyr/projects/ams/zephyrproject/zephyr/subsys/mgmt/buf.c:10:
/home/icephyr/projects/ams/zephyrproject/zephyr/include/mgmt/buf.h:11:10: fatal error: cbor_encoder_writer.h: No such file or directory
 #include "cbor_encoder_writer.h"
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
zephyr/subsys/mgmt/CMakeFiles/subsys__mgmt.dir/build.make:62: recipe for target 'zephyr/subsys/mgmt/CMakeFiles/subsys__mgmt.dir/buf.c.obj' failed
make[2]: *** [zephyr/subsys/mgmt/CMakeFiles/subsys__mgmt.dir/buf.c.obj] Error 1
CMakeFiles/Makefile2:1948: recipe for target 'zephyr/subsys/mgmt/CMakeFiles/subsys__mgmt.dir/all' failed
make[1]: *** [zephyr/subsys/mgmt/CMakeFiles/subsys__mgmt.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

So I wonder if the hci_uart sample can work with DFU by mcumgr, or any solutions to solve the compilation error.

carlescufi, 2019-06-21:
You want to update the hci_uart sample with mcumgr? That is not going to work because mcumgr requires the Bluetooth Host, and this sample only enables the Controller. Here are the options that you have:

1. Use mcumgr on the main CPU, the one that you use to connect to the CPU running hci_uart
2. Use MCUboot's serial recovery mode. Enable that in the chip running hci_uart and then the main CPU can update the firmware using it.

icephyr, 2019-06-24:
Thanks @carlescufi , I will have a try.

ioannisg, 2019-06-25:
@icephyr could this question be closed?

mfiumara, 2019-06-27:
I think this is a problem with the SMP Server example code. When I try to build on master I get the same error:
```
FAILED: CMakeFiles/app.dir/src/main.c.obj 
ccache /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v1.14.0-116-g21609c25621b -DKERNEL -DNRF52832_XXAA -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I/home/mattia/bitbucket/diamond/zephyr/kernel/include -I/home/mattia/bitbucket/diamond/zephyr/arch/arm/include -I/home/mattia/bitbucket/diamond/zephyr/include -I/home/mattia/bitbucket/diamond/zephyr/include/drivers -Izephyr/include/generated -I/home/mattia/bitbucket/diamond/zephyr/soc/arm/nordic_nrf/nrf52 -I/home/mattia/bitbucket/diamond/zephyr/soc/arm/nordic_nrf/include -I/home/mattia/bitbucket/diamond/zephyr/lib/libc/minimal/include -I/home/mattia/bitbucket/diamond/zephyr/ext/lib/crypto/tinycrypt/include -I/home/mattia/bitbucket/diamond/zephyr/ext/lib/fnmatch/. -I/home/mattia/bitbucket/diamond/zephyr/ext/hal/cmsis/Include -I/home/mattia/bitbucket/diamond/zephyr/ext/hal/nordic/nrfx -I/home/mattia/bitbucket/diamond/zephyr/ext/hal/nordic/nrfx/drivers/include -I/home/mattia/bitbucket/diamond/zephyr/ext/hal/nordic/nrfx/hal -I/home/mattia/bitbucket/diamond/zephyr/ext/hal/nordic/nrfx/mdk -I/home/mattia/bitbucket/diamond/zephyr/ext/hal/nordic/. -I/home/mattia/bitbucket/diamond/zephyr/subsys/bluetooth -I/home/mattia/bitbucket/diamond/zephyr/subsys/bluetooth/controller/ll_sw/nordic -I/home/mattia/bitbucket/diamond/zephyr/ext/fs/nffs/include -I/home/mattia/bitbucket/diamond/zephyr/ext/lib/mgmt/mcumgr/cborattr/include -I/home/mattia/bitbucket/diamond/zephyr/ext/lib/mgmt/mcumgr/cmd/fs_mgmt/include -I/home/mattia/bitbucket/diamond/zephyr/ext/lib/mgmt/mcumgr/cmd/img_mgmt/include -I/home/mattia/bitbucket/diamond/zephyr/ext/lib/mgmt/mcumgr/cmd/os_mgmt/include -I/home/mattia/bitbucket/diamond/zephyr/ext/lib/mgmt/mcumgr/cmd/stat_mgmt/include -I/home/mattia/bitbucket/diamond/zephyr/ext/lib/mgmt/mcumgr/mgmt/include -I/home/mattia/bitbucket/diamond/zephyr/ext/lib/mgmt/mcumgr/mgmt/port/zephyr/include -I/home/mattia/bitbucket/diamond/zephyr/ext/lib/mgmt/mcumgr/smp/include -I/home/mattia/bitbucket/diamond/zephyr/ext/lib/mgmt/mcumgr/util/include -isystem /opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include -isystem /opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include-fixed -Os -nostdinc -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -imacros /home/mattia/bitbucket/diamond/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/zephyr/include/generated/autoconf.h -ffreestanding -Wno-main -fno-common -mthumb -mcpu=cortex-m4 -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Werror=implicit-int -Wpointer-arith -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -std=c99 -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj   -c ../src/main.c
In file included from ../src/main.c:12:
/home/mattia/bitbucket/diamond/zephyr/include/mgmt/buf.h:11:10: fatal error: cbor_encoder_writer.h: No such file or directory
 #include "cbor_encoder_writer.h"
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

mfiumara, 2019-06-27:
It seems tinycbor was moved out of zephyr repo and into https://github.com/zephyrproject-rtos/tinycbor. How are you compiling exactly?

icephyr, 2019-07-16:
> @icephyr could this question be closed?

Sorry for replying so late, yes , the issue can be closed, thanks a lot for your suggestion.

icephyr, 2019-07-16:
> It seems tinycbor was moved out of zephyr repo and into https://github.com/zephyrproject-rtos/tinycbor. How are you compiling exactly? And what branch are you on of zephyr? For me it compiles on latest master but it does not on `v1.14-branch`.

so you compile successfully on the latest master ? I use the master branch, but failed.

closed 2019-07-16 by icephyr