zephyrproject-rtos/zephyr #16988: Packet isn't received by server during stepping

GeeseHoward, 2019-06-22:
**Summary:**
I have two applications (A and B) and I'm trying to communicate some data from application B to A over **UDP**/**OpenThread**.

Following are excerpts from the two applications (BP indicates breakpoints):
Application A:
```
	while(true){
		received = recvfrom(read_sock, recv_buffer,sizeof(recv_buffer), 0,
				&client_addr, &client_addr_len);
		printf("Packet received!");
BP		if (strstr(recv_buffer, param_is_header) == NULL){
			continue;
		}
		else{
			//header=(char*) malloc(sizeof(char)*received);
			//break;
			continue;
		}
	}
```
Application B:
```
	while(true){
BP		ret = send(write_sock, header, start, 0);
		if (ret < 0) {
			printf("UDP: Failed to send header\n");
			return;
		}
		received = recv(read_sock, recv_buf, sizeof(recv_buf),
				MSG_DONTWAIT);
		if (received <= 0) {
			continue;
		}
		else{
			break;
		}
	}
```
I launch and resume application A which hits the blocking call recvfrom.
I then launch application B. My expectation was that when I step over the send call, the breakpoint in application A would hit.

However, that doesn't seem to happen. When I step over the the send call, nothing happens. I tried to step over the call multiple times and the breakpoint at A still wouldn't hit.
But when I remove the breakpoint in B and resume the application, the breakpoint at A eventually hits. It sometimes takes several seconds before the breakpoint eventually hits.

I would like some insight as to why this is happening. Thank you.

carlescufi, 2019-06-22:
I will delegate the full response to @rlubos, but my guess is that when you stop the execution flow the relevant interrupts are not executed, causing the stack to go into undefined behavior. This is definitely the case for Bluetooth Low Energy, and I wouldn't be surprised if this was the case with Thread/15.4 as well.

GeeseHoward, 2019-06-22:
Thank you very much for the response, first of all.
I have one further observation to add.
I have now modified application B and added a sleep immediately after the send call:

Application B (with sleep):
```
	while(true){
BP		ret = send(write_sock, header, start, 0);
                k_sleep(1000);
		if (ret < 0) {
			printf("UDP: Failed to send header\n");
			return;
		}
		received = recv(read_sock, recv_buf, sizeof(recv_buf),
				MSG_DONTWAIT);
		if (received <= 0) {
			continue;
		}
		else{
			break;
		}
	}
```

When I step over the send call in B, the breakpoint in A doesn't hit like before. However, as soon as I step over k_sleep, the breakpoint in A consistently hits.

rlubos, 2019-06-24:
Hi @GeeseHoward ,

What thread priority does your application run on?

The networking stack in Zephyr uses its own internal thread for transmitting packets. Any interaction with the underlying L2 takes place from within that thread.

When you call `send` function from your application, the function quits after it queues your packet for sending. The packet will only be sent to the network after the TX queue is processed, which takes place in the networking TX thread. So the packet will not hit the network until the kernel switches the threads., and it could happen that it does not do so unless you enter `recv` or `k_sleep` call, which will yield the application thread.

You could use a sniffer to verify that, and see when the packet actually hits the network.

GeeseHoward, 2019-06-24:
Thanks for the response, @rlubos 

> What thread priority does your application run on?
I just have a main function in my application and no other threads, and I don't know what the priority of the main is.
The problem I'm facing is that this makes debugging a bit inconvenient. Right now I'm working around this by introducing `k_sleep` after the function calls, but if there's a better way to debug then that'd be great.
Would using threads with a different priority work?

> You could use a sniffer to verify that, and see when the packet actually hits the network
I do not know how to sniff 802.15.4 traffic. I did a quick google search and came across this project:
https://github.com/NordicSemiconductor/nRF-Sniffer-for-802.15.4
Is this what you use to sniff OT traffic? If not, are there any other guides etc that I can use to set up a sniffer?

rlubos, 2019-06-24:
@GeeseHoward 

Regarding sniffer, I use the one you've pointed to. Here're some extra docs on it: https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.thread_zigbee.v3.0.0%2Fnrf802154_sniffer.html&cp=5_2_4_0
All you need is an extra DK/dongle to flash the sniffer firmware into and a Wireshark. 

Regarding threads, I had an impression that if the application runs on a lower priority than the networking TX thread, the kernel would reschedule just after the TX queue is updated, but I might be wrong here. I'd need to build an app an play with it on my own to be certain here.
There's also an OpenThread thread in-between that could delay the packet, I'm not sure though if it's involved in the TX path.

rlubos, 2020-05-05:
Closing for now, @GeeseHoward please reopen if you wish to resume discussion.

closed 2020-05-05 by rlubos