zephyrproject-rtos/zephyr #16989: Errors when building application in Eclipse

GeeseHoward, 2019-06-22:
**Description**
I'm using Eclipse for application development.
When I clean and rebuild the application, the console log shows errors, but binary gets built nonetheless. However, the C/C++ Editor starts showing lots of 'Could not find symbol in index' errors.
If you make a minor modification in the application source and build the application again, the errors disappear.

The error seems to be somewhere around here, because these lines are highlighted in red in the build log:
```
configure.ac:198: installing 'third_party/nlbuild-autotools/repo/third_party/autoconf/compile'
configure.ac:114: installing 'third_party/nlbuild-autotools/repo/third_party/autoconf/missing'
```

While I'm not certain, I think this wasn't happening a few weeks ago.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja -DCONFIG_DEBUG_OPTIMIZATIONS=y -DCMAKE_BUILD_TYPE=Debug -DBOARD=nrf52840_pca10056 -DCONF_FILE="prj.conf overlay-ot.conf" -GEclipse CDT4 - Ninja /zephyrproject/zephyr/samples/net/sockets/echo_client
3. ninja
4. Import the application in Eclipse
5. Clean the application (Right-click on the project and click 'Clean Project')
6. Rebuild the application (Right-click on the project and click 'Build Project')

**Expected behavior**
No errors in the build log, no problems in the Problems tab and no errors in the C/C++ Editor in Eclipse.

**Impact**
Uncertain. For me it's a nuisance, but it might be causing some other issues as well.

**Screenshots or console output**
Screenshot:
[image]
[image]

Build log:
```
11:55:45 **** Build of project sockets_echo_client-Debug@echo_client_build ****
/usr/bin/ninja all 
[1/216] Preparing syscall dependency handling

[2/216] Creating directories for 'ot'
[3/216] Performing download step (git clone) for 'ot'
-- Avoiding repeated git clone, stamp file is up to date: '/home/<USERNAME>/zephyrproject/zephyr/samples/net/sockets/echo_client_build/zephyr/ext_proj/Stamp/ot/ot-gitclone-lastrun.txt'
[4/216] Performing update step for 'ot'
[5/216] No patch step for 'ot'
[6/216] Generating include/generated/syscall_macros.h
[7/216] Generating zephyr/include/generated/echo-apps-cert.der.inc
[8/216] Generating misc/generated/syscalls_subdirs.trigger
[9/216] Generating include/generated/driver-validation.h
[10/216] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h
[11/216] Generating misc/generated/syscalls.json
[12/216] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h
[13/216] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
[14/216] Generating include/generated/offsets.h
[15/216] Building C object CMakeFiles/app.dir/src/udp.c.obj
[16/216] Building C object CMakeFiles/app.dir/src/echo-client.c.obj
[17/216] Building C object zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/irq_init.c.obj
[18/216] Building C object zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/swap.c.obj
[19/216] Building C object zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/fault.c.obj
[20/216] Generating linker.cmd
[21/216] Building ASM object zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/fault_s.S.obj
[22/216] Building C object zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/irq_manage.c.obj
[23/216] Building C object zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/thread.c.obj
[24/216] Building C object zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/fatal.c.obj
[25/216] Linking C static library app/libapp.a
[26/216] Building ASM object zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/exc_exit.S.obj
[27/216] Building C object zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/sys_fatal_error_handler.c.obj
[28/216] Building C object zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/thread_abort.c.obj
[29/216] Building ASM object zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/isr_wrapper.S.obj
[30/216] Building C object zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/__aeabi_atexit.c.obj
[31/216] Building ASM object zephyr/arch/arch/arm/core/cortex_m/CMakeFiles/arch__arm__core__cortex_m.dir/vector_table.S.obj
[... 597 lines cut ...]
[204/216] Building C object zephyr/subsys/net/lib/openthread/platform/CMakeFiles/openthread_platform.dir/radio.c.obj
[205/216] Building C object zephyr/subsys/net/lib/openthread/platform/CMakeFiles/openthread_platform.dir/misc.c.obj
[206/216] Building C object zephyr/subsys/net/lib/openthread/platform/CMakeFiles/openthread_platform.dir/spi.c.obj
[207/216] Building C object zephyr/subsys/net/lib/openthread/platform/CMakeFiles/openthread_platform.dir/shell.c.obj
[208/216] Linking C static library zephyr/subsys/net/l2/openthread/libsubsys__net__ip__l2__openthread.a
[209/216] Linking C static library zephyr/subsys/net/lib/openthread/platform/libopenthread_platform.a
[210/216] Linking CXX static library zephyr/libzephyr.a
[211/216] Linking CXX executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      265164 B         1 MB     25.29%
            SRAM:       84280 B       256 KB     32.15%
        IDT_LIST:         104 B         2 KB      5.08%
[212/216] Generating linker_pass_final.cmd
[213/216] Generating isr_tables.c
[214/216] Building C object zephyr/CMakeFiles/zephyr_final.dir/misc/empty_file.c.obj
[215/216] Building C object zephyr/CMakeFiles/zephyr_final.dir/isr_tables.c.obj
[216/216] Linking CXX executable zephyr/zephyr.elf

11:56:17 Build Failed. 2 errors, 0 warnings. (took 31s.795ms)

```

**Environment:**
 - OS: Ubuntu
 - Toolchain: GNU ARM Embedded

carlescufi, 2019-06-22:
@GeeseHoward thanks for the report. Would it be possible for you to bisect this to find out which commit introduced the error?

GeeseHoward, 2019-06-22:
@carlescufi I quickly tried going back a few commits and the error is still reproducing. Going too far back starts giving me different errors entirely. I will keep trying.
One question: When you build an application with OpenThread the build system clones the OpenThread project as well. The files which are highlighted in red during the build are all located in the OpenThread project. Does this imply that the error is with OpenThread and not with Zephyr?

carlescufi, 2019-06-23:
>One question: When you build an application with OpenThread the build system clones the OpenThread project as well. The files which are highlighted in red during the build are all located in the OpenThread project. Does this imply that the error is with OpenThread and not with Zephyr?

It is likely, yes. But without more feedback from the experts on OpenThread I cannot say for sure.

rlubos, 2019-06-24:
So these two are indeed autotools reports, which OpenThread uses:
```
configure.ac:198: installing 'third_party/nlbuild-autotools/repo/third_party/autoconf/compile'
configure.ac:114: installing 'third_party/nlbuild-autotools/repo/third_party/autoconf/missing'
```
I don't see though why are they treated as errors. Possibly Eclipse simply misinterprets the results? All in all, there are two different build systems running for one project.
 
All other semantic errors refer to Zephyrs symbols - most likely an issue with Eclipse configuration. I haven't been using Eclipse with Zephyr for some time now so I won't be much help here.

carlescufi, 2019-06-25:
Those are really not errors @GeeseHoward. They are just autotools reporting installation of tools, this is probably Eclipse just being confused.

GeeseHoward, 2019-06-26:
> Those are really not errors @GeeseHoward. They are just autotools reporting installation of tools, this is probably Eclipse just being confused.
> 
> I reproduced them when building from the command-line, so this is just autotools displaying something.

I see.
Thank you for clarifying this!

closed 2019-08-13 by SebastianBoe