zephyrproject-rtos/zephyr #17004: arch/x86: build errors with newest build-grub.sh scripts

gnuless, 2019-06-23:
**Describe the bug**

Attempts to build grub with the script fail under Ubuntu 18.04.2 LTS.  This is a regression introduced with changes starting with #14468.

**Screenshots or console output**

```
$ ./build_grub.sh x86_64  ~/zephyr/zephyr/boards/x86/common/scripts/grub ~/zephyr/zephyr/boards/x86/common/scripts
Cloning into 'src'...
remote: Counting objects: 94083, done.
remote: Compressing objects: 100% (21614/21614), done.
remote: Total 94083 (delta 70136), reused 94067 (delta 70128)
Receiving objects: 100% (94083/94083), 70.21 MiB | 2.14 MiB/s, done.
Resolving deltas: 100% (70136/70136), done.
~/zephyr/zephyr/boards/x86/common/scripts/grub/src ~/zephyr/zephyr/boards/x86/common/scripts/grub ~/zephyr/zephyr/boards/x86/common/scripts
Note: checking out 'grub-2.04-rc1-17-g8e8723a6b'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 8e8723a6b f2fs: Disable gcc9 -Waddress-of-packed-member
~/zephyr/zephyr/boards/x86/common/scripts/grub ~/zephyr/zephyr/boards/x86/common/scripts
~/zephyr/zephyr/boards/x86/common/scripts/grub/src ~/zephyr/zephyr/boards/x86/common/scripts/grub ~/zephyr/zephyr/boards/x86/common/scripts
./bootstrap: Bootstrapping from checked-out grub sources...
./bootstrap: consider installing git-merge-changelog from gnulib
./bootstrap: getting gnulib files...
Cloning into 'gnulib'...

```
_system hangs here for quite some time, there's a host lookup issue with the specified gnulib repo in the script, it then continues.._

```
remote: Counting objects: 204496, done.
remote: Compressing objects: 100% (27335/27335), done.
remote: Total 204496 (delta 177392), reused 204109 (delta 177095)
Receiving objects: 100% (204496/204496), 42.68 MiB | 5.06 MiB/s, done.
Resolving deltas: 100% (177392/177392), done.
Note: checking out 'd271f868a8df9bbec29049d01e056481b7a1a263'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at d271f868a lock: Fix link error with --enable-threads=pth.
./bootstrap: autopoint --force
Copying file ABOUT-NLS
Copying file build-aux/config.rpath
Creating directory m4
Copying file m4/codeset.m4
Copying file m4/extern-inline.m4
Copying file m4/fcntl-o.m4
Copying file m4/gettext.m4
Copying file m4/glibc2.m4
Copying file m4/glibc21.m4
Copying file m4/iconv.m4
Copying file m4/intdiv0.m4
Copying file m4/intl.m4
Copying file m4/intldir.m4
Copying file m4/intlmacosx.m4
[... 868 lines cut ...]
autoreconf: Entering directory `.'
autoreconf: running: autopoint --force
Copying file build-aux/config.rpath
Copying file m4/extern-inline.m4
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
configure.ac:414: error: possibly undefined macro: AC_CHECK_HEADERS
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:505: error: possibly undefined macro: AC_LINK_IFELSE
configure.ac:1550: error: possibly undefined macro: AC_LANG_CALL
autoreconf: /usr/bin/autoconf failed with exit status: 1
configure: error: cannot find install-sh, install.sh, or shtool in build-aux "."/build-aux
make: *** No targets specified and no makefile found.  Stop.
./build_grub.sh: line 39: ./grub-mkimage: No such file or directory
~/zephyr/zephyr/boards/x86/common/scripts/grub ~/zephyr/zephyr/boards/x86/common/scripts
cp: cannot stat 'src/grub_x86_64.efi': No such file or directory
```

**Environment (please complete the following information):**

Ubuntu 18.04.2 LTS
SDK 0.10.1

gvancuts, 2019-07-02:
I cannot reproduce this on my side (using a container). I believe this error can be caused by a missing dependency, specifically: `pkg-config`. Can you confirm all these packages are installed on your host: `bison autoconf libopts25-dev flex automake pkg-config gettext autopoint`?

Regarding the lookup issue (for `gnulib`), you are not the first one to report it but I have never seen it myself... I wonder if it's one of those weird global DNS lookup issues that are different based on where you are located.

nashif, 2019-08-21:
low impact, lowering priority.

rossburton, 2019-11-19:
FWIW I just had this exact problem and installing pkg-config was the solution.

gvancuts, 2019-11-20:
Thanks for the feedback @rossburton 

@gnuless let me close this issue since we cannot reproduce it and have had no further feedback from your side on it. Feel free to re-open it if it is still an issue.

closed 2019-11-20 by gvancuts

caomufei, 2019-11-22:
> FWIW I just had this exact problem and installing pkg-config was the solution.
excuse me, where can I  download the pkg-config ? I has this problem too!

rossburton, 2019-11-22:
@caomufei your distribution will have a package for this.

SauloAlessandre, 2023-04-13:
I had the same problem, even after installing quoted packages by [gvancuts](https://github.com/gvancuts). But in my case it was the lack of python, because I only had the python3 installed on my computer, so installing python or creating a link from /usr/bin/python to /usr/bin/python3, solved the problem.