zephyrproject-rtos/zephyr #17007: USB mass demo format fails on frdm_k64f

MarkWangChinese, 2019-06-24:
**Describe the bug**
with the native kinetis usb controller driver, the mass demo runs unstable.

**To Reproduce**
Steps to reproduce the behavior:
1. Increase the ram disk size by changing #define RAMDISK_VOLUME_SIZE (32 * 512) to #define RAMDISK_VOLUME_SIZE (300 * 512) in file fat12_ramdisk.h
2. Execute the command line: cmake -GNinja -DOVERLAY_CONFIG="overlay-ram-disk.conf" -DBOARD=frdm_k64f  ..
3. build and download the demo to board.
4. Connect to PC (windows), and format the disk, it will fail. my PC is Win10.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: arm gcc in windows.
 - 0bed1007edc1c73dbe3e84263cdd99cd09ed7c48

MarkWangChinese, 2019-06-24:
@jfischer-phytec-iot Do you have time to check it? or I will check it later. I tried the shim driver, and it is OK.

jfischer-phytec-iot, 2019-07-05:
Work fine for me, tested on Linux.

dd if=/dev/urandom of=test.img bs=512 count=300
dd if=test.img of=/dev/sdd bs=512 count=300
dd of=test1.img if=/dev/sdd bs=512 count=300
diff test1.img test.img

I had to disable MPU. I usually have problems on the frdm_k64f when MPU is enabled. Bisect identified aa826960d89e066bd2219b0576a0c4f3a475cb3f, but I did not have more time to investigate.

ioannisg, 2019-07-28:
There have been some recent MPU fixes in #17603: @jfischer-phytec-iot, could you check if this still needs MPU to be disabled, in order to work?

agansari, 2019-07-29:
@MarkWangChinese @jfischer-phytec-iot  i've added a new pull #17834 that should fix USB's MPU access to RAM, not sure about formatting in Windows. Can you please test?
USB is a different master (4) than ethernet (3) @ioannisg.

ioannisg, 2019-07-29:
@MarkWangChinese @jfischer-phytec-iot can you test this patch?
Thanks @agansari

MarkWangChinese, 2019-07-30:
Yes, I do one simple test, it seems fix the issue. I will do more testes later (today), and give the result.

ioannisg, 2019-07-30:
> 
> 
> Yes, I do one simple test, it seems fix the issue. I will do more testes later (today), and give the result.

Thanks @MarkWangChinese , let us know of the result of your testing!

MarkWangChinese, 2019-07-30:
I did the test and it solved this issue.

ioannisg, 2019-07-30:
Then it seems we are good to go FWIW, I ran the complete sanitycheck on frdm-k64f on the related PR, to confirm that no tests are affected.

PR #17834 (2019-07-30): soc: k64f MPU configured to always allow USB
Because NXP MPU's regions are dynamically enabled/disabled, USB
device's access maybe restricted when switching out of a task.
Background DMA transfers to/from RAM may happen during MPU region
reconfiguration or core idling.
Enabled USB (Kinetis MPU Master 4) to always have access to RAM address
space.

Fixes #17007
closed 2019-07-30 by ioannisg