zephyrproject-rtos/zephyr #17015: #15910 Breaks Ethernet on STM32F7

alexanderwachter, 2019-06-24:
**Describe the bug**
After #15910 Ethernet seems to be broken.

**To Reproduce**
Steps to reproduce the behavior:
1. `git checkout a4f3e628e51a1d5cd6fe835aac4058dc3bd56a28`
2. `west build -b nucleo_f746zg zephyr/samples/net/zperf && west flash`
3. open terminal and type `net iface` to get the IPv6 address
4. Connect to host with Ethernet
5. Try to ping the board
6. Board answer correctly
7. `git checkout d94c0d39f561626c55e2e9ddc054d1ec4f055817`
8. `rm -r build && west build -b nucleo_f746zg zephyr/samples/net/zperf && west flash`
9. open terminal and type `net iface` to get the IPv6 address
10. Connect to host with Ethernet
11. Try to ping the board
12. Board answers sometimes but late.

**Expected behavior**
Echo reply should arrive in about 1-5ms

**Impact**
Ethernet doesn't work on STM32F7

erwango, 2019-06-24:
Likely a consequence of cache being enabled as a result of moving from DTCM to SRAM.
Can you test  to disable cache in case ethernet is used?
So in `soc/arm/st_stm32/stm32f7/soc.c`:
```
+#ifndef CONFIG_ETH_STM32_HAL
 	SCB_EnableICache();
 	if (!(SCB->CCR & SCB_CCR_DC_Msk)) {
 		SCB_EnableDCache();
 	}
+#endif /* CONFIG_ETH_STM32_HAL */
```
Txs

alexanderwachter, 2019-06-25:
@erwango deactivating the caches doesn't help.

alexanderwachter, 2019-06-26:
Findings so far:
 - Deactivating DCache doesn't help
 - Adding `SCB_InvalidateDCache_by_Addr` before and after DMA operations doesn't help
 - Enabling and adding __nocache to the DMA buffers and structures doesn't help

erwango, 2019-07-23:
#17123 has been merged.
Let me close this issue

closed 2019-07-23 by alexanderwachter