zephyrproject-rtos/zephyr #17017: #16827 Breaks Ethernet on FRDM-K64F

alexanderwachter, 2019-06-24:
**Describe the bug**
After #16827 Ethernet doesn't work on FRDM board.
This seems weird because it has nothing to do with Ethernet.
After removing this commit from master Ethernet works again

**To Reproduce**
Steps to reproduce the behavior:
1. `git checkout 2b32059a618bf61e153a0fd29b21f0009dcd785d~1`
2. `west build -b frdm_k64f zephyr/samples/net/zperf && west flash`
3. open terminal and type `net iface` to get the IPv6 address
4. Connect to host with Ethernet
5. Try to ping the board
6. Board answer correctly
7. `git checkout 2b32059a618bf61e153a0fd29b21f0009dcd785d`
8. `rm -r build && west build -b frdm_k64f zephyr/samples/net/zperf && west flash`
11. Try to ping the board
12. Board doesn't send a response .

**Expected behavior**
Board should answer echo request within some milliseconds

**Impact**
Ethernet on FRDM (and maybe for other NXP MCUS) is not working.

aescolar, 2019-06-24:
CC @npitre  for your information

npitre, 2019-06-24:
Is there a way to reproduce this in qemu or something without actual
hardware?

jukkar, 2019-06-24:
I tested this and Ethernet works properly with qemu_x86, native_posix and also with sam_e70_xplained boards. I used `samples/net/sockets/echo-server` app in my tests.
I also verified that Ethernet did not work with latest master in frdm-k64f, the board was not able to send or receive packets. I did not bother to bisect the offending commit as @alexanderwachter already had done that.

npitre, 2019-06-24:
Unless commit 2b32059a61 introduced a bug I can't spot, the only other 
effect would be a possible code placement difference. I'm afraid this 
will require some investigation using pyOCD on the affected hardware, or 
the addition of various printk statements in key areas of the code to 
find out what's going on.

agansari, 2019-07-08:
@alexanderwachter i'm sorry but i can't reproduce this issue, i followed the steps you mentioned in the PR description and it responds on both with pings. I've also tested master and pinging works.
Can you please recheck with current master?

It maybe something in my setup that I'm missing. I keep my board pluged into the eth cable.
Do you mean "Connect to host with Ethernet" you plug the cable after flashing the board? In that case it maybe related to #16089

agansari, 2019-07-18:
This issue does not reproduce 100% time (i didn't manage to reproduce it)
Perhaps this issue is the same case #17603 fixes. Sometimes ENET's DMA access to RAM is cut off when switching context (random behavior depending on code behavior and execution time).

Does not reproduce on latest master.
@alexanderwachter I think this issue can be closed.

alexanderwachter, 2019-07-18:
Ethernet seems to always work on actual master. Was not able to find the actual commit that fixes it with bisect. Seems to be random somehow.

closed 2019-07-18 by alexanderwachter