zephyrproject-rtos/zephyr #17031: Compiler warnings in settings module in Zephyr 1.14

DeclanTraill, 2019-06-25:
@nvlsianpu 

We are getting a number of Compiler Warnings when building Zephyr 1.14 (See below).
The variable rc is not initialized in these functions and could potentially get returned without being set to a value. In actual fact it may not be possible for this to happen due to the values passed into these functions being correct, but the Compiler cannot determine this, and it would be safer to actively prevent this possibility in the functions themselves.

Can you please advise on the best way to correct these warnings?
It seems that simply initializing the rc variables to an invalid value when they are declared would achieve this - do you agree?

Regards,
Declan Traill
Setec Pty Ltd

[50/254] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/settings/src/settings_store.c.obj
C:/Users/declan.traill/RVMN101/zephyr/subsys/settings/src/settings_store.c: In function 'settings_cmp':
C:/Users/declan.traill/RVMN101/zephyr/subsys/settings/src/settings_store.c:113:9: warning: 'rc' may be used uninitialized in this function [-Wmaybe-uninitialized]
  return rc;
         ^~
[52/254] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/settings/src/settings_line.c.obj
C:/Users/declan.traill/RVMN101/zephyr/subsys/settings/src/settings_line.c: In function 'settings_line_raw_read_until':
C:/Users/declan.traill/RVMN101/zephyr/subsys/settings/src/settings_line.c:272:19: warning: 'rc' may be used uninitialized in this function [-Wmaybe-uninitialized]
   return (rc) ? 0 : 1;
          ~~~~~~~~~^~~
C:/Users/declan.traill/RVMN101/zephyr/subsys/settings/src/settings_line.c: In function 'settings_entry_copy':
C:/Users/declan.traill/RVMN101/zephyr/subsys/settings/src/settings_line.c:444:9: warning: 'rc' may be used uninitialized in this function [-Wmaybe-uninitialized]
  return rc;
         ^~
[55/254] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/settings/src/settings_fcb.c.obj
C:/Users/declan.traill/RVMN101/zephyr/subsys/settings/src/settings_fcb.c: In function 'settings_fcb_save':
C:/Users/declan.traill/RVMN101/zephyr/subsys/settings/src/settings_fcb.c:270:5: warning: 'rc' may be used uninitialized in this function [-Wmaybe-uninitialized]
  if (rc) {
     ^

ioannisg, 2019-06-25:
@DeclanTraill do we get the same warnings in master?

DeclanTraill, 2019-06-26:
> 
> 
> @DeclanTraill do we get the same warnings in master?

Looking at master, some of the code has changed but some still have the compiler warning issue.
Also, we noticed that the warnings seem to only happen in a Debug optimized build (not a Release build).

ioannisg, 2019-06-26:
Thanks! If you see warnings in master, pls, could you open a separate issue?

nvlsianpu, 2019-08-12:
Was fixed in #16078

PR #17078 (2019-07-31): Settings: Fixes for compiler warnings
@nvlsianpu
Fixes #17031:
Compiler warnings in settings module in Zephyr 1.14
closed 2019-08-12 by nvlsianpu