zephyrproject-rtos/zephyr #17053: Bluetooth Mesh: Periodic Publishing

EddLeon, 2019-06-25:
**Describe the bug**
I am working with bbc_microbit modules, using Bluetooth-Mesh examples from the Bluetooth official website as a base (self provisioning and configuring). I want to set a periodic publishing but the callback function is not being called.

**To Reproduce**
Steps to reproduce the behavior:
```c
static int vend_pub_update(struct bt_mesh_model *mod){
       printk("pub_update called \n");
       return 0;
}
BT_MESH_MODEL_PUB_DEFINE(vnd_pub, vend_pub_update, 3 + 1); 
static struct bt_mesh_model vnd_models[] = {                                                                                                                                                                                          
          BT_MESH_MODEL_VND(BT_COMP_ID_LF, MOD_LF, vnd_ops, &vnd_pub, NULL),                                                                                                                                                            
  }; 
```

Inside the configuration function:
``` c
struct bt_mesh_cfg_mod_pub pub = {                                                                                                                                                                                            
                  .addr = GROUP_ADDR,                                                                                                                                                                                                   
                  .app_idx = APP_IDX,                                                                                                                                                                                                   
                  .ttl = DEFAULT_TTL,                                                                                                                                                                                                   
                  .period = BT_MESH_PUB_PERIOD_SEC(5),                                                                                                                                                                                  
          }; 

bt_mesh_cfg_mod_pub_set_vnd(NET_IDX, NODE_ADDR, NODE_ADDR, MOD_LF, BT_COMP_ID_LF,  &pub, NULL);
```

Relevant Kernel configs:
```
CONFIG_BT=y 
CONFIG_BT_BROADCASTER=y
CONFIG_BT_OBSERVER=y
CONFIG_BT_MESH=y  
CONFIG_BT_MESH_RELAY=y
CONFIG_BT_DEBUG_LOG=y
CONFIG_BT_MESH_DEBUG=y
```

**Expected behavior**
Seeing the printk on the callback function (*update) every n seconds (as set on the bt_mesh_cfg_mod_pub .period) in order to update the message to be published

**Impact**
I want to set up an experiment where multiple sensors publish at different/random moments in time. Not having the periodic callbacks complicates the development.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit:  e0943cfaab29100c5212f74aa9f3eb1a4aa58f0b

jhedberg, 2019-06-25:
Could you enable `CONFIG_BT_DEBUG_LOG=y` and see if the stack is outputting any warnings or errors to the console? If I remember right the set vendor publication message is so large that it has to be segmented, so you need to be careful to make sure you have sufficient buffers and segmentation contexts to deal with it as well as the response, *especially* if you're sending the message to yourself (as you seem to be doing) rather than using a proper external configuration client. I can imagine you've got the various resources tuned to the absolute minimum due to the limited memory (16k) on the micro:bit.

EddLeon, 2019-06-25:
Thanks for the reply!

The objective is to periodically broadcast data to different group addresses. The problem, as mentioned before is that the callback function on the "broadcaster" is never called and thus, the publishing does not take place (in a periodic fashion), if I make a normal publish (bt_mesh_model_publish()) it works fine.

This is the output of my current test code:
```
***** Booting Zephyr OS build zephyr-v1.14.0-1721-ge0943cfaab29 *****
Bluetooth initialised OK
Provisioning completed
self-provisioned OK
configuring...
added appkey
bound appkey to generic onoff server model
bound appkey to generic onoff server model
bound appkey to health server model
subscribed to group address
Mesh initialised OK
[00:00:00.047,454] <dbg> bt_mesh_main.bt_mesh_provision: Storing network information persistently
--- 11 messages dropped ---
[00:00:00.137,390] <wrn> bt_mesh_cfg_cli: Unexpected Model App Status message
[00:00:00.138,214] <wrn> bt_mesh_cfg_cli: Unexpected Model App Status message
[00:00:00.139,007] <wrn> bt_mesh_cfg_cli: Unexpected Model Subscription Status message
```

I don't know what the dropped packets mean and so far I assumed that the Status messages where the periodic Heartbeats sent by the same node. This I am not entirely sure, maybe @jhedberg  can clear out this.

jhedberg, 2019-06-26:
@EddLeon the "11 messages dropped" means you have a too small log buffer size and some log messages have been dropped because of it. You could try increasing CONFIG_LOG_BUFFER_SIZE to the extent that available RAM allows.

jhedberg, 2019-06-26:
@EddLeon for the record, I don't see anything obviously wrong with the code snippets you've shown so far. The fact that `bt_mesh_model_publish()` works indicates that the the configuration message should have been successfully processed. I've most recently been using `samples/boards/reel_board/mesh_badge` on reel boards, which also uses periodic publishing, and has worked fine. You may want to compare the code in `samples/boards/reel_board/mesh_badge/src/mesh.c`. I'll try to find time to test that again today to see if there might have been a regression since I last tried it.

EddLeon, 2019-06-26:
Thanks for the tip on the CONFIG_LOG_BUFFER_SIZE, it seems that the default size is 128 after some tweaking I pushed the ram to 99% usage with CONFIG_LOG_BUFFER_SIZE=496 and got the following output:

```
***** Booting Zephyr OS build zephyr-v1.14.0-1721-ge0943cfaab29 *****
light node 7
Bluetooth initialised OK
Provisioning completed
self-provisioned OK
configuring...
added appkey
bound appkey to generic onoff server model
bound appkey to generic onoff server model
bound appkey to health server model
subscribed to group address
Mesh initialised OK
[00:00:00.020,050] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 1.14 Build 99
--- 2 messages dropped ---
[00:00:00.020,935] <wrn> bt_hci_core: No ID address. App must call settings_load()
[00:00:00.023,315] <inf> bt_mesh_main: Primary Element: 0x0007
[00:00:00.023,376] <dbg> bt_mesh_main.bt_mesh_provision: net_idx 0x0000 flags 0x00 iv_index 0x0000
[00:00:00.049,102] <dbg> bt_mesh_main.bt_mesh_provision: Storing network information persistently
[00:00:00.049,316] <err> bt_mesh_adv: starting scan failed (err -11)
[00:00:00.080,200] <err> bt_mesh_transport: No multi-segment message contexts available
[00:00:00.080,200] <err> bt_mesh_cfg_cli: model_send() failed (err -16)
[00:00:00.132,080] <wrn> bt_mesh_cfg_cli: Unexpected App Key Status message
[00:00:00.132,843] <wrn> bt_mesh_cfg_cli: Unexpected Model App Status message
[00:00:00.133,636] <wrn> bt_mesh_cfg_cli: Unexpected Model App Status message
[00:00:00.134,490] <wrn> bt_mesh_cfg_cli: Unexpected Model App Status message
[00:00:00.135,314] <wrn> bt_mesh_cfg_cli: Unexpected Model Subscription Status message
```

By the way, I was originally trying to setup a Sensor Server publisher and was already using the samples on the reel_board, nrf52 and the microbit's. Tried multiple tests, backtracked to a basic example trying to reproduce the same Vendor publishing as on the `samples/boards/reel_board/mesh_badge/src/mesh.c` without any success so far.

Any insight on the issue (queues  and triggers) that might not be working or if the log I just updated exposes some mistake I've done, please let me know @jhedberg I would really appreciate it.

jhedberg, 2019-06-26:
> [00:00:00.049,316] <err> bt_mesh_adv: starting scan failed (err -11)

That's quite odd, and serious since that means you wouldn't be able to receive anything over mesh (since we're not scanning for advertising packets). It's as if the controller is in some strange state. Haven't seen that before - would probably need to see the full application source code to understand better.

> [00:00:00.080,200] <err> bt_mesh_transport: No multi-segment message contexts available

That one is easy, and matches what I originally suspected, i.e. the one or more of your configuration messages, or possibly their responses, requires segmentation, and you don't have sufficient buffers/contexts available. You would need to increase `CONFIG_BT_MESH_TX_SEG_MSG_COUNT` to fix it.

EddLeon, 2019-06-26:
@jhedberg 
> [00:00:00.049,316] bt_mesh_adv: starting scan failed (err -11)

 Was caused for some reason by the `[*] Store Bluetooth state and configuration persitently` option,  which I think is `CONFIG_BT_SETTINGS`. Maybe the self-provisioning/configuration has something to do with that. If the option is removed, the error disappears and the node works again. I am still checking if the `bt_mesh_model_publish()` actually works since I may have introduced some errors in this particular test code (such as the previous config).

I will make another "clean" test sample and upload it to git with the increased `CONFIG_BT_MESH_TX_SEG_MSG_COUNT`, maybe it will become clearer what the problem is and you'll have some spare time to check it out!

Thanks again for the feedback.

EddLeon, 2019-06-27:
@jhedberg 
> CONFIG_BT_MESH_TX_SEG_MSG_COUNT=3

Did the trick! The callback is getting triggered and I am able to publish data from the Sensor Server; subscribers receive the Status properly.
Thank you so much.

jhedberg, 2019-06-27:
@EddLeon good to hear! I'll close this issue then.

closed 2019-06-27 by jhedberg