zephyrproject-rtos/zephyr #17063: tests/kernel/tickless/tickless_concept/ fails even outside of CI

carlescufi, 2019-06-25:
I am getting these failures on master on my machine:

```
SeaBIOS (version rel-1.12.0-0-ga698c8995f-prebuilt.qemu.org)
Booting from ROM..***** Booting Zephyr OS build zephyr-v1.14.0-1721-ge0943cfaab29 *****
Running test suite tickless_concept
===================================================================
starting test - test_tickless_sysclock
time 10, 220
time 230, 430
PASS - test_tickless_sysclock
===================================================================
starting test - test_tickless_slice
elapsed slice 10118461, expected: <9900000, 10100000>

    Assertion failed at ZEPHYR_BASE/tests/kernel/tickless/tickless_concept/src/main.c:77: thread_tslice: t <= SLICE_SIZE_MAX_CYCLES is false

elapsed slice 219997, expected: <9900000, 10100000>

    Assertion failed at ZEPHYR_BASE/tests/kernel/tickless/tickless_concept/src/main.c:76: thread_tslice: t >= SLICE_SIZE_MIN_CYCLES is false

elapsed slice 158424, expected: <9900000, 10100000>

    Assertion failed at ZEPHYR_BASE/tests/kernel/tickless/tickless_concept/src/main.c:76: thread_tslice: t >= SLICE_SIZE_MIN_CYCLES is false

elapsed slice 328287, expected: <9900000, 10100000>

    Assertion failed at ZEPHYR_BASE/tests/kernel/tickless/tickless_concept/src/main.c:76: thread_tslice: t >= SLICE_SIZE_MIN_CYCLES is false

FAIL - test_tickless_slice
===================================================================
Test suite tickless_concept failed.
===================================================================
PROJECT EXECUTION FAILED
QEMU: Terminated
```

carlescufi, 2019-06-25:
I can reproduce locally consistently on latest master

andrewboie, 2019-06-25:
Also seen on qemu_xtensa

andrewboie, 2019-06-25:
Also this test failed 100% of the time if CONFIG_COVERAGE was enabled, I had to ifdef out the asserts in test_tickless_slice to work around it

andyross, 2019-06-26:
A somewhat tedious[1] bisection pointed at commit 62c71dc4d82f351.  That patch changed the error threshold for measuring a timeslice length from half of CONFIG_TICKLESS_IDLE_THRESHOLD (which was, OK, really weird -- I don't know what the test was trying to do) to +/- 1ms.  Timeslicing is measured in ticks, so it won't work to try to set this value faster than the tick rate (which is 10ms in HEAD, and will remain so on all the emulation platforms until we get deterministic emulation working).

[1] Turns out there was a big stretch in the series which would fail to build because of a DTS warning about QMSI (introduced when QMSI moved out of tree and into Westworld), and I had to dig out the patch to apply to fix that to see through the intermediate bisection points.

PR #17082 (2019-06-26): Revert "tests: tickless_concept: Fix slicing time measurement"
Unfortunately this seems to have introduced spurious failures on (at
least) qemu_x86 and qemu_xtensa.

The change limits the timeslice tolerance to +/- 1ms, which isn't
necessarily correct when the tick rate is less than 1ms (though it
will probably work on deterministic hardware as long as the system is
hitting the target at exactly the right tick), and isn't even
theoretically achievable on emulation environments where timing
granularity is limited by the host scheduling quantum.

What this needs to do is check the deadline is of [... cut]
commit c0630346 (2019-06-26): Revert "tests: tickless_concept: Fix slicing time measurement"

Unfortunately this seems to have introduced spurious failures on (at
least) qemu_x86 and qemu_xtensa.

The change limits the timeslice tolerance to +/- 1ms, which isn't
necessarily correct when the tick rate is less than 1ms (though it
will probably work on deterministic hardware as long as the system is
hitting the target at exactly the right tick), and isn't even
theoretically achievable on emulation environments where timing
granularity is limited by the host scheduling quantum.

What this needs to do is check the deadline is o [... cut]
closed 2019-06-26 by carlescufi