zephyrproject-rtos/zephyr #17095: Building with Xtensa toolchain fails

jhedberg, 2019-06-26:
This worked until very recently, now I get:
```
$ west build -b intel_s1000_crb samples/hello_world
...
8/93] Building C object zephyr/CMakeFiles/offsets.dir/arch/xtensa/core/offsets/offsets.c.obj
FAILED: zephyr/CMakeFiles/offsets.dir/arch/xtensa/core/offsets/offsets.c.obj 
ccache /home/jh/xtensa/XtDevTools/install/tools/RF-2016.4-linux/XtensaTools/bin/xt-xcc -DBUILD_VERSION=zephyr-v1.14.0-1764-gef6c4d8d3b15 -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../kernel/include -I../arch/xtensa/include -I../include -I../include/drivers -Izephyr/include/generated -I../soc/xtensa/intel_s1000 -I../soc/xtensa/intel_s1000/include -I../lib/libc/minimal/include -Os --imacros=/home/jh/src/zephyr/zephyr/build/zephyr/include/generated/autoconf.h --imacros=/home/jh/src/zephyr/zephyr/include/toolchain/zephyr_stdint.h -ffreestanding -fno-common -g -fms-extensions -D__SIZEOF_LONG__=4 -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Werror=implicit-int -fno-pic -fno-strict-overflow -ffunction-sections -fdata-sections -mlongcalls -std=c99 -MD -MT zephyr/CMakeFiles/offsets.dir/arch/xtensa/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/xtensa/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/xtensa/core/offsets/offsets.c.obj   -c /home/jh/src/zephyr/zephyr/arch/xtensa/core/offsets/offsets.c
cc1: error: =/home/jh/src/zephyr/zephyr/build/zephyr/include/generated/autoconf.h: No such file or directory
cc1: error: =/home/jh/src/zephyr/zephyr/include/toolchain/zephyr_stdint.h: No such file or directory
In file included from ../include/toolchain/xcc.h:10,
                 from ../include/toolchain.h:37,
                 from ../kernel/include/gen_offset.h:74,
                 from /home/jh/src/zephyr/zephyr/arch/xtensa/core/offsets/offsets.c:26:
```

If I rebase back to a few days old tree everything works. I suspect #16645 might be the culprit.

jhedberg, 2019-06-26:
If I reset the tree to commit 549bd60624ed96dfe1583b9f21415ae2b035b0fd (i.e. before the recent patches from @npitre ) then I can build without issues.

npitre, 2019-06-26:
Let me find a copy of xcc and I'll fix this.

npitre, 2019-06-27:
I got an SDK evaluation license but can't figure out how to install it 
properly. Nothing I can download from cadence.com matches the Zephyr 
documentation for the Intel S1000 CRB either. Fixing the actual problem 
certainly would have taken less time than futzing around with this 
toolchain to no avail.

Is there a way for me to just log into somewhere with all this stuff 
already working?

In the mean time it would be very helpful if you could identify the 
first commit that broke compilation for you. I really doubt it is commit 
cf5c22d8bef7.

In particular, does it still work with commit f57ba2d30c2d?

If so then the -D__SIZEOF_LONG__=4 in 
zephyr/cmake/toolchain/xcc/generic.cmake would be a big clue already.

Let me know.

jhedberg, 2019-06-27:
Resetting my tree to commit f57ba2d30c2d7aaea8298f4d931901f09f5c1fc0 still fails, however resetting to commit f57ba2d30c2d7aaea8298f4d931901f09f5c1fc0 (the one before it) works. So looks like commit f57ba2d30c2d7aaea8298f4d931901f09f5c1fc0 is at fault. I tried reverting it on top of my tree but that caused a different build failure (presumably because subsequent patches depend on it).

npitre, 2019-06-27:
On Thu, 27 Jun 2019, Johan Hedberg wrote:

> So looks like commit f57ba2d30c2d7aaea8298f4d931901f09f5c1fc0 is at fault.

OK.  This is not nice.

Do you know where I could find documentation on xcc command line usage?

> I tried reverting it on top of my tree but that caused a different build failure (presumably because subsequent patches depend on it).

Rather than reverting it, could you try the following:

In cmake/compiler/gcc/target_imacros.cmake, replace

    zephyr_compile_options(--imacros=${header_file})

with this

    zephyr_compile_options(-imacros${header_file})

and let me know what error you then get.

npitre, 2019-07-02:
FYI: I managed to install xcc and I know what the problem is. Expect a fix shortly.

jhedberg, 2019-07-02:
> FYI: I managed to install xcc and I know what the problem is. Expect a fix shortly.

@npitre thanks, and sorry for not being able to provide you with timely feedback - I've had to focus on other priorities.

npitre, 2019-07-02:
Fixes are in PR #17267.

PR #17267 (2019-07-03): make compilation with xcc work again (fix issue #17095)
Fixes #17095
closed 2019-07-03 by nashif