zephyrproject-rtos/zephyr #17111: nucleo_f030r8 build error

maweicoltsmart, 2019-06-27:
in master branch with board nucleo_f030r8,build error .
the log below

/arch/arm/core/offsets/offsets.c
In file included from ../../../../include/arch/arm/cortex_m/cmsis.h:17,
                 from ../../../../arch/arm/include/cortex_m/stack.h:27,
                 from ../../../../arch/arm/include/kernel_arch_data.h:53,
                 from ../../../../kernel/include/kernel_structs.h:68,
                 from /home/coltsmart/ethdtu/arch/arm/core/offsets/offsets.c:26:
../../../../soc/arm/st_stm32/stm32f0/soc.h:23:10: fatal error: stm32f0xx.h: No such file or directory
 #include <stm32f0xx.h>
          ^~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.

erwango, 2019-06-27:
You might need to run `west update`

Gagamama, 2019-06-27:
Same thing for me with <stm32f4xx.h>. It was previously in ext\hal\st\stm32cube... 
I don't use west

erwango, 2019-06-28:
Well, all these have moved to `hal_smt32` zephyr repo (https://github.com/zephyrproject-rtos/hal_stm32.git).
So, in order to build, you'll need to get packages from there manually if not using `west`.

ioannisg, 2019-06-30:
@maweicoltsmart (or @Gagamama ) can you follow the procedure suggested by @erwango and report whether this is still an issue?

kadamski, 2019-07-01:
I'm trying to follow the getting-started tutorial and I did:
```
$ west init zephyr_test2
$ cd zephyr_test2
$ west update
$ export ZEPHYR_TOOLCHAIN_VARIANT=cross-compile
$ export CROSS_COMPILE=/usr/bin/arm-none-eabi-
$ west build -b nucleo_f030r8 zephyr/samples/hello_world
[8/109] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
FAILED: zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj 
/usr/bin/arm-none-eabi-gcc -DBUILD_VERSION=zephyr-v1.14.0-1934-gb7d4b0057e2c -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I/home/k/PROJECTS/zephyr_test/zephyr/kernel/include -I/home/k/PROJECTS/zephyr_test/zephyr/arch/arm/include -I/home/k/PROJECTS/zephyr_test/zephyr/include -I/home/k/PROJECTS/zephyr_test/zephyr/include/drivers -Izephyr/include/generated -I/home/k/PROJECTS/zephyr_test/zephyr/soc/arm/st_stm32/stm32f0 -I/home/k/PROJECTS/zephyr_test/zephyr/lib/libc/minimal/include -I/home/k/PROJECTS/zephyr_test/zephyr/drivers -I/home/k/PROJECTS/zephyr_test/zephyr/ext/hal/cmsis/Include -isystem /usr/lib/gcc/arm-none-eabi/7.4.0/include -isystem /usr/lib/gcc/arm-none-eabi/7.4.0/include-fixed -Os -nostdinc --imacros=/home/k/PROJECTS/zephyr_test2/build/zephyr/include/generated/autoconf.h --imacros=/home/k/PROJECTS/zephyr_test/zephyr/include/toolchain/zephyr_stdint.h -ffreestanding -fno-common -g -mthumb -mcpu=cortex-m0 -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -ffunction-sections -fdata-sections -mabi=aapcs -march=armv6s-m -std=c99 -MD -MT zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj   -c /home/k/PROJECTS/zephyr_test/zephyr/arch/arm/core/offsets/offsets.c
In file included from /home/k/PROJECTS/zephyr_test/zephyr/include/arch/arm/cortex_m/cmsis.h:17:0,
                 from /home/k/PROJECTS/zephyr_test/zephyr/arch/arm/include/cortex_m/stack.h:27,
                 from /home/k/PROJECTS/zephyr_test/zephyr/arch/arm/include/kernel_arch_data.h:53,
                 from /home/k/PROJECTS/zephyr_test/zephyr/kernel/include/kernel_structs.h:68,
                 from /home/k/PROJECTS/zephyr_test/zephyr/arch/arm/core/offsets/offsets.c:26:
/home/k/PROJECTS/zephyr_test/zephyr/soc/arm/st_stm32/stm32f0/soc.h:23:10: fatal error: stm32f0xx.h: No such file or directory
 #include <stm32f0xx.h>
          ^~~~~~~~~~~~~
compilation terminated.
```

The stm32f0xx.h is there:
```
$ find -name "stm32f0xx.h"
./modules/hal/stm32/stm32cube/stm32f0xx/soc/stm32f0xx.h
```
But as can be seen, it is not added as an include path.

The same happens for any other stm32 board (stm32f4_disco, stm32_min_dev_blue). Isn't this the same issue? Or maybe I'm doing something wrong, actually?

erwango, 2019-07-01:
Can you run `west list`?

kadamski, 2019-07-01:
@erwango:
```
$ west list
zephyr         zephyr             (not set)     (not set) (cloned)
ci-tools       tools/ci-tools     672cb7623b89a41370f95881290a97cccdc35cb0 https://github.com/zephyrproject-rtos/ci-tools (cloned)
esp-idf        modules/hal/esp-idf 6835bfc741bf15e98fb7971293913f770df6081f https://github.com/zephyrproject-rtos/esp-idf (cloned)
fatfs          modules/fs/fatfs   df96914dd989907f3a5de4cb95b116c2f305820d https://github.com/zephyrproject-rtos/fatfs (cloned)
hal_qmsi       modules/hal/qmsi   9195fe6f97e4f7f25a3fc9e5a515f1b7af13762c https://github.com/zephyrproject-rtos/hal_qmsi (cloned)
hal_cypress    modules/hal/cypress a12d92816a53a521d79cefcf5c38b9dc8a4fed6e https://github.com/zephyrproject-rtos/hal_cypress (cloned)
hal_silabs     modules/hal/silabs 9151e614c23997074acd1096a3e8a9e5c255d5b9 https://github.com/zephyrproject-rtos/hal_silabs (cloned)
hal_st         modules/hal/st     0ec40aed8087f26bd9ac1b70fb5a6c326a6451aa https://github.com/zephyrproject-rtos/hal_st (cloned)
hal_stm32      modules/hal/stm32  8066e1d2477860a106e3c2e6ca841b4abbfbaad2 https://github.com/zephyrproject-rtos/hal_stm32 (cloned)
libmetal       modules/hal/libmetal 45e630d6152824f807d3f919958605c4626cbdff https://github.com/zephyrproject-rtos/libmetal (cloned)
mbedtls        modules/crypto/mbedtls ca32746072ce3381f1c9ae46ba6cd34c69f8c0ee https://github.com/zephyrproject-rtos/mbedtls (cloned)
mcumgr         modules/lib/mcumgr 84934959d2d1722a23b7e7e200191ae4a6f96168 https://github.com/zephyrproject-rtos/mcumgr (cloned)
net-tools      tools/net-tools    30b7efa827b04d2e47840716b0372737fe7d6c92 https://github.com/zephyrproject-rtos/net-tools (cloned)
nffs           modules/fs/nffs    bc62a2fa9d98ddb5d633c932ea199bc68e10f194 https://github.com/zephyrproject-rtos/nffs (cloned)
open-amp       modules/lib/open-amp 9b591b289e1f37339bd038b5a1f0e6c8ad39c63a https://github.com/zephyrproject-rtos/open-amp (cloned)
segger         modules/debug/segger 6fcf61606d6012d2c44129edc033f59331e268bc https://github.com/zephyrproject-rtos/segger (cloned)
tinycbor       modules/lib/tinycbor 31ae89e4b768612722620cb6cb173a0de4a19cc9 https://github.com/zephyrproject-rtos/tinycbor (cloned)
```

kadamski, 2019-07-01:
Sorry, turned out to be a noob mistake. Seeing this issue I though there are others who have it as well but in practice I had environment problem.

After some tracing I found out that find_program() in the zephyr/cmake/host-tools.cmake failed to find "west" binary, even though I was actually running via west and apparently that caused zephyr_modules.txt not being created.

erwango, 2019-07-01:
@kadamski , thanks for feedback. Can we close the issue then?

Also, if you think logging could be enhanced in west, do not hestistate to submit an issue in https://github.com/zephyrproject-rtos/west

kadamski, 2019-07-01:
@erwango I was not the author of the issue so I'm not sure, for me all seems to be working now.

maweicoltsmart, 2019-07-01:
it's ok now, thanks all of you!

closed 2019-07-01 by maweicoltsmart