zephyrproject-rtos/zephyr #17114: /usb/device/usb_dc_stm32.c

StefJar, 2019-06-27:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

erwango, 2019-06-27:
@StefJar, can you provide target on which this was spotted? 

Txs

StefJar, 2019-06-27:
I am using an board based on the STM32F412RG

Think you can use the _nucleo_f412zg_ for testing.

In case you want to rebuild my config:

the dts is based on _st/f4/stm32f412Xg.dtsi_
and I activate the usb via
```
&usbotg_fs {
	status = "ok";
};
```
the defconfig looks like
```
# usb
CONFIG_USB=y
CONFIG_USB_DC_STM32=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_VID=0x2FE1
CONFIG_USB_DEVICE_PID=0x69
CONFIG_USB_DEVICE_MANUFACTURER="xxx"
CONFIG_USB_DEVICE_PRODUCT="xxx"
CONFIG_USB_CDC_ACM=y
CONFIG_USB_VBUS_GPIO=y
CONFIG_USB_VBUS_GPIO_DEV_NAME="GPIOA"
CONFIG_USB_VBUS_GPIO_PIN_NUM=9
```
thx for checking

jfischer-phytec-iot, 2019-06-27:
@erwango e.g. stm32f4_disco

This `low-speed` option is superfluous in my opinion. Workaround:

```diff
diff --git a/drivers/usb/device/usb_dc_stm32.c b/drivers/usb/device/usb_dc_stm32.c                 
index b16f6ab7a3..89c0292965 100644                                                                     
--- a/drivers/usb/device/usb_dc_stm32.c                                   
+++ b/drivers/usb/device/usb_dc_stm32.c                      
@@ -322,8 +322,6 @@ static u32_t usb_dc_stm32_get_maximum_speed(void)                                   
 #else                                                                                                  
                speed = USB_OTG_SPEED_FULL;                                                             
 #endif /* DT_COMPAT_ST_STM32_USBPHYC && DT_USB_HS_BASE_ADDRESS */                                      
-       } else if (!strncmp(DT_USB_MAXIMUM_SPEED, "low-speed", 9)) {       
-               speed = USB_OTG_SPEED_LOW;                                                              
        } else {                                                                                        
                LOG_DBG("Unsupported maximum speed defined in device tree. "                            
                        "USB controller will default to its maximum HW "   
```

StefJar, 2019-06-27:
this fixes the error

erwango, 2019-06-27:
@jfischer-phytec-iot, indeed, I'm checking internally this is the case for all STM32 references (and code could be removed) or if it's supported in some samples (then we need to protect these lines).

PR #17120 (2019-06-28): drivers/usb/device: stm32: Remove reference to unsupported low speed
Low speed isn't supported in device mode for any of the STM32
references.
Remove the code that refer to it.

Fixes #17114
commit 6b40394d (2019-06-28): drivers/usb/device: stm32: Remove reference to unsupported low speed

Low speed isn't supported in device mode for any of the STM32
references.
Remove the code that refer to it.

Fixes #17114
closed 2019-06-28 by nashif