zephyrproject-rtos/zephyr #17135: Cannot flash LWM2M example for ESP32

mutrack, 2019-06-28:
I am trying to run the LWM2M sample on an ESP32 board. It builds fine but fails to flash with an "Invalid segment count 17 (max 16)" error.
Other samples e.g. hello_world, mqtt build and flash without problem.

Build environment : Ubuntu 4.18.0
ESP-IDF Version : v4.0-dev-1018-gbbafd5066
Zephyr version: 1.14.99

Full output:

```
steve@ubuntu:~/zephyrproject/zephyr$ cmake -Bbuild -GNinja -DBOARD=esp32 -DCONF_FILE=prj.conf samples/net/lwm2m_client
Zephyr version: 1.14.99
-- Selected BOARD esp32
-- Found west: /home/steve/.local/bin/west (found suitable version "0.5.7", minimum required is "0.5.6")
-- Loading /home/steve/zephyrproject/zephyr/boards/xtensa/esp32/esp32.dts as base
-- Overlaying /home/steve/zephyrproject/zephyr/dts/common/common.dts
Parsing Kconfig tree in /home/steve/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/steve/zephyrproject/zephyr/build/zephyr/.config'
No change to '/home/steve/zephyrproject/zephyr/build/zephyr/.config'
-- Cache files will be written to: /home/steve/.cache/zephyr
CMake Warning at /home/steve/zephyrproject/zephyr/boards/common/esp32.board.cmake:7 (message):
  Setting ESP_IDF_PATH in the environment is deprecated.  Use cmake
  -DESP_IDF_PATH=...  instead.
Call Stack (most recent call first):
  /home/steve/zephyrproject/zephyr/boards/xtensa/esp32/board.cmake:3 (include)
  /home/steve/zephyrproject/zephyr/cmake/app/boilerplate.cmake:463 (include)
  CMakeLists.txt:4 (include)

Including module: esp-idf in path: /home/steve/zephyrproject/modules/hal/esp-idf/zephyr
Including module: fatfs in path: /home/steve/zephyrproject/modules/fs/fatfs
Including module: qmsi in path: /home/steve/zephyrproject/modules/hal/qmsi
Including module: cypress in path: /home/steve/zephyrproject/modules/hal/cypress
Including module: silabs in path: /home/steve/zephyrproject/modules/hal/silabs
Including module: st in path: /home/steve/zephyrproject/modules/hal/st
Including module: stm32 in path: /home/steve/zephyrproject/modules/hal/stm32
Including module: libmetal in path: /home/steve/zephyrproject/modules/hal/libmetal
Including module: mbedtls in path: /home/steve/zephyrproject/modules/crypto/mbedtls
Including module: mcumgr in path: /home/steve/zephyrproject/modules/lib/mcumgr
Including module: nffs in path: /home/steve/zephyrproject/modules/fs/nffs
Including module: open-amp in path: /home/steve/zephyrproject/modules/lib/open-amp
Including module: segger in path: /home/steve/zephyrproject/modules/debug/segger
Including module: tinycbor in path: /home/steve/zephyrproject/modules/lib/tinycbor
-- Configuring done
-- Generating done
-- Build files have been written to: /home/steve/zephyrproject/zephyr/build
steve@ubuntu:~/zephyrproject/zephyr$ ninja -Cbuild flash
ninja: Entering directory `build'
[121/126] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
     iram0_0_seg:      121918 B       128 KB     93.02%
     iram0_2_seg:          0 GB      3264 KB      0.00%
     dram0_0_seg:       64176 B       320 KB     19.58%
     drom0_0_seg:          0 GB         8 MB      0.00%
    rtc_iram_seg:          0 GB         8 KB      0.00%
    rtc_slow_seg:          0 GB         4 KB      0.00%
        IDT_LIST:          72 B         8 KB      0.88%
[125/126] Flashing esp32
-- west flash: using runner esp32
-- runners.esp32: Converting ELF to BIN
esptool.py v2.7-dev

A fatal error occurred: Invalid segment count 17 (max 16). Usually this indicates a linker script problem.
ERROR: command exited with status 2: /home/steve/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 elf2image /home/steve/zephyrproject/zephyr/build/zephyr/zephyr.elf
run as "west -v flash --skip-rebuild" for a stack trace
FAILED: zephyr/cmake/flash/CMakeFiles/flash
cd /home/steve/zephyrproject/zephyr/build && /home/steve/.bin/cmake/cmake-3.13.1-Linux-x86_64/bin/cmake -E env /home/steve/.local/bin/west flash --skip-rebuild
ninja: build stopped: subcommand failed.

```

fake991, 2019-10-15:
has anyone managed to make it work?

Deliver88, 2019-11-14:
@fake991 , @nashif 
I faced this issue too, when tried running the WiFi example at samples/net/wifi. The error is caused from the esp-idf. At esp-idf/components/esptool_py/esptool you have to fix it in the esptool.phy, there you will find the section
def verify(self):
        if len(self.segments) > 16:
            raise FatalError('Invalid segment count %d (max 16). Usually this indicates a linker script problem.' % len(self.segments))

Changing it to a nother value (e.g. to 17 or what ever you like) will fix the error!

nashif, 2020-03-11:
duplicate of #20980

closed 2020-03-11 by nashif