zephyrproject-rtos/zephyr #17147: UARTE device has no API when run on nrf52810

pdunaj, 2019-06-28:
**Describe the bug**
I tried to enable console via UARTE on nrf52810 but SOC crashes at early stage.
Quick investigation revealed two things.
* First is that `uart_console_init` does not check if get binding returned anything and so crash occurs much later when console is actually used.
* Second - get binding returns NULL even if device is present... I checked that it is NULL as API is NULL. This is strange as API is statically added when device is defined (see `drivers/serial/uart_nrfx_uarte.c`).

**To Reproduce**
Steps to reproduce the behavior:
1. Enable console over uarte on nrf52810
2. start zephyr
3. observe crash when zephyr os tag is printed

**Expected behavior**
No crash, everything works

**Impact**
Showstopper

**Screenshots or console output**
```
1: info->driver_api = (const void *) 0x20bf8 <_k32src_clock_control_api>
2: info->config->name = 0x224a5 "CLOCK_32K"
1: info->driver_api = (const void *) 0x20c04 <_m16src_clock_control_api>
2: info->config->name = 0x224af "CLOCK_16M"
1: info->driver_api = (const void *) 0x0 <z_finalize_fd>
2: info->config->name = 0x22412 "UART_0"
1: info->driver_api = (const void *) 0x21084 <entropy_nrf5_api_funcs>
2: info->config->name = 0x22211 "ENTROPY_0"
```

**Environment (please complete the following information):**
master

**Additional context**
N/A

pdunaj, 2019-06-28:
btw - why device information is stored in RAM not in ROM?

anangl, 2019-07-01:
> **To Reproduce**
> Steps to reproduce the behavior:
> 
> 1. Enable console over uarte on nrf52810

Could you provide some more details regarding the configuration needed to reproduce the issue? The only board in the Zephyr source tree that currently uses the nRF52810 SoC is `nrf52810_pca10040` and it has the console over UARTE enabled by default. For this board the console (in the hello_world sample, for instance) seems to be working fine.

pdunaj, 2019-07-03:
Nothing special was set. I enabled
`SERIAL`, `UART_NRFX`, `UART_0_NRF_UARTE` , `UART_0_NRF_FLOW_CONTROL`
and `CONSOLE`, `UART_CONSOLE`.

I tried again and it seems that the problem is caused by `UART_0_NRF_FLOW_CONTROL`. That is, if the option is not enabled I application works, if it is enabled UART api is not specified and I get crash as described.

pdunaj, 2019-07-03:
Seems API is set to NULL as `uarte_0_init` fails and `kernel/device.c` has a code that will set API to NULL in such a case (great idea btw Zepyr to set API to NULL instead of doing anything else that would point to the actual error).

pdunaj, 2019-07-03:
I found the problem is in the dts file as instead of cts I have rts defined twice...

closed 2019-07-03 by pdunaj