zephyrproject-rtos/zephyr #17167: ARC crash with spinning user thread

andrewboie, 2019-06-29:
**Describe the bug**
I have found that the ARC nsim_em and nsim_sem targets crash out if a preemptible user thread hogs the CPU.

**To Reproduce**
Run samples/hello_world in this branch:
https://github.com/andrewboie/zephyr/commit/80d2a939feea719cb181b165ea810d8d9709c7e7

**Expected behavior**
The expected output is:

```
***** Booting Zephyr OS build zephyr-v1.14.0-1925-gca3549d9f540 *****
Hello World! nsim
done
```

Instead I get:

```
***** Booting Zephyr OS build zephyr-v1.14.0-1925-gca3549d9f540 *****
Hello World! nsim
***** Exception vector: 0x0, cause code: 0x0, parameter 0x0
Address 0x0
Reset
Current thread ID = 0x80000100
Faulting instruction address = 0x15b4
Fatal fault in essential thread! Spinning...
```

Now if I add a k_yield() statement in the child thread's spinning loop, or start the child thread without K_USER(), eveything works normally.

I seem to get random, other exceptions depending on the application. When I originally found this in another project I was working on, I got:

```
***** Exception vector: 0x6, cause code: 0x1, parameter 0x2
Address 0x877c
EV_ProtV: Memory read protection violation: stack checking scheme
***** Stack Check Fail! *****
Current thread ID = 0x800006b8
Fatal fault in thread 0x800006b8! Aborting.
```

**Impact**
Showstopper

andrewboie, 2019-06-29:
@vonhust this does not seem directly related to your recent switch changes, even if I revert that patch I still get a crash, with the console log:

```
***** Exception vector: 0x6, cause code: 0x2, parameter 0x4
Address 0xfffffffc
EV_ProtV: Memory write protection violation: MPU
Current thread ID = 0x80000100
Faulting instruction address = 0x4290
Fatal fault in essential thread! Spinning...
```

andrewboie, 2019-06-29:
This can also be reproduced with:

https://github.com/zephyrproject-rtos/zephyr/pull/17071

If you comment out the `k_yield()` call in thread_name_entry() and run tests/kernel/threads/thread_apis test case.

vonhust, 2019-07-01:
@andrewboie Thanks, it's a bug. I make a fix in #17184 and re-run you provided tests case. The result is ok

PR #17184 (2019-07-03): arch: arc: fix the bug caused by hardware sp switch in interrupt
* if thread switchs in interrupt, the target sp must be in
thread's kernel stack, no need to do hardware sp switch

Fixes #17167
closed 2019-07-03 by andrewboie