zephyrproject-rtos/zephyr #17171: Insufficient code coverage for lib/os/fdtable.c

andrewboie, 2019-06-30:
**Is your enhancement proposal related to a problem? Please describe.**
Code coverage reports indicate we have a lack of unit tests for the code in lib/os/fdtable.c

Lines: 37/96 38.5%
Functions: 9/18 50%
Branches: 11/54 20.4%

At bare minimum we need 100% function coverage and greater than 90% line coverage, ideally 100% across the board.

The command line I am using for gathering coverage data is:

```
sanitycheck -n -p qemu_x86 -p mps2_an385 --coverage --coverage-platform qemu_x86 --coverage-platform mps2_an385 --gcov-tool /opt/zephyr-sdk/i586-zephyr-elf/bin/i586-zephyr-elf-gcov -T tests/kernel/ -T tests/lib/ -T tests/posix/ -j12 -x=CONFIG_ASSERT=n
```

Fixes should be backported to the v1.14 branch

andrewboie, 2019-07-02:
It doesn't look like we have any dedicated unit tests for this code at all, the number we have are only because the APIs are used by some fs tests.

Need to write some tests under tests/lib/fdtable to prove that these APIs work. There may be bugs lurking.

pfalcon, 2020-05-18:
A ticket for particular sub-issue was created: https://github.com/zephyrproject-rtos/zephyr/issues/25407, to hopefully be addressed for 2.3.

nashif, 2022-11-17:
seems to be addressed.

closed 2022-11-17 by nashif