zephyrproject-rtos/zephyr #17182: "tests/subsys/usb/device/" fails on reel_board.

LeiW000, 2019-07-01:
**Describe the bug**
"tests/subsys/usb/device/" fails on reel_board.

**To Reproduce**
1. mkdir build; cd build
2. cmake -DBOARD=reel_board
3. make
4. make flash

**Screenshots or console output**

Booting Zephyr OS build zephyr-v1.14.0-1797-gb53ce02afdb8 
Running test suite test_device
===================================================================
starting test - test_usb_dc_api_invalid

    Assertion failed at ZEPHYR_BASE/tests/subsys/usb/device/src/main.c:157: test_usb_dc_api_invalid: (usb_dc_ep_set_callback(INVALID_EP, NULL) equal to TC_PASS)
usb_dc_ep_set_callback(INVALID_EP, NULL)
FAIL - test_usb_dc_api_invalid
===================================================================
starting test - test_usb_dc_api

    Assertion failed at ZEPHYR_BASE/tests/subsys/usb/device/src/main.c:116: test_usb_dc_api: (usb_dc_set_address(0x01) not equal to TC_PASS)
usb_dc_set_address(0x01)
FAIL - test_usb_dc_api
===================================================================
starting test - test_usb_dc_api_read_write
PASS - test_usb_dc_api_read_write
===================================================================
starting test - test_usb_dc_api_invalid

    Assertion failed at ZEPHYR_BASE/tests/subsys/usb/device/src/main.c:127: test_usb_dc_api_invalid: (usb_dc_ep_set_stall(INVALID_EP) equal to TC_PASS)
usb_dc_ep_set_stall(INVALID_EP)
FAIL - test_usb_dc_api_invalid
===================================================================
starting test - test_usb_deconfig
PASS - test_usb_deconfig
===================================================================
starting test - test_usb_disable
ASSERTION FAIL [m_drv_state == NRFX_DRV_STATE_POWERED_ON] @ ZEPHYR_BASE/ext/hal/nordic/nrfx/drivers/src/nrfx_usbd.c:1880
[00:00:00.005,828] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB detected
--- 29 messages dropped ---
[00:00:00.005,828] <dbg> usb_nrfx.hf_clock_enable: HF clock start success (-68)
[00:00:00.006,011] <dbg> usb_device.composite_setup_ep_cb: set cb, ep: 0x81
[00:00:00.006,011] <dbg> usb_nrfx.usb_dc_ep_enable: EP enable: 0x00
[00:00:00.006,042] <dbg> usb_nrfx.usb_dc_ep_enable: EP enable: 0x80
[00:00:00.006,896] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB Powered
[00:00:00.010,009] <dbg> usb_nrfx.hf_clock_enable: HF clock stop success (0)
[00:00:00.010,009] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB Removed
[00:00:00.036,163] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB detected
[00:00:00.036,193] <dbg> usb_nrfx.hf_clock_enable: HF clock start success (-68)
[00:00:00.037,261] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB Powered
[00:00:00.040,344] <dbg> usb_nrfx.hf_clock_enable: HF clock stop success (0)
[00:00:00.040,344] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB Removed
[00:00:00.062,530] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB detected
[00:00:00.062,561] <dbg> usb_nrfx.hf_clock_enable: HF clock start success (-68)
[00:00:00.063,598] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB Powered
[00:00:00.066,711] <dbg> usb_nrfx.hf_clock_enable: HF clock stop success (0)
[00:00:00.066,711] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB Removed
[00:00:00.074,584] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x20, maxlen 1
[00:00:00.074,615] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x20, len 1
[00:00:00.086,059] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB detected
[00:00:00.086,090] <dbg> usb_nrfx.hf_clock_enable: HF clock start success (-68)
[00:00:00.087,158] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB Powered
[00:00:00.087,310] <dbg> usb_nrfx.usb_dc_ep_set_stall: STALL on EP 0x20
[00:00:00.103,546] <dbg> usb_nrfx.hf_clock_enable: HF clock stop success (0)
[00:00:00.103,576] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB Removed
[00:00:00.114,776] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB detected
[00:00:00.114,807] <dbg> usb_nrfx.hf_clock_enable: HF clock start success (-68)
[00:00:00.115,875] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB Powered
[00:00:00.118,957] <dbg> usb_nrfx.hf_clock_enable: HF clock stop success (0)
[00:00:00.118,988] <dbg> usb_nrfx.usbd_work_process_pwr_events: USB Removed
Kernel Panic!
Current thread ID = 0x200016d4 (unknown)
Faulting instruction address = 0xb686

jfischer-phytec-iot, 2019-07-01:
It does not depend on the reel board, similar behavior on nrf52840_pca10056. There are some incompatibilities between usb_dc_nrfx driver and device test. I will look at it.

closed 2019-07-04 by ioannisg

LeiW000, 2019-07-08:
Thanks for fixing it.

PR #17236 (2019-07-04): drivers: usb_dc_nrfx: fix usb device test
tests: usb: do not set device address
Forced attempt to set the device address is unpredictable
and also should not be done during testing.

drivers/usb_dc_nrfx: 
Do not use NRF_USBD_EP_NR_GET in ep_is_valid because
it masks the higher nibble. Although this behavior is valid
to get an index from an endpoint, it is not suitable to check
if the address is incorrect, such as: 0x11.

Check the device state before nrfx_usbd_disable and
nrfx_usbd_uninit. Otherwise it may lead to an ASSERT
inside the HAL driver.

Validate pointer argument in usb_dc_ep_is_stal [... cut]