zephyrproject-rtos/zephyr #17190: net-mgmt should pass info element size to callback

jukkar, 2019-07-01:
**Describe the bug**
The information element size should be passed to the user defined callback. The system works atm if the info element can be determined from the network management event type. Sometimes this is not possible so the application needs to know the info element size.

**Impact**
The application might not be able to utilize the information element tied to the network management event.

This issue is created so that we can backport few commits from PR #16743 to 1.14

tbursztyka, 2019-07-01:
"Sometimes this is not possible so the application" Currently it is always possible. Even for the socket layer.

jukkar, 2019-07-03:
The PR was merged -> closing.

PR #17192 (2019-07-02): [Backport v1.14-branch] net-mgmt event needs length of info element
This PR backports relevant parts from #16743 to 1.14

Fixes #17190
closed 2019-07-03 by jukkar