zephyrproject-rtos/zephyr #17239: crash when running "sanitycheck -j12"

andrewboie, 2019-07-02:
```
$ time sanitycheck -j12
JOBS: 12
Selecting default platforms per test case
Building testcase defconfigs...
Filtering test cases...
2449 tests selected, 113686 tests discarded due to filters
Exception in thread qemu_xtensa/tests/lib/rbtree/libraries.data_structures.rbtree:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 865, in run
  File "/home/apboie/projects/zephyr1/zephyr/scripts/sanitycheck", line 762, in _thread
OSError: [Errno 24] Too many open files: '/home/apboie/projects/zephyr1/zephyr/sanity-out/qemu_xtensa/tests/lib/rbtree/libraries.data_structures.rbtree/qemu-fifo.in'

libgcc_s.so.1 must be installed for pthread_cancel to work
Aborted (core dumped)
```

Master branch, I'm running Ubuntu 19.04.
Reproduced on another machine running Fedora 30

andrewboie, 2019-07-02:
@nashif are you seeing this

nashif, 2019-07-02:
> @nashif are you seeing this

yes, we are hitting the file limit, we launch too many thrreads in sanitycheck which open too many files, try to increase limit: ulimit -n 4096.

I am already looking into this.

marc-hb, 2019-07-02:
I run multiple large sanitycheck runs daily and I never encountered this. I suspect it's because this local hack has been an _accidental workaround_ (**not** a fix cause JOBS != Python threads):
```diff
                                  "last_sanity.xml")
 RELEASE_DATA = os.path.join(ZEPHYR_BASE, "scripts", "sanity_chk",
                             "sanity_last_release.csv")
-JOBS = multiprocessing.cpu_count() * 2
+JOBS = min(int(multiprocessing.cpu_count() * 1.2), 40)
 
 if os.isatty(sys.stdout.fileno()):
     TERMINAL = True
```
> we launch too many thrreads in sanitycheck

For anyone interested: "top" doesn't show threads which hides the issue. "htop" shows 2000+ Python threads with default sanitycheck parameters.

andrewboie, 2019-07-03:
For me the --jobs option doesn't affect it, I was seeing this with -j12.
Think this is a function of how many tests are selected, in the brief look at the code I took, it appears that it tries to open a QEMU pipe for every single emulated test, all at once.

andrewboie, 2019-07-03:
I posted a workaround patch which has sanitycheck automatically increase the file limit.

If we agree to merge it, I think we can lower this to 'medium' and give us some breathing room to develop a proper fix.

nashif, 2019-08-20:
lowering priority as we have a workaround in place and a long term solution in the works.

thst-nordic, 2019-08-21:
I have a [PR](https://github.com/zephyrproject-rtos/zephyr/pull/16619) to run `sanitycheck` on windows, but the resource package only supports posix systems.  

I'll wrap this temporary change with `if os.name == 'posix'`

However, I am curious what the long term solution is.

marc-hb, 2019-08-21:
> I have a PR to run sanitycheck on windows

\o/

> However, I am curious what the long term solution is.

The longer term solution is an extensive rewrite of the job management part of the code. @nashif has a prototype somewhere, check his github you may find it there.

nashif, 2019-10-18:
fixed in new sanitycheck

closed 2019-10-18 by nashif