zephyrproject-rtos/zephyr #17266: CDC_ACM USB not recognized by windows as serial port

pfvogel, 2019-07-02:
**Describe the bug**
The USB CDC-ACM port is usable under Linux but not usable under Windows 10 unless it is built as a composite device.

**To Reproduce**
Using the nrf52840 Development Kit board, with a USB cable plugged into the JLink USB connector for programming.

Steps to reproduce the behavior:
1. cd samples/subsys/usb/cdc_acm
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=nrf52840_pca10056 ..
3. ninja
4. ninja flash
5. Plug USB cable into "nrf USB" connector.

**Expected behavior**
Under Linux, /dev/ttyACM1 is created when the second USB is plugged in and is usable as a serial port with a serial terminal such as minicom.  Under Windows 10, a COM port is created but has an error and is not usable.
```
This device cannot start. (Code 10)

A device which does not exist was specified.
```

**Impact**
The serial port is not accessible to Windows 10 applications.  A workaround is to add the define
CONFIG_USB_COMPOSITE_DEVICE=y
to the _prj.conf_ file and rebuild.  This results in a CDC ACM port that is usable under both Linux and Windows 10.

**Screenshots or console output**
```
Screenshot of Windows 10 port with error
````
[image]

**Environment (please complete the following information):**
 - OS: Built under Ubuntu 18.04
 - Toolchain: gcc-arm-none-eabi-7-2018-q2-update
 - Commit SHA or Version used: d8c5c9dcf1

**Additional context**
None

jfischer-phytec-iot, 2019-07-03:
Windows 10?, weird. Could you please try `CONFIG_CDC_ACM_IAD=y`?

pfvogel, 2019-07-03:
Just tried it with CONFIG_CDC_ACM_IAD=y instead of CONFIG_USB_COMPOSITE_DEVICE=y and it works under both Linux and Windows 10.
Thanks for the suggestion!

jfischer-phytec-iot, 2019-07-03:
CONFIG_CDC_ACM_IAD was introduced as workaround for Windows 7, as I remember, there should be no issues on Windows10. @masz-nordic any thoughts?

masz-nordic, 2019-07-05:
I was able to reproduce this on Windows 10 1809 with current master. I will try bisecting and testing on various OS versions when I have some time.

@pfvogel What Windows 10 version are you using?

pfvogel, 2019-07-05:
@masz-nordic Windows 10 Home 1809

aolowin, 2019-07-11:
I have seen this on Windows 10 Pro 1803.
CONFIG_CDC_ACM_IAD=y works fine.

closed 2019-08-29 by carlescufi

traavnik, 2019-09-05:
I have experienced the same problem using v1.14 (LTS) of Zephy. Sadly "CONFIG_CDC_ACM_IAD" doesn't seem to be supported in that version. 
Are there any plans to backport relevant fixes?

PR #18722 (2019-08-29): usb: mitigate the check of request buffer length and enable CONFIG_CDC_ACM_IAD by default 
9ba269f580bb754ef566129eab262106c3afacac  has caused a few surprises, so a few more urgent fixes:

Mitigate the check of the request buffer length because
the host may not know the real length for some responds.

Increase the length of the request buffer.
Current value is not enough for an average
device descriptor.

Enable CONFIG_CDC_ACM_IAD by default so that
CDC ACM class works out of the box on Windows OS.

Fixes #17266