zephyrproject-rtos/zephyr #17280: How to use UART1 for nrf52_pca10040

LengedZhao, 2019-07-03:
**Describe the bug**
I want to use uart1 for communiction while uart0 is using for debuger print. So after referencing "Documenting enablement of UART1 support for nrf52840_pca10056"(https://github.com/zephyrproject-rtos/zephyr/pull/13077/files) , I try to add some code to my nrf52_pca10040.overlay such as "&uart1 {
	compatible = "nordic,nrf-uarte";
	current-speed = <9600>;
	status = "ok";
	tx-pin = <12>;
	rx-pin = <11>;
};".
But when I run " cmake -GNinja -DBOARD=nrf52_pca10040 ..", An error is report "Error: nrf52_pca10040.dts.pre.tmp:470.1-7 Label or path uart1 not found
FATAL ERROR: Syntax error parsing input tree
CMake Error at C:/Users/zhao/zephyr-mesh/cmake/dts.cmake:124 (message):
  command failed with return code: 1".

So how to use uart1 for nrf52_pca10040 ? Or ZEPHYR just can't suport uart1 for nrf52_pca10040.

ioannisg, 2019-07-03:
@anangl could you please answer this?

anangl, 2019-07-03:
> So how to use uart1 for nrf52_pca10040 ? Or ZEPHYR just can't suport uart1 for nrf52_pca10040.

The board (nrf52_pca10040) is equipped with the nRF52832 SoC which has only one UART. Thus, what you are trying to achieve is simply not possible on this board, since there is no corresponding hardware there.

closed 2019-07-03 by anangl