zephyrproject-rtos/zephyr #17281: sanitycheck error on mimxrt1050_evk samples/gui/lvgl/sample.gui.lvgl with no network connection

andrewboie, 2019-07-03:
```
$ cat sanity-out/mimxrt1050_evk/samples/gui/lvgl/sample.gui.lvgl/build.log
Zephyr version: 1.14.99
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.7.3", minimum required is "3.4") 
-- Selected BOARD mimxrt1050_evk
-- Found west: /home/apboie/.local/bin/west (found suitable version "0.5.7", minimum required is "0.5.6")
-- Loading /home/apboie/projects/zephyr1/zephyr/boards/arm/mimxrt1050_evk/mimxrt1050_evk.dts as base
-- Overlaying /home/apboie/projects/zephyr1/zephyr/dts/common/common.dts
Parsing Kconfig tree in /home/apboie/projects/zephyr1/zephyr/Kconfig
Loaded configuration '/home/apboie/projects/zephyr1/zephyr/boards/arm/mimxrt1050_evk/mimxrt1050_evk_defconfig'
Merged configuration '/home/apboie/projects/zephyr1/zephyr/samples/gui/lvgl/prj.conf'
Merged configuration '/home/apboie/projects/zephyr1/zephyr/sanity-out/mimxrt1050_evk/samples/gui/lvgl/sample.gui.lvgl/overlay.conf'
Configuration saved to '/home/apboie/projects/zephyr1/zephyr/sanity-out/mimxrt1050_evk/samples/gui/lvgl/sample.gui.lvgl/zephyr/.config'
-- Cache files will be written to: /home/apboie/.cache/zephyr
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
make[1]: Entering directory '/home/apboie/projects/zephyr1/zephyr/sanity-out/mimxrt1050_evk/samples/gui/lvgl/sample.gui.lvgl/zephyr/ext_proj/Subbuild/lvgl'
make[2]: Entering directory '/home/apboie/projects/zephyr1/zephyr/sanity-out/mimxrt1050_evk/samples/gui/lvgl/sample.gui.lvgl/zephyr/ext_proj/Subbuild/lvgl'
make[3]: Entering directory '/home/apboie/projects/zephyr1/zephyr/sanity-out/mimxrt1050_evk/samples/gui/lvgl/sample.gui.lvgl/zephyr/ext_proj/Subbuild/lvgl'
Scanning dependencies of target lvgl-populate
make[3]: Leaving directory '/home/apboie/projects/zephyr1/zephyr/sanity-out/mimxrt1050_evk/samples/gui/lvgl/sample.gui.lvgl/zephyr/ext_proj/Subbuild/lvgl'
make[3]: Entering directory '/home/apboie/projects/zephyr1/zephyr/sanity-out/mimxrt1050_evk/samples/gui/lvgl/sample.gui.lvgl/zephyr/ext_proj/Subbuild/lvgl'
[ 11%] Creating directories for 'lvgl-populate'
[ 22%] Performing download step (git clone) for 'lvgl-populate'
Cloning into 'lvgl'...
fatal: unable to access 'https://github.com/littlevgl/lvgl.git/': Failed to connect to github.com port 443: Connection timed out
Cloning into 'lvgl'...
fatal: unable to access 'https://github.com/littlevgl/lvgl.git/': Failed to connect to github.com port 443: Connection timed out
Cloning into 'lvgl'...
fatal: unable to access 'https://github.com/littlevgl/lvgl.git/': Failed to connect to github.com port 443: Connection timed out
-- Had to git clone more than once:
          3 times.
CMake Error at /home/apboie/projects/zephyr1/zephyr/sanity-out/mimxrt1050_evk/samples/gui/lvgl/sample.gui.lvgl/zephyr/ext_proj/Subbuild/lvgl/lvgl-populate-prefix/tmp/lvgl-populate-gitclone.cmake:66 (message):
  Failed to clone repository: 'https://github.com/littlevgl/lvgl.git'

make[3]: *** [CMakeFiles/lvgl-populate.dir/build.make:91: lvgl-populate-prefix/src/lvgl-populate-stamp/lvgl-populate-download] Error 1
make[3]: Leaving directory '/home/apboie/projects/zephyr1/zephyr/sanity-out/mimxrt1050_evk/samples/gui/lvgl/sample.gui.lvgl/zephyr/ext_proj/Subbuild/lvgl'
make[2]: *** [CMakeFiles/Makefile2:73: CMakeFiles/lvgl-populate.dir/all] Error 2
make[2]: Leaving directory '/home/apboie/projects/zephyr1/zephyr/sanity-out/mimxrt1050_evk/samples/gui/lvgl/sample.gui.lvgl/zephyr/ext_proj/Subbuild/lvgl'
make[1]: *** [Makefile:84: all] Error 2
make[1]: Leaving directory '/home/apboie/projects/zephyr1/zephyr/sanity-out/mimxrt1050_evk/samples/gui/lvgl/sample.gui.lvgl/zephyr/ext_proj/Subbuild/lvgl'

CMake Error at /home/apboie/.local/lib/python3.7/site-packages/cmake/data/share/cmake-3.13/Modules/FetchContent.cmake:798 (message):
  Build step for lvgl failed: 2
Call Stack (most recent call first):
  /home/apboie/.local/lib/python3.7/site-packages/cmake/data/share/cmake-3.13/Modules/FetchContent.cmake:889 (__FetchContent_directPopulate)
  ../../../lib/gui/lvgl/CMakeLists.txt:24 (FetchContent_Populate)

-- Configuring incomplete, errors occurred!
See also "/home/apboie/projects/zephyr1/zephyr/sanity-out/mimxrt1050_evk/samples/gui/lvgl/sample.gui.lvgl/CMakeFiles/CMakeOutput.log".
See also "/home/apboie/projects/zephyr1/zephyr/sanity-out/mimxrt1050_evk/samples/gui/lvgl/sample.gui.lvgl/CMakeFiles/CMakeError.log".
```

The test appears to be trying to clone a git repository. I was offline when the error occurred. This extra project should be added to the west manifest instead.

vanwinkeljan, 2019-07-03:
This is a known limitation and the intention is to move lvgl to a dedicated west repo after the migration of ext is done.

vanwinkeljan, 2019-07-03:
@nashif, @carlescufi, @MaureenHelm  as already 2 people ran into issues with the lvgl download (this issue and https://github.com/zephyrproject-rtos/zephyr/pull/6826#issuecomment-494923867) would it be an option to create an empty repo `lvgl` in `zephyrproject-rtos`, then I could take care of migrating `lvgl` in to the west manifest?

nashif, 2019-07-03:
@vanwinkeljan sure, can you prepare a repo in your fork that would act as a module, I will then create it in the zephyr project and push it and make it ready for PRs and integration.

vanwinkeljan, 2019-07-03:
@nashif Create a branch (`west_module`) in my [lvgl fork](https://github.com/vanwinkeljan/lvgl/commits/west_module).

But would it not be better to directly fork the main lvgl repo to `zephyrproject-rtos` and that I send a PR to that repo?

(zephyr part can be found in draft pr #17320)

PR #17320 (2019-07-06): Move lvgl into a zephyr module
Updated  `lib/gui/lvgl/CMakeLists.txt` and `west.yml` so that lvgl can be moved in to a dedicated zephyr module.

Glue logic between zephyr and lvgl is kept in the main zephyr tree.

fixes #17281
closed 2019-07-06 by nashif