zephyrproject-rtos/zephyr #17284: unrecognized binary sections: ['_settings_handlers_area']

andrewboie, 2019-07-03:
```
FAILED: bbc_microbit/samples/bluetooth/mesh/sample.bluetooth.mesh has unrecognized binary sections: ['_settings_handlers_area']
FAILED: bbc_microbit/samples/bluetooth/mesh_demo/sample.bluetooth.mesh_demo has unrecognized binary sections: ['_settings_handlers_area']
FAILED: nrf51_pca10028/tests/bluetooth/mesh_shell/bluetooth.mesh_shell has unrecognized binary sections: ['_settings_handlers_area']
FAILED: nrf52840_pca10056/samples/boards/nrf52/mesh/onoff-app/sample.bluetooth.mesh.onoff has unrecognized binary sections: ['_settings_handlers_area']
FAILED: nrf52840_pca10056/samples/boards/nrf52/mesh/onoff_level_lighting_vnd_app/sample.bluetooth.mesh.onoff_level_lighting_vnd has unrecognized binary sections: ['_settings_handlers_area']
FAILED: nrf52840_pca10056/tests/bluetooth/shell/bluetooth.shell has unrecognized binary sections: ['_settings_handlers_area']
FAILED: qemu_cortex_m3/samples/bluetooth/peripheral/sample.bluetooth.peripheral has unrecognized binary sections: ['_settings_handlers_area']
FAILED: qemu_cortex_m3/samples/bluetooth/peripheral_dis/sample.bluetooth.peripheral_dis has unrecognized binary sections: ['_settings_handlers_area']
FAILED: qemu_cortex_m3/samples/bluetooth/peripheral_hids/sample.bluetooth.peripheral_hids has unrecognized binary sections: ['_settings_handlers_area']
FAILED: qemu_x86/samples/bluetooth/mesh/sample.bluetooth.mesh has unrecognized binary sections: ['_settings_handlers_area']
FAILED: qemu_x86/samples/bluetooth/mesh_demo/sample.bluetooth.mesh_demo has unrecognized binary sections: ['_settings_handlers_area']
FAILED: qemu_x86/samples/bluetooth/peripheral/sample.bluetooth.peripheral has unrecognized binary sections: ['_settings_handlers_area']
FAILED: qemu_x86/samples/bluetooth/peripheral_dis/sample.bluetooth.peripheral_dis has unrecognized binary sections: ['_settings_handlers_area']
FAILED: qemu_x86/samples/bluetooth/peripheral_hids/sample.bluetooth.peripheral_hids has unrecognized binary sections: ['_settings_handlers_area']
FAILED: qemu_x86/tests/bluetooth/mesh_shell/bluetooth.mesh_shell has unrecognized binary sections: ['_settings_handlers_area']
FAILED: qemu_x86/tests/bluetooth/shell/bluetooth.shell has unrecognized binary sections: ['_settings_handlers_area']
FAILED: reel_board/samples/boards/reel_board/mesh_badge/sample.board.reel_board.mesh_badge has unrecognized binary sections: ['_settings_handlers_area']
```

Seeing this in sanitycheck output on master branch.

jhedberg, 2019-07-03:
@Vudentz is this related to the “iterable” stuff?

jhedberg, 2019-07-03:
It seems like this area was introduced by commit c20ff1150f717d79374997ed3b429b2449d2c3b8

jhedberg, 2019-07-03:
If I understand this right, all that's needed is for the new section to be added to the right array in `scripts/sanitycheck`

Laczen, 2019-07-03:
@jhedberg, do you mind if I would rename the `_bt_settings_area` in `scripts/sanitycheck` to `_settings_handlers_area` ?

jhedberg, 2019-07-03:
> @jhedberg, do you mind if I would rename the _bt_settings_area in scripts/sanitycheck to _settings_handlers_area ?

@Laczen yes that's fine with me - there are no more users of _bt_settings_area, right?

Laczen, 2019-07-03:
@jhedberg, the area is not even there anymore.

@andrewboie, I am not a user of sanitycheck, but there has been a change in the storage of settings_handlers. Could you try in line 911 of scripts/sanitycheck to replace `"_bt_settings_area"` to `"_settings_handlers_area"`. Thanks.

Laczen, 2019-07-12:
Fixed by #17500

closed 2019-07-12 by Laczen