zephyrproject-rtos/zephyr #17300: [Zephyr v1.14.0] mcumgr: stm32: Strange Build warnings when counter is enabled

gon1332, 2019-07-03:
**Describe the bug**
When I enable the mcumgr module in our project along with the counter driver, the building process emits warnings about the `uint32_t` vs `u32_t` definitions on code that mcumgr uses.

My main target is the STM32 L4 series and, but it seems that also `nucleo_f401re` is affected.
OTOH, on `nrf51_pca10028` with counter and mcumgr enabled, no warning is emitted.

It seems that although Zephyr defines `PRIx32` as `"%x"`, something else manages and predefines `PRIx32` as `"%lx"` for STM32 platforms. Zephyr does not like that.

**To Reproduce**

Add this `prj.conf` in `samples/hello_world`:
```
CONFIG_MCUMGR=y
CONFIG_MCUMGR_CMD_IMG_MGMT=y

CONFIG_NET_BUF=y
CONFIG_NET_BUF_USER_DATA_SIZE=7

CONFIG_FLASH=y

CONFIG_COUNTER=y
```

Steps to reproduce the behavior:
1. west build -b nucleo_f401re -s zephyr/samples/hello_world
2. See warnings

**Expected behavior**
I expected to have a seamless integration. My main working version is v1.14, where there are warnings about the `uint32_t` vs `u32_t` definitions and the LOG macros. In master everything is OK.

These warning do not seem dangerous in the current context, but in what we do, they are expanded on pointer to `u32_t` vs pointers to `uint32_t` warnings.

**Impact**
Annoyance..

**Screenshots or console output**
```
~/D/zephyrproject ❯❯❯ west build -b nucleo_f401re -s zephyr/samples/hello_world -d build_f401re                 ✘ 130                                                                                                                          
source directory: /home/gon1332/Development/zephyrproject/zephyr/samples/hello_world                                                                                                                                                           
build directory: /home/gon1332/Development/zephyrproject/build_f401re                                                                                                                                                                          
BOARD: nucleo_f401re                                                                                                                                                                                                                           
[0/1] Re-running CMake...                                                                                                                                                                                                                      
Zephyr version: 1.14.0                                                                                                                                                                                                                         
-- Selected BOARD nucleo_f401re                                                                                                                                                                                                                
-- Found west: /home/gon1332/.local/bin/west (found suitable version "0.5.8", minimum required is "0.5.6")                                                                                                                                     
-- Loading /home/gon1332/Development/zephyrproject/zephyr/boards/arm/nucleo_f401re/nucleo_f401re.dts as base                                                                                                                                   
-- Overlaying /home/gon1332/Development/zephyrproject/zephyr/dts/common/common.dts                                                                                                                                                             
Parsing Kconfig tree in /home/gon1332/Development/zephyrproject/zephyr/Kconfig                                                                                                                                                                 
Loading /home/gon1332/Development/zephyrproject/zephyr/boards/arm/nucleo_f401re/nucleo_f401re_defconfig as base                                                                                                                                
Merging /home/gon1332/Development/zephyrproject/zephyr/samples/hello_world/prj.conf                                                                                                                                                            
Configuration written to '/home/gon1332/Development/zephyrproject/build_f401re/zephyr/.config'                                                                                                                                                 
-- Cache files will be written to: /home/gon1332/.cache/zephyr  
Including module: tinycbor in path: /home/gon1332/Development/zephyrproject/modules/lib/tinycbor
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gon1332/Development/zephyrproject/build_f401re
[30/132] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/dfu/img_util/flash_img.c.obj
In file included from /home/gon1332/Development/zephyrproject/zephyr/include/logging/log.h:11,
                 from /home/gon1332/Development/zephyrproject/zephyr/subsys/dfu/img_util/flash_img.c:10:
/home/gon1332/Development/zephyrproject/zephyr/subsys/dfu/img_util/flash_img.c: In function 'flash_verify':
/home/gon1332/Development/zephyrproject/zephyr/subsys/dfu/img_util/flash_img.c:36:12: warning: format '%lx' expects argument of type 'long unsigned int', but argument 3 has type 'unsigned int' [-Wformat=]
    LOG_ERR("flash_read error %d offset=0x%08"PRIx32,
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     rc, (u32_t)offset);
         ~~~~~~~~~~~~~
/home/gon1332/Development/zephyrproject/zephyr/include/logging/log_core.h:236:27: note: in definition of macro '__LOG'
    log_printf_arg_checker(__VA_ARGS__);      \
                           ^~~~~~~~~~~
/home/gon1332/Development/zephyrproject/zephyr/include/logging/log.h:46:25: note: in expansion of macro 'Z_LOG'
 #define LOG_ERR(...)    Z_LOG(LOG_LEVEL_ERR, __VA_ARGS__)
                         ^~~~~
/home/gon1332/Development/zephyrproject/zephyr/subsys/dfu/img_util/flash_img.c:36:4: note: in expansion of macro 'LOG_ERR'
    LOG_ERR("flash_read error %d offset=0x%08"PRIx32,
    ^~~~~~~
/home/gon1332/Development/zephyrproject/zephyr/subsys/dfu/img_util/flash_img.c:42:12: warning: format '%lx' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
    LOG_ERR("offset=0x%08"PRIx32" VERIFY FAIL. "
            ^~~~~~~~~~~~~~
/home/gon1332/Development/zephyrproject/zephyr/subsys/dfu/img_util/flash_img.c:44:5:
     (u32_t)offset, temp, UNALIGNED_GET(data));
     ~~~~~~~~~~~~~
/home/gon1332/Development/zephyrproject/zephyr/include/logging/log_core.h:236:27: note: in definition of macro '__LOG'
    log_printf_arg_checker(__VA_ARGS__);      \
                           ^~~~~~~~~~~
/home/gon1332/Development/zephyrproject/zephyr/include/logging/log.h:46:25: note: in expansion of macro 'Z_LOG'
 #define LOG_ERR(...)    Z_LOG(LOG_LEVEL_ERR, __VA_ARGS__)
                         ^~~~~
/home/gon1332/Development/zephyrproject/zephyr/subsys/dfu/img_util/flash_img.c:42:4: note: in expansion of macro 'LOG_ERR'
    LOG_ERR("offset=0x%08"PRIx32" VERIFY FAIL. "
    ^~~~~~~
/home/gon1332/Development/zephyrproject/zephyr/subsys/dfu/img_util/flash_img.c: In function 'flash_sync':
/home/gon1332/Development/zephyrproject/zephyr/subsys/dfu/img_util/flash_img.c:67:11: warning: format '%lx' expects argument of type 'long unsigned int', but argument 3 has type 'unsigned int' [-Wformat=]
   LOG_ERR("flash_write error %d offset=0x%08" PRIx32, rc,
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    (u32_t)ctx->bytes_written);
    ~~~~~~~~~~~~~~~~~~~~~~~~~
/home/gon1332/Development/zephyrproject/zephyr/include/logging/log_core.h:236:27: note: in definition of macro '__LOG'
    log_printf_arg_checker(__VA_ARGS__);      \
                           ^~~~~~~~~~~
/home/gon1332/Development/zephyrproject/zephyr/include/logging/log.h:46:25: note: in expansion of macro 'Z_LOG'
 #define LOG_ERR(...)    Z_LOG(LOG_LEVEL_ERR, __VA_ARGS__)
                         ^~~~~
/home/gon1332/Development/zephyrproject/zephyr/subsys/dfu/img_util/flash_img.c:67:3: note: in expansion of macro 'LOG_ERR'
   LOG_ERR("flash_write error %d offset=0x%08" PRIx32, rc,
   ^~~~~~~
[127/132] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       26560 B       512 KB      5.07%
            SRAM:        7912 B        96 KB      8.05%
        IDT_LIST:         200 B         2 KB      9.77%
[132/132] Linking C executable zephyr/zephyr.elf
```

**Environment (please complete the following information):**
 - GNU/Linux
 - Zephyr SDK
 - db50b4ec2a0be70832c2a3580cfe446b93359c30 (v1.14-branch)

gon1332, 2019-07-03:
@galazari

gon1332, 2019-07-03:
Furthermore, if I enable the `counter` and only this in `mcuboot/boot/zephyr/prj.conf` and build it, I can see similar warnings. The following is just a snippet:

```
/home/gon1332/Development/Centaur/ioannis-end-device/3rd-party/mcuboot/boot/zephyr/include/mcuboot_config/mcuboot_logging.h:23:30: note: in expansion of macro 'LOG_DBG'                                                                       
 #define MCUBOOT_LOG_DBG(...) LOG_DBG(__VA_ARGS__)                                                                                                                                                                                             
                              ^~~~~~~                                                                                                                                                                                                          
/home/gon1332/Development/Centaur/ioannis-end-device/3rd-party/mcuboot/boot/bootutil/include/bootutil/bootutil_log.h:34:27: note: in expansion of macro 'MCUBOOT_LOG_DBG'                                                                      
 #define BOOT_LOG_DBG(...) MCUBOOT_LOG_DBG(__VA_ARGS__)                                                                                                                                                                                        
                           ^~~~~~~~~~~~~~~                                                                                                                                                                                                     
/home/gon1332/Development/Centaur/ioannis-end-device/3rd-party/mcuboot/boot/bootutil/src/bootutil_misc.c:502:5: note: in expansion of macro 'BOOT_LOG_DBG'                                                                                     
     BOOT_LOG_DBG("writing swap_type; fa_id=%d off=0x%x (0x%x), swap_type=0x%x",                                                                                                                                                               
     ^~~~~~~~~~~~                                                                                                                                                                                                                              
/home/gon1332/Development/Centaur/ioannis-end-device/3rd-party/mcuboot/boot/bootutil/src/bootutil_misc.c: In function 'boot_write_swap_size':                                                                                                  
/home/gon1332/Development/Centaur/ioannis-end-device/3rd-party/mcuboot/boot/bootutil/src/bootutil_misc.c:526:18: warning: format '%x' expects argument of type 'unsigned int', but argument 3 has type 'uint32_t' {aka 'long unsigned int'} [-W
format=]                                                                                                                                                                                                                                       
     BOOT_LOG_DBG("writing swap_size; fa_id=%d off=0x%x (0x%x)",                                                                                                                                                                               
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                                                
                  fap->fa_id, off, fap->fa_off + off);                                                                                                                                                                                         
                              ~~~                                                                                                                                                                                                              
/home/gon1332/Development/Centaur/ioannis-end-device/3rd-party/zephyr/include/logging/log_core.h:236:27: note: in definition of macro '__LOG'                                                                                                  
    log_printf_arg_checker(__VA_ARGS__);      \                                                                                                                                                                                                
                           ^~~~~~~~~~~                                                                                                                                                                                                         
/home/gon1332/Development/Centaur/ioannis-end-device/3rd-party/zephyr/include/logging/log.h:77:25: note: in expansion of macro 'Z_LOG'                                                                                                         
 #define LOG_DBG(...)    Z_LOG(LOG_LEVEL_DBG, __VA_ARGS__)                                                                                                                                                                                     
                         ^~~~~                                                                                                                                                                                                                 
[... cut ...]
[38/216] Building C object CMakeFiles/app.dir/home/gon1332...e/3rd-party/mcuboot/boot/bootutil/src/image_validate.c.obj                                                                                                                        
In file included from /home/gon1332/Development/Centaur/ioannis-end-device/3rd-party/mcuboot/boot/bootutil/src/image_validate.c:44:                                                                                                            
/home/gon1332/Development/Centaur/ioannis-end-device/3rd-party/mcuboot/boot/bootutil/src/bootutil_priv.h: In function 'boot_initialize_area':   
```

gon1332, 2019-07-03:
OK, this https://github.com/zephyrproject-rtos/zephyr/commit/f32330b22c333012cf5f771b0eae9ef378e446d4#diff-af3b638bc2a3e6c650974192a53c7291 seems like a good candidate to solve the problem.  I'll test it.

nvlsianpu, 2019-08-12:
I believe warning you have are fixed by https://github.com/zephyrproject-rtos/zephyr/pull/18158

mnkp, 2019-08-13:
To fix the warnings in 1.14 branch it may be better to backport #15925. I haven't tested this approach.

closed 2020-10-04 by github-actions[bot]