zephyrproject-rtos/zephyr #17307: device tree bindings disallow strings that begin with integers

pabigot, 2019-07-03:
I have a yaml file with this:

    epd-variant:
      type: string
      category: required
      generation: define
      enum:
        - "1.54b"

When processing this:

	epd@0 {
	      compatible = "waveshare,e-paper";
	      reg = <0>;
	      rst-gpios = <&gpio0 2 0>;
	      busy-gpios = <&gpio0 3 0>;
	      dc-gpios = <&gpio0 28 0>;
	      epd-variant = "1.54b";
	};

I get this:
```
  File "/mnt/devel/external/zp/zephyr/scripts/dts/devicetree.py", line 114, in parse_value
    return parse_values(value, '"', '"', ',')
  File "/mnt/devel/external/zp/zephyr/scripts/dts/devicetree.py", line 96, in parse_values
    out = parse_values_internal(value, start, end, separator)
  File "/mnt/devel/external/zp/zephyr/scripts/dts/devicetree.py", line 91, in parse_values_internal
    return parse_value(out[0])
  File "/mnt/devel/external/zp/zephyr/scripts/dts/devicetree.py", line 126, in parse_value
    return int(value, 10)
ValueError: invalid literal for int() with base 10: '1.54b'
```

AFAICT it should be legitimate to have strings that start with digits.

galak, 2019-07-03:
I think this might be fixed by:

https://github.com/zephyrproject-rtos/zephyr/pull/15599

ioannisg, 2019-07-12:
@pabigot could you please check again, since now #15599 is merged?

pabigot, 2019-07-12:
Yes, this appears to work now.

closed 2019-07-12 by pabigot