zephyrproject-rtos/zephyr #17309: device tree yaml generation property unused?

pabigot, 2019-07-03:
As far as I can tell the `generation` property is `define` in all cases except  a few where it's `structures`.

Also, as far as I can tell the `generation` property is read but has absolutely no effect on generation.  The yaml template even says it's ignored.

This is unfortunate, as I had hoped there was an option to produce the macro value for a string property without the quotes around the string.  (And there are cases where I'd love to see arrays and compounds generated as structure initializers like `{1, 2, 3, 4, 5}` instead of independent macros for each element, which can be a real pain when the array isn't a known length.)

What's going on with `generation`?  Can we add new generation options, like `token` to eliminate quotes on strings and `initializer` for compounds/arrays that aren't broken down?  It's still there, so presumably there's a reason....

galak, 2019-07-03:
I'm trying to phase out `generation` in the YAML files.  If you've got an example of what you want generated we can look at doing it.

We're working on a new parser and generation script/tooling.  I'm not sure if it will get completed in time for merging in 2.0.  So we can see if what you need makes sense to add there or to add to the existing code.

pabigot, 2019-07-03:
Below are three enhancements to generation that I would like to see, with real-world examples

### GPIOS

In addition to
```
#define DT_GPIO_KEYS_BUTTON_0_GPIOS_CONTROLLER "GPIO_0"
#define DT_GPIO_KEYS_BUTTON_0_GPIOS_FLAGS      256
#define DT_GPIO_KEYS_BUTTON_0_GPIOS_PIN        18
```
I would like to see
```
#define DT_GPIO_KEYS_BUTTON_0_GPIOS {"GPIO_0", 256, 18}
```
in conjunction with a standard `struct gpio_descriptor` in the `gpio.h` header, allowing:
```
   .mode_btn = DT_GPIO_KEYS_BUTTON_0_GPIOS,
```
This would reduce the amount of noisy boilerplate when instantiating GPIO uses.

### Tokenizable Values

[Waveshare's e-paper displays](https://www.waveshare.com/product/modules/oleds-lcds/e-paper.htm) use the same API with a large number of device-specific values, including initialization sequences and look-up tables.  I would like the driver I'm  developing to allow selection of the display like this:

```
    variant:
      type: string
      category: required
      generation: define
      enum:
        - "epd1p54"
        - "epd1p54b"
        - "epd1p54c"
        - "epd4p2"
        # ...
```

Currently what I get is:
```
#define DT_NORDIC_NRF_SPI_40004000_WAVESHARE_EPD_0_VARIANT "epd1p54b"
```

In the driver I have things like:
```
static u8_t epd1p54b_init[] = { /* lots of bytes here */};
```

Since the value of the property is expressed as a string rather than a token, I can't use it in something like:
```
   .init = CONCAT(DT_INST_0_WAVESHARE_EPD_VARIANT, _init),
```

So I want an option to generate string-valued properties as tokens.

### Array Initializers

It would be nice if an external board that had a display not built into the driver could provide the necessary information in the device-tree bindings, e.g.:

```
vcom-lut = <0x0E 0x14 0x01 0x0A 0x06 0x04 0x0A 0x0A
            0x0F 0x03 0x03 0x0C 0x06 0x0A 0x00>;
```

Currently that expands to:
```
#define DT_NORDIC_NRF_SPI_40004000_WAVESHARE_EPD_0_VCOM_LUT_0   14
#define DT_NORDIC_NRF_SPI_40004000_WAVESHARE_EPD_0_VCOM_LUT_1   20
#define DT_NORDIC_NRF_SPI_40004000_WAVESHARE_EPD_0_VCOM_LUT_10  3
#define DT_NORDIC_NRF_SPI_40004000_WAVESHARE_EPD_0_VCOM_LUT_11  12
#define DT_NORDIC_NRF_SPI_40004000_WAVESHARE_EPD_0_VCOM_LUT_12  6
#define DT_NORDIC_NRF_SPI_40004000_WAVESHARE_EPD_0_VCOM_LUT_13  10
#define DT_NORDIC_NRF_SPI_40004000_WAVESHARE_EPD_0_VCOM_LUT_14  0
#define DT_NORDIC_NRF_SPI_40004000_WAVESHARE_EPD_0_VCOM_LUT_2   1
#define DT_NORDIC_NRF_SPI_40004000_WAVESHARE_EPD_0_VCOM_LUT_3   10
#define DT_NORDIC_NRF_SPI_40004000_WAVESHARE_EPD_0_VCOM_LUT_4   6
#define DT_NORDIC_NRF_SPI_40004000_WAVESHARE_EPD_0_VCOM_LUT_5   4
#define DT_NORDIC_NRF_SPI_40004000_WAVESHARE_EPD_0_VCOM_LUT_6   10
#define DT_NORDIC_NRF_SPI_40004000_WAVESHARE_EPD_0_VCOM_LUT_7   10
#define DT_NORDIC_NRF_SPI_40004000_WAVESHARE_EPD_0_VCOM_LUT_8   15
#define DT_NORDIC_NRF_SPI_40004000_WAVESHARE_EPD_0_VCOM_LUT_9   3
```
It should instead be expanded to an initializer so I can do:
```
#ifdef DT_INST_0_WAVESHARE_EPD_VCOM_LUT
static u8_t vcom_lut[] = DT_INST_0_WAVESHARE_EPD_VCOM_LUT
#else
static u8_t vcom_lut[] = CONCAT(DT_INST_0_WAVESHARE_EPD_VARIANT,_vcom_lut);
#endif
```

jfischer-phytec-iot, 2019-07-03:
I join in here. But I prefer to use `a-byte-data-property` for that, like:

```
        ssd1673fb@0 {                                                                         
                compatible = "solomon,ssd1673fb", "gd,gde0213b1";                             
                label = "SSD1673";                                                            
                spi-max-frequency = <4000000>;
....
                vcom = <0xa8>;                                                                
                border-waveform = <0x71>;
                lut_initial = [22, 55, AA, 55, AA, 55, AA, 11]; 
                lut_default = [22, 11, 00, 55, 00, 55, 00, 00];                            
        };
```

@pabigot OT: I wonder why you write driver for a display and not a display controller. btw, ssd1673 already supports HINK E0154A05.

pabigot, 2019-07-03:
@jfischer-phytec-iot I'm intending to write a driver that conforms to Zephyr's display API so it can be used with lvgl.  I don't believe these displays use the ssd1673, but I'll check into that.

pabigot, 2019-07-03:
Agree, support for `bytestring` generating an initializer is desirable.

jfischer-phytec-iot, 2019-07-03:
@pabigot The driver should be for a display controller not a display and the parameters for a specific panel have to be passed from DT. For example, ssd1673 (will rename it to something generic) supports different controllers (SSD1673 (gde0213b1) and SSD1608 (GDE029A1) compatible),  see `reel_board.dts` and `ssd1673.c`. But because of the problem described above, the LUTs are still in the driver itself.

I prepare ssd1673 to support IL3897 (GDEH0213B72).

galak, 2019-10-28:
@pabigot just doing a quick scan of this, but have we dealt with the issues described here?

pabigot, 2019-10-28:
* GPIOS -- yes (#18588)
* Tokenizable values, Generated Enums -- No.
* Array initializers -- yes (#17395), and in-progress (#20174)

If I create a new issue for the token/enum capability and #20174 gets merged (I will update per your comment), I can close this.  Is that plan OK?

pabigot, 2019-12-10:
The unimplemented capability is now in its own issue.

closed 2019-12-10 by pabigot