zephyrproject-rtos/zephyr #17313: drivers: usb_dc_mcux_ehci

jfischer-phytec-iot, 2019-07-03:
**Describe the bug**
usb_dc_mcux_ehci.c does not compile

**To Reproduce**
Steps to reproduce the behavior:

0. cd tests/subsys/usb/device
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1050_evk ../
3. make
4. See error

**Impact**
CI fails on `tests/subsys/usb/device`, blocks #17236

**Screenshots or console output**
```
In file included from /home/john/zephyr-project/zephyr/include/toolchain.h:39,
                 from /home/john/zephyr-project/zephyr/include/kernel_includes.h:19,
                 from /home/john/zephyr-project/zephyr/soc/arm/nxp_imx/rt/soc.h:24,
                 from /home/john/zephyr-project/zephyr/drivers/usb/device/usb_dc_mcux_ehci.c:7:
/home/john/zephyr-project/zephyr/include/toolchain/gcc.h:27:28: error: expected identifier or '(' before '_Static_assert'
 #define BUILD_ASSERT(EXPR) _Static_assert(EXPR, "")
                            ^~~~~~~~~~~~~~
/home/john/zephyr-project/zephyr/include/kernel.h:4160:2: note: in expansion of macro 'BUILD_ASSERT'
  BUILD_ASSERT(WB_UP(maxsz) >= _MPOOL_MINBLK);   \
  ^~~~~~~~~~~~
/home/john/zephyr-project/zephyr/drivers/usb/device/usb_dc_mcux_ehci.c:42:11: note: in expansion of macro 'K_MEM_POOL_DEFINE'
 __nocache K_MEM_POOL_DEFINE(ep_buf_pool, 16, 1024, EP_BUF_NUMOF_BLOCKS, 4);
           ^~~~~~~~~~~~~~~~~

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.10.1
 - 2f95f4afe7bc437b3038f9bb1336668527a6baf4

jfischer-phytec-iot, 2019-07-03:
removing `__nocache` section macro fixes build. @andyross @andrewboie  any thoughts why it fails on `BUILD_ASSERT`?

ioannisg, 2019-07-03:
This likely breaks CIs for usb -related PRs

ioannisg, 2019-07-03:
I'll investigate tomorrow

ioannisg, 2019-07-04:
Git bisect gives me ace11bbefde384af7e0bf7c2f7b9fc6a1c9c88ca, and this makes sense: the PR added the build-time asserts in kernel.h which give, now errors. (FYI @npitre)

But perhaps that patch itself is not to blame, it might be the some sort of inclusion cycles

jfischer-phytec-iot, 2019-07-04:
@ioannisg [__nocache section macro](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/usb/device/usb_dc_mcux_ehci.c#L42) triggers some issue in K_MEM_POOL_DEFINE

ioannisg, 2019-07-04:
@jfischer-phytec-iot the problem is the build-time assert in kernel.h. If you comment that out, the project builds.

jfischer-phytec-iot, 2019-07-04:
@ioannisg I see, but It builds too if you comment __nocache in usb_dc_mcux_ehci.

ioannisg, 2019-07-04:
Yes! But the problem is not the whole think of defining a mem-pool, it is just the build-time assert inside it, which was added in the commit that i mentioned

jfischer-phytec-iot, 2019-07-04:
`BUILD_ASSERT(WB_UP(maxsz) >= _MPOOL_MINBLK);` conflicts with __section macro, how should it be processed?

I have not looked at everything exactly, e.g. patch below fixes the problem:

```diff
diff --git a/include/kernel.h b/include/kernel.h
index 5cc05767ae..5bf5a90f71 100644
--- a/include/kernel.h
+++ b/include/kernel.h
@@ -4157,7 +4157,6 @@ struct k_mem_pool {
  * @req K-MPOOL-001
  */
 #define K_MEM_POOL_DEFINE(name, minsz, maxsz, nmax, align)             \
-       BUILD_ASSERT(WB_UP(maxsz) >= _MPOOL_MINBLK);                    \
        char __aligned(WB_UP(align)) _mpool_buf_##name[WB_UP(maxsz) * nmax \
                                  + _MPOOL_BITS_SIZE(maxsz, minsz, nmax)]; \
        struct sys_mem_pool_lvl _mpool_lvls_##name[Z_MPOOL_LVLS(maxsz, minsz)]; \
@@ -4170,7 +4169,8 @@ struct k_mem_pool {
                        .levels = _mpool_lvls_##name,                   \
                        .flags = SYS_MEM_POOL_KERNEL                    \
                } \
-       }
+       }; \
+       BUILD_ASSERT(WB_UP(maxsz) >= _MPOOL_MINBLK);

```

jfischer-phytec-iot, 2019-07-04:
I suppose no one thought that an K_MEM_POOL_DEFINE user would combine it with an other macro.

ioannisg, 2019-07-04:
I confirm, at least,  that this is a fix @jfischer-phytec-iot 

Thanks, I think we can live with that. This is a hotfix - it will likely break lots of PR CIs if we don't address it, and, after all, it won't allow developers to work with USBD on that board....

npitre, 2019-07-04:
On Thu, 4 Jul 2019, Johann Fischer wrote:

> I have not looked at everything exactly, e.g. patch below fixes the problem:

If that fixes the problem, then go for it.

ioannisg, 2019-07-04:
@npitre can you comment on #17352 instead?

npitre, 2019-07-04:
On Thu, 4 Jul 2019, Ioannis Glaropoulos wrote:

> @npitre can you comment on #17352 instead?

Yes. Please tag me there so I can reply by email. The web interface is a 
pain to use for me.

PR #17352 (2019-07-04): mempool: move BUILD_ASSERT to the end of K_MEM_POOL_DEFINE
Move BUILD_ASSERT to the end of K_MEM_POOL_DEFINE.
K_MEM_POOL_DEFINE can not be processed if
combined with an other macro like __section.

Fixes: #17313
commit 223a2b95 (2019-07-04): mempool: move BUILD_ASSERT to the end of K_MEM_POOL_DEFINE

Move BUILD_ASSERT to the end of K_MEM_POOL_DEFINE.
K_MEM_POOL_DEFINE can not be processed if
combined with an other macro like __section.

Fixes: #17313
closed 2019-07-04 by andrewboie